openwrt/openwrt #14702: Latest Snapshot Bricks Xiaomi Mi 4A Gigabit Edition

khanjui, 2024-02-23:
### Describe the bug

Power LED becomes solid orange and the device is unreachable.

### OpenWrt version

r25280-08e249d43c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Steps to reproduce

Download the image from https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=xiaomi_mi-router-4a-gigabit or upgrade using attendedsysupgrade.

### Actual behaviour

Device becomes unreachable, I had to de-brick it.

### Expected behaviour

Should work, I guess.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-23:
Orange means it is booting.
How long you waited?
Did you reset configuration upgrading?
Power off device?
Does īt ever get to fast blink led for failsafe boot?

khanjui, 2024-02-23:
> Orange means it is booting. How long you waited? Did you reset configuration upgrading? Power off device? Does īt ever get to fast blink led for failsafe boot?

I've waited 15+ minutes.
Kept configurations.
I couldn't make it into failsafe.

robimarko, 2024-02-23:
This should already be fixed, there was a breakage if attended sysupgrade was used

khanjui, 2024-03-09:
> This should already be fixed, there was a breakage if attended sysupgrade was used

Actually the problem continues, still with today's snapshot using attendedaysupgrade. The symptoms are same.