openwrt/openwrt #14704: Ds-Lite package does not work with https-dns-proxy

beatstick, 2024-02-23:
### Describe the bug

I recently switched my ISP. 
Now I only get ds-lite via cable, meaning the ipv4 Adress is only provided remotely on an aftr server. 

I installed the package ds-lite on openwrt, and it creates a virtual dynamic interface on startup.

Upon installing and setting up https-dns-proxy, it stops working after a reboot of openwrt. I cannot get an internet connection any more. 

I tried starting the service for http-dns-proxy later, so the proxying starts only after the virtual dynamic interface is set up. That kind of works for a time, but the luci interface for htpps-dns-proxy cannot be loaded any more afterwards. Also adblock lists cannot be loaded any more, but that is a different issue maybe.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/64

### Device

10ZiG Technology. 44xx

### Image kind

Official downloaded image

### Steps to reproduce

Install package ds-lite.
Install package https-dns-proxy

### Actual behaviour

Internet cannot be reached via any device on the network any more. 

### Expected behaviour

Internet available.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

beatstick, 2024-02-23:
I replaced one line in the script of /etc/init.d/https-dns-proxy to wait for the wan to connect before starting the service.
Everything seems to be working now, except for adblock, but that might be related to the dns-hppts-proxy configuration.

``
 boot() {
        #ubus -t 30 wait_for network.interface 2>/dev/null
        # above line with line below to wait for internet connnection to come up before starting doh
        ubus -t 30 wait_for network.interface network.wan 2>/dev/null
``

missing233, 2024-02-23:
If your AFTR IP is static, it's better to use the IP address rather than the domain.

brada4, 2024-02-23:
You need to report package bugs in package bug tracker.