openwrt/openwrt #14710: fsck causes oom on large drive

reukiodo, 2024-02-23:
### Describe the bug

When using a large drive formatted ext4 with data, the boot is very slow, and seems to be caused by a file check which runs out of memory.

I do NOT have 'Run filesystem check' enabled in System > Mount Points, yet it seems to run at boot anyway.
I've enabled zram-swap and upped its priority to 11 and lowered fstab to 12, but the zram still seems to enable after the file check. I can successfully `umount /dev/sda && fsck.ext4 /dev/sda` with zram swap enabled and active.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7620

### Device

Kimax U35WF

### Image kind

Official downloaded image

### Steps to reproduce

1. install >14TB drive
2. `mkfs.ext4 /dev/sda`
3. copy >3TB to drive
4. install zram-swap
5. set /dev/sda mount without filesystem check at boot
6. swap priorities of zram and fstab startup
7. reboot

### Actual behaviour

Device takes quite a long time before becoming responsive on the network again after poweron / reboot.

### Expected behaviour

Device boots up quickly with zram enabled before filesystem mounts, and fsck only runs when the checkbox is enabled.

### Additional info

https://forum.openwrt.org/t/blueendless-kimax-bs-u35-wf/187587
[image]
```
[    0.000000] Linux version 5.15.137 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Tue Nov 14 13:38:11 2023
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is Kimax U35WF
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[... 239 lines cut ...]
[  599.587571] zram0: detected capacity change from 0 to 262144
[  599.714812] Adding 131068k swap on /dev/zram0.  Priority:100 extents:1 across:131068k SSDsc
[  633.663228] 8021q: adding VLAN 0 to HW filter on device eth0
[  633.753422] br-lan: port 1(eth0) entered blocking state
[  633.763950] br-lan: port 1(eth0) entered disabled state
[  633.774809] device eth0 entered promiscuous mode
[  633.923155] br-lan: port 1(eth0) entered blocking state
[  633.933698] br-lan: port 1(eth0) entered forwarding state
[  634.678859] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  637.343750] ksmbd: The ksmbd server is experimental, use at your own risk.
[  645.311674] ksmbd: The ksmbd server is experimental, use at your own risk.
[  666.423770] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
```

### Diffconfig

```text
root@U35WF:~#  ./scripts/diffconfig.sh
-ash: ./scripts/diffconfig.sh: not found
root@U35WF:~#
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

reukiodo, 2024-02-23:
```
root@U35WF:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 5.3M      5.3M         0 100% /rom
tmpfs                    59.7M    112.0K     59.6M   0% /tmp
/dev/mtdblock6            8.4M      3.4M      5.0M  40% /overlay
overlayfs:/overlay        8.4M      3.4M      5.0M  40% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/sda                 14.4T      3.6T     10.1T  26% /home
root@U35WF:~# free -h
              total        used        free      shared  buff/cache   available
Mem:         122220       40748       41308          24       40164       36804
Swap:        131068        2816      128252
root@U35WF:~# lsblk
NAME      MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda         8:0    0 14.6T  0 disk /home
mtdblock0  31:0    0  192K  1 disk 
mtdblock1  31:1    0   64K  1 disk 
mtdblock2  31:2    0   64K  1 disk 
mtdblock3  31:3    0 15.7M  0 disk 
mtdblock4  31:4    0  2.2M  1 disk 
mtdblock5  31:5    0 13.5M  1 disk /rom
mtdblock6  31:6    0  8.4M  0 disk /overlay
zram0     254:0    0  128M  0 disk [SWAP]
root@U35WF:~# 
```

reukiodo, 2024-02-23:
[image]

brada4, 2024-02-23:
You need serious swap partition.
Or xfs if it mounts at all it never needs fsck.
You should not set zramswap bigger than memtotal in meminfo.
Probably vendor meant something saying 6tb is maximum disk size.

reukiodo, 2024-02-23:
What is interesting is that fsck doesn't fail or OOM after zram swap is useable, it can fsck the entire ~15TB after a while.

I think maybe there are actually 2 issues here::
1. fsck runs even when configured not to
2. zram swap doesn't complete until after fsck OOMs

Is there some service other than fstab which could cause the fsck to start before zram? Or is there a way to force fsck to wait for zram to finish?

reukiodo, 2024-02-23:
Well, I found a workaround to vastly speed up boot:

in System > Mount Points : disable automount for /dev/sda
in System > Startup > Local Startup : add `mount /dev/sda /home` before the `exit 0`

Now reboot takes ~1m vs the ~10m it used to.

boot log:
```
[    0.000000] Linux version 5.15.137 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Tue Nov 14 13:38:11 2023
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is Kimax U35WF
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 120996K/131072K available (5929K kernel code, 608K rwdata, 740K rodata, 1224K init, 225K bss, 10076K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
[    0.000000] systick: enable autosleep mode
[    0.000000] systick: running - mult: 214748, shift: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
[    0.000002] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
[    0.015584] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.087799] pid_max: default: 32768 minimum: 301
[    0.098157] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.112595] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.136530] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.156043] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.169753] pinctrl core: initialized pinctrl subsystem
[    0.181876] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.194321] thermal_sys: Registered thermal governor 'step_wise'
[    0.210697] rt2880_gpio 10000600.gpio: registering 24 gpios
[    0.233664] rt2880_gpio 10000600.gpio: registering 24 irq handlers
[    0.246455] rt2880_gpio 10000660.gpio: registering 32 gpios
[    0.257501] rt2880_gpio 10000660.gpio: registering 32 irq handlers
[    0.270172] rt2880_gpio 10000688.gpio: registering 1 gpios
[    0.281032] rt2880_gpio 10000688.gpio: registering 1 irq handlers
[    0.295959] clocksource: Switched to clocksource systick
[    0.308415] NET: Registered PF_INET protocol family
[    0.318372] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.333893] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.350488] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.365813] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.381053] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.395073] TCP: Hash tables configured (established 1024 bind 1024)
[    0.407909] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.420850] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.435163] NET: Registered PF_UNIX/PF_LOCAL protocol family
[... 133 lines cut ...]
[   22.197620] NET: Registered PF_PPPOX protocol family
[   22.343115] rt2800_wmac 10180000.wmac: loaded eeprom from mtd device "factory"
[   22.357580] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected
[   22.373075] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 7620 detected
[   22.388684] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   22.498642] kmodloader: done loading kernel modules from /etc/modules.d/*
[  113.528046] zram0: detected capacity change from 0 to 262144
[  113.706988] Adding 131068k swap on /dev/zram0.  Priority:100 extents:1 across:131068k SSDsc
[  125.141032] random: crng init done
[  125.147852] random: 89 urandom warning(s) missed due to ratelimiting
[  159.678315] exFAT-fs (sda): invalid boot record signature
[  159.689183] exFAT-fs (sda): failed to read boot sector
[  159.699457] exFAT-fs (sda): failed to recognize exfat type
[  160.932392] urngd: v1.0.2 started.
[... cut ...]
[  164.933236] device eth0 entered promiscuous mode
[  165.101219] br-lan: port 1(eth0) entered blocking state
[  165.111700] br-lan: port 1(eth0) entered forwarding state
[  165.871110] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  179.188312] EXT4-fs (sda): mounted filesystem with ordered data mode. Opts: (null). Quota mode: disabled.
[  211.881619] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[  212.549435] ksmbd: The ksmbd server is experimental, use at your own risk.
```

brada4, 2024-02-23:
Sotty you have surplus mounts, probably  intended fsck config works if you use blkid.
[  159.678315] exFAT-fs (sda): invalid boot record signature

brada4, 2024-02-24:
You have to use guid from block detect/block info.
Show /etc/config/fstab ?

reukiodo, 2024-02-25:
With issue:
```
root@U35WF:~# cat /etc/config/fstab 

config global
        option anon_swap '0'
        option anon_mount '0'
        option auto_swap '1'
        option auto_mount '1'
        option delay_root '5'
        option check_fs '1'

config mount
        option target '/home'
        option uuid '4b41b3ca-9e12-48cd-8d01-7cbdd88c1261'
        option enabled '1'

root@U35WF:~# 
```
With workaround:
```
root@U35WF:~# cat /etc/config/fstab 

config global
        option anon_swap '0'
        option anon_mount '0'
        option auto_swap '1'
        option auto_mount '1'
        option delay_root '5'
        option check_fs '1'

config mount
        option target '/home'
        option uuid '4b41b3ca-9e12-48cd-8d01-7cbdd88c1261'
        option enabled '0'

root@U35WF:~# cat /etc/rc.local 
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

mount -t ext4 /dev/sda /home
exit 0
root@U35WF:~# 
```

Without the `-t ext4` the kernel logs show the `exFAT-fs (sda)` errors during filesystem auto-detection. When `-t ext4` is present in the local startup, then those messages disappear from the kernel log.

reukiodo, 2024-02-25:
Another working example, where zram is fully initialized and fsck completes without OoM:
```
root@U35WF:~# cat /etc/config/fstab 

config global
        option anon_swap '0'
        option anon_mount '0'
        option auto_swap '1'
        option auto_mount '1'
        option delay_root '5'
        option check_fs '1'

config mount
        option target '/home'
        option uuid '4b41b3ca-9e12-48cd-8d01-7cbdd88c1261'
        option enabled '0'

root@U35WF:~# cat /etc/rc.local 
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

fsck.ext4 -fy /dev/sda
mount -t ext4 /dev/sda /home
exit 0
root@U35WF:~# 
```

brada4, 2024-02-25:
Just set to zero/ remove delay_root. Root filesystem contains zram script. Forcing fsck is waste of time, it is generally not needed with journaling fs, and in worst cases will not mount corrupted fs anyway

reukiodo, 2024-02-27:
Ah ha! So the global 'check_fs' config was overriding the individual mount's file check preference!

Perhaps it is because when the option is modified in LuCI, with the mount's fsck:
```
root@U35WF:~# cat /etc/config/fstab 

config global
        option anon_swap '0'
        option anon_mount '0'
        option auto_swap '1'
        option auto_mount '1'
        option delay_root '0'
        option check_fs '1'

config mount
        option target '/home'
        option uuid '4b41b3ca-9e12-48cd-8d01-7cbdd88c1261'
        option enabled '0'
        option enabled_fsck '1'

root@U35WF:~# 
```
where without it is:
```
root@U35WF:~# cat /etc/config/fstab 

config global
        option anon_swap '0'
        option anon_mount '0'
        option auto_swap '1'
        option auto_mount '1'
        option delay_root '0'
        option check_fs '1'

config mount
        option target '/home'
        option uuid '4b41b3ca-9e12-48cd-8d01-7cbdd88c1261'
        option enabled '0'

root@U35WF:~# 
```
So by removing the explicit check, it just falls back to the default / global option?

brada4, 2024-02-27:
You can also uninstall fsck,xyz
Yes, the default in individual option absent is taken from global one.

reukiodo, 2024-02-29:
Uninstalling fsck isn't a great workaround as you then lose the ability to manually check the filesystem when needed.

The other workarounds all seem to mask the bug, though not ideal.

brada4, 2024-03-01:
I think 5s root delay to accomodate extroot is confusing. Including scripts.

Fs should not be checked every time:
https://github.com/openwrt/fstools/blob/08cd7083cac4bddf88459efa0881ee52858e7d0a/block.c#L808
And normally only replay journal before mount.

brada4, 2024-03-01:
Hmm, add `STOP=89` to have specific block unmount to run before umount -a at STOP=90, maybe it keeps filesystem clean enough to go with quick fsck.

brada4, 2024-03-01:
i.e I see 2 problems:
- zram-swap start order is late (fstab start is too early, there is no space between 10 start and 11 fstab), you are absolutely right it needs to be early.
- unmount might not be clean - you need to stop nfsd/samba before unmounting, also syslog if you store that on that disk. it should be 5-seconds for fsck -p run, seems you get full fsck every run hinting dirty filesystem shutdown.
once you change start/stop priorities you need to "enable" script to adjust start symlinks.

reukiodo, 2024-03-01:
for zram / fstab - what priorities should the be set at to guarantee zram completes before fstab?

reukiodo, 2024-03-01:
For unmount, it is probably always unclean, as there is no 'shutdown' command in LuCI, and this device has a hard on/off switch in the back - so... no early warning to OpenWrt that someone is turning it off

This device will be an offsite backup, so I cannot ensure how they will operate it, and must assume that someone seeing the switch will use it, so that leads to the preference of running fsck before mount to assume every shutdown was unclean.

brada4, 2024-03-01:
There are poweroff and reboot commands, later is exactly one you need for tests, and former helps before poweroff IF clean umount is attempted.

It is old init that executes scripts from /etc/rc.d in order.
Make start of /etc/init.d/zram-swap START=5
And add STOP=89 in /etc/init.d/fstab
And move last /etc/init.d/umount to STOP=91

Then disable fileseyetem automount including /etc/rc.local and try few times fsck -p after reboots with filesystem mounted. It should always say that filesystem is clean, if it ever replays log it means clean umount was not achieved.

Then make any NFS or SMB server explicitly stop earlier.
Note you need to enable each script after changing start priorites.

reukiodo, 2024-03-01:
At least we've now identified 2 clear ways to work around the issue:
1. disable all fsck and set root wait to 0:
```
config global
        option delay_root '0'
        option check_fs '0'
```
2. disable mount in /etc/config/fstab and manually fsck and mount the drive in /etc/rc.local after zram completes

brada4, 2024-03-01:
Earlier start of zram eliminates need of disabling fsck. Just that there is no space between 10boot and 11fstab

reukiodo, 2024-03-01:
Does zram have a dependence on fstab ?

brada4, 2024-03-02:
no, but it needs sys proc and dev mounted.