openwrt/openwrt #14721: Stack trace: gcc_usb1_master_clk status stuck at 'on'

testuser7, 2024-02-25:
### Describe the bug

During boot from persistent storage this stack trace is displayed:
```
[    3.296810] ------------[ cut here ]------------
[    3.301529] gcc_usb1_master_clk status stuck at 'on'
[    3.301547] WARNING: CPU: 2 PID: 1 at clk_branch_wait+0x11c/0x130
[    3.311254] Modules linked in:
[    3.317239] CPU: 2 PID: 1 Comm: swapper/0 Not tainted 6.1.79 #0
[    3.320194] Hardware name: Linksys MX5300 (DT)
[    3.326009] pstate: 604000c5 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    3.330525] pc : clk_branch_wait+0x11c/0x130
[    3.337379] lr : clk_branch_wait+0x11c/0x130
[    3.341892] sp : ffffffc008c7bc70
[    3.346143] x29: ffffffc008c7bc70 x28: 0000000000000000 x27: ffffffc008a000c8
[    3.349363] x26: ffffffc008a56068 x25: ffffffc008a56078 x24: ffffffc0089e7710
[    3.356482] x23: ffffffc0089697f8 x22: 0000000000000000 x21: ffffffc0084050b0
[    3.363599] x20: 0000000000000000 x19: ffffffc008be45d0 x18: 00000000000000d7
[    3.370717] x17: ffffffc008becd08 x16: ffffffc008becc98 x15: ffffffc008b57248
[    3.377836] x14: 0000000000000285 x13: 00000000000000d7 x12: 00000000ffffffea
[    3.384954] x11: 00000000ffffefff x10: ffffffc008baf248 x9 : ffffffc008b571f0
[    3.392071] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
[    3.399190] x5 : 0000000000000fff x4 : 0000000000000000 x3 : ffffffc008c7ba50
[    3.406308] x2 : ffffffc008b57188 x1 : ffffffc008b57188 x0 : 0000000000000028
[    3.413427] Call trace:
[    3.420535]  clk_branch_wait+0x11c/0x130
[    3.422795]  clk_branch2_disable+0x2c/0x40
[    3.426961]  clk_disable_unused_subtree+0xa8/0x108
[    3.430869]  clk_disable_unused_subtree+0x2c/0x108
[    3.435642]  clk_disable_unused_subtree+0x2c/0x108
[    3.440417]  clk_disable_unused_subtree+0x2c/0x108
[    3.445192]  clk_disable_unused_subtree+0x2c/0x108
[    3.449967]  clk_disable_unused+0x50/0xd8
[    3.454738]  do_one_initcall+0x50/0x1d0
[    3.458818]  kernel_init_freeable+0x234/0x29c
[    3.462465]  kernel_init+0x28/0x130
[    3.466977]  ret_from_fork+0x10/0x20
[    3.470277] ---[ end trace 0000000000000000 ]---
[    3.474417] ------------[ cut here ]------------
[    3.478699] gcc_usb0_master_clk status stuck at 'on'
[    3.478711] WARNING: CPU: 2 PID: 1 at clk_branch_wait+0x11c/0x130
[    3.488250] Modules linked in:
[    3.494235] CPU: 2 PID: 1 Comm: swapper/0 Tainted: G        W          6.1.79 #0
[    3.497194] Hardware name: Linksys MX5300 (DT)
[    3.504739] pstate: 604000c5 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    3.508997] pc : clk_branch_wait+0x11c/0x130
[    3.515851] lr : clk_branch_wait+0x11c/0x130
[    3.520364] sp : ffffffc008c7bc70
[    3.524617] x29: ffffffc008c7bc70 x28: 0000000000000000 x27: ffffffc008a000c8
[    3.527835] x26: ffffffc008a56068 x25: ffffffc008a56078 x24: ffffffc0089e7710
[    3.534953] x23: ffffffc0089698b0 x22: 0000000000000000 x21: ffffffc0084050b0
[    3.542071] x20: 0000000000000000 x19: ffffffc008be4950 x18: 00000000000000f9
[    3.549190] x17: ffffffc008becd08 x16: ffffffc008becc98 x15: ffffffc008b57248
[    3.556308] x14: 00000000000002eb x13: 00000000000000f9 x12: 00000000ffffffea
[    3.563425] x11: 00000000ffffefff x10: ffffffc008baf248 x9 : ffffffc008b571f0
[    3.570543] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
[    3.577662] x5 : 0000000000000fff x4 : 0000000000000000 x3 : ffffffc008c7ba50
[    3.584780] x2 : ffffffc008b57188 x1 : ffffffc008b57188 x0 : 0000000000000028
[    3.591898] Call trace:
[    3.599007]  clk_branch_wait+0x11c/0x130
[    3.601266]  clk_branch2_disable+0x2c/0x40
[    3.605433]  clk_disable_unused_subtree+0xa8/0x108
[    3.609340]  clk_disable_unused_subtree+0x2c/0x108
[    3.614116]  clk_disable_unused_subtree+0x2c/0x108
[    3.618890]  clk_disable_unused_subtree+0x2c/0x108
[    3.623664]  clk_disable_unused_subtree+0x2c/0x108
[    3.628438]  clk_disable_unused+0x50/0xd8
[    3.633212]  do_one_initcall+0x50/0x1d0
[    3.637290]  kernel_init_freeable+0x234/0x29c
[    3.640937]  kernel_init+0x28/0x130
[    3.645450]  ret_from_fork+0x10/0x20
[    3.648749] ---[ end trace 0000000000000000 ]---
```

### OpenWrt version

r25318-3e8d2f2439

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Linksys MX5300

### Image kind

Official downloaded image

### Additional info

This stack trace did not exist before.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-02-25:
Hm, this should not be possible as there is a GDSC that needs to be enabled before any USB clocks can be powered and since USB1 is disabled it should be as well.

Can you post the whole bootlog?

testuser7, 2024-02-25:
> Can you post the whole bootlog?

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.79 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r25318-3e8d2f2439) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sun Feb 25 01:50:31 2024
[    0.000000] Machine model: Linksys MX5300
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x00000000510fffff]
[    0.000000]   node   0: [mem 0x0000000051100000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] psci: OSI mode supported.
[    0.000000] psci: failed to set PC mode: -1
[    0.000000] percpu: Embedded 18 pages/cpu s34664 r8192 d30872 u73728
[    0.000000] pcpu-alloc: s34664 r8192 d30872 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=ubi0:ubifs rootfstype=ubifs rootwait swiotlb=1 root=/dev/ubiblock0_0 rootfstype=squashfs ro
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 887000K/1048576K available (8192K kernel code, 878K rwdata, 2392K rodata, 1344K init, 291K bss, 161576K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000108] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000121] pid_max: default: 32768 minimum: 301
[    0.000582] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000596] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002427] cblist_init_generic: Setting adjustable number of callback queues.
[    0.002436] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.002628] rcu: Hierarchical SRCU implementation.
[    0.002632] rcu: 	Max phase no-delay instances is 1000.
[    0.003452] smp: Bringing up secondary CPUs ...
[    0.004088] Detected VIPT I-cache on CPU1
[    0.004231] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.004892] Detected VIPT I-cache on CPU2
[    0.004994] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.005609] Detected VIPT I-cache on CPU3
[    0.005710] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.005776] smp: Brought up 1 node, 4 CPUs
[    0.005785] SMP: Total of 4 processors activated.
[    0.005790] CPU features: detected: 32-bit EL0 Support
[    0.005794] CPU features: detected: CRC32 instructions
[    0.005858] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.005864] CPU: All CPU(s) started at EL1
[    0.005867] alternatives: applying system-wide alternatives
[    0.016790] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016817] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[... 298 lines cut ...]
[   16.206581] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
[   17.424343] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[   17.427872] kmodloader: done loading kernel modules from /etc/modules.d/*
[   17.468470] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[   17.475879] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   17.475926] ath11k c000000.wifi: fw_version 0x290104a5 fw_build_timestamp 2023-08-02 20:32 fw_build_id WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1
[   17.810758] ath11k c000000.wifi: htt event 48 not handled
[   19.536668] br-lan: port 1(lan1) entered blocking state
[   19.536710] br-lan: port 1(lan1) entered disabled state
[... cut ...]
[   19.572160] device lan3 entered promiscuous mode
[   19.580487] br-lan: port 4(lan4) entered blocking state
[   19.582022] br-lan: port 4(lan4) entered disabled state
[   19.587218] device lan4 entered promiscuous mode
[   22.714596] nss-dp 3a001600.dp4 lan4: PHY Link up speed: 1000
[   22.714678] br-lan: port 4(lan4) entered blocking state
[   22.719333] br-lan: port 4(lan4) entered forwarding state
[   22.724681] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   33.754279] l11: disabling
```

robimarko, 2024-03-06:
That is just so weird as I cannot reproduce this.

Does this device have USB3 or USB2 wired up and enabled?
Because my Dynalink has only USB3 enabled and I am not getting any issues, also this networking issue is completely unrelated.

testuser7, 2024-03-07:
The problem is observed after this commit: https://github.com/openwrt/openwrt/commit/33e72e0b3d3ba90b2eeeab1c61ff8597693c000d

There is only one USB port:
```
root@OpenWrt:~# cat /sys/kernel/debug/usb/devices 

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.01
S:  Manufacturer=Linux 6.1.79 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.01
S:  Manufacturer=Linux 6.1.79 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms
```

robimarko, 2024-03-09:
I really tried to reproduce this on 3 boards but I just cannot, does it happen only when booting initramfs or when flashed?

testuser7, 2024-03-11:
> does it happen only when booting initramfs or when flashed?

Both.

I will try to revert this changes:
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb?h=v6.1.79&id=3fd6c16ec78da4349de068529e2a175723dbdd0a
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb?h=v6.1.79&id=d31b886ed6a5095214062ee4fb55037eb930adb6
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb?h=v6.1.79&id=9f754d009483791c14eef98095369a3a4ea66cac
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb?h=v6.1.79&id=2888258144e08d1f7905c0ee5df8c14d26d23699
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb?h=v6.1.79&id=0ecc97c81e3aea3335ea4a0b0f1777aae42b71ec
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/usb?h=v6.1.79&id=57e2e42ccd3cd6183228269715ed032f44536751

testuser7, 2024-03-11:
USB info from u-boot:
```
IPQ807x# usb start
starting USB...
USB0:   Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus 0 for devices... 1 USB Device(s) found
USB1:   Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus 1 for devices... 1 USB Device(s) found
IPQ807x# usb info
1: Hub,  USB Revision 3.0
 - U-Boot XHCI Host Controller 
 - Class: Hub
 - PacketSize: 9  Configurations: 1
 - Vendor: 0x0000  Product 0x0000 Version 1.0
   Configuration: 1
   - Interfaces: 1 Self Powered 0mA
     Interface: 0
     - Alternate Setting 0, Endpoints: 1
     - Class Hub
     - Endpoint 1 In Interrupt MaxPacket 8 Interval 255ms

1: Hub,  USB Revision 3.0
 - U-Boot XHCI Host Controller 
 - Class: Hub
 - PacketSize: 9  Configurations: 1
 - Vendor: 0x0000  Product 0x0000 Version 1.0
   Configuration: 1
   - Interfaces: 1 Self Powered 0mA
     Interface: 0
     - Alternate Setting 0, Endpoints: 1
     - Class Hub
     - Endpoint 1 In Interrupt MaxPacket 8 Interval 255ms

IPQ807x# usb tree
USB device tree:
  1  Hub (5 Gb/s, 0mA)
     U-Boot XHCI Host Controller 
   
  1  Hub (5 Gb/s, 0mA)
     U-Boot XHCI Host Controller 
   
```

testuser7, 2024-03-22:
Is it possible to configure build system to download a specific kernel snapshot version from a url: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/snapshot/linux-57e2e42ccd3cd6183228269715ed032f44536751.tar.gz ?

robimarko, 2024-03-22:
No, but you can use an external kernel with OpenWrt buildroot

testuser7, 2024-03-23:
The problem appeared with this commit: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.1.79&id=9359ff1a4501ea7286aec70b90f88fac66a6aea5

robimarko, 2024-03-23:
Well, that commit then probably just uncovered a bug about the GDSC configuration and is not the root cause.

But, I just cannot reproduce this on 3 different boards.

What is the USB config your board has?
Both controllers active, if not which one is enabled?

testuser7, 2024-03-27:
> Well, that commit then probably just uncovered a bug about the GDSC configuration and is not the root cause.

Is there anything I can check on the OEM firmware?

robimarko, 2024-03-27:
No, since in the stock FW they just enable GDSC by default via regmap and I think that even USB clocks are marked as not to be disabled.

Without a datasheet I can only think of the GDSC being buggy as in the IPQ6018

testuser7, 2024-03-27:
> Without a datasheet I can only think of the GDSC being buggy as in the IPQ6018

Datasheet for IPQ8072A?

robimarko, 2024-03-27:
Yes, or IPQ6018 cause they seem to have identical GDSC-s

testuser7, 2024-03-27:
Can [this](https://download.siliconexpert.com/pdfs/2018/6/1/5/4/14/17/qcm_/manual/ipq-8078-0-772fcbga-mt-02-0.pdf) document be helpful?

robimarko, 2024-03-27:
Sadly no, I have those but they are just specification sheets, only useful thing in them is usually GPIO pin functions

testuser7, 2024-03-28:
I found something regarding GDSC in the GPL sources:
```
+	/* Enable SW_COLLAPSE for USB0 GDSCR */
+	regmap_update_bits(regmap, 0x3e078, BIT(0), 0x0);
+	/* Enable SW_COLLAPSE for USB1 GDSCR */
+	regmap_update_bits(regmap, 0x3f078, BIT(0), 0x0); 
```
But it's probably already in the kernel sources.

robimarko, 2024-03-28:
That is exatly what I said, in QSDK they just manually enable GDSC-s on probe and that's it

testuser7, 2024-03-28:
The GPL sources for the MX8500 have the following code:
```
+	/* Disable SW_COLLAPSE for USB0 GDSCR */
+	regmap_update_bits(regmap, 0x3e078, BIT(0), 0x0);
+	/* Enable SW_OVERRIDE for USB0 GDSCR */
+	regmap_update_bits(regmap, 0x3e078, BIT(2), BIT(2));
+	/* Disable SW_COLLAPSE for USB1 GDSCR */
+	regmap_update_bits(regmap, 0x3f078, BIT(0), 0x0);
+	/* Enable SW_OVERRIDE for USB1 GDSCR */
+	regmap_update_bits(regmap, 0x3f078, BIT(2), BIT(2));
```

testuser7, 2024-03-28:
Is there any workaround for this problem?

robimarko, 2024-03-28:
Not that I know of as details about these GDSC-s are not public other than their MMIO address and some control bits.

Basically, for now you can just ignore it since USB should still work just fine, kernel just wont be able to shut down GDSC and clocks in order to save on power if USB is not enabled on the board