openwrt/openwrt #14734: bcm2712 (rpi5) snapshot is hardcoded for a root file system on an SD card

greyltc, 2024-02-26:
### Describe the bug

These days, the raspberry pi firmware is capable of booting from nvme, USB or micro sd. The factory squashfs image i've just tested [provided here](https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2712/openwrt-bcm27xx-bcm2712-rpi-5-squashfs-factory.img.gz) (date Mon Feb 26 03:49:12 2024) has root=/dev/mmcblk0p2 in cmdline.txt in the boot partition, meaning that the distro is hardcoded to only find a root fs on the 2nd partition of an sd card.

I don't think there's a good reason to hardcode that in. Replacing `root=/dev/mmcblk0p2` with `root=PARTUUID=5452574f-02` in this image means I can boot without issue if I flash the image onto either a USB or an sd card (nvme drive untested).

### OpenWrt version

r25325

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Steps to reproduce

Flash a rpi5 image onto a usb stick and try to boot. I assume other rpi images are also hard coded for only sd card boots too.

### Actual behaviour

Device hangs forever during boot trying to find a root file system at /dev/mmcblk0p2, a place that doesn't exist.

### Expected behaviour

Device boots

### Additional info

I suggest you change the `root=` kernel parameter in `cmdline.txt` from `root=/dev/mmcblk0p2` to something like `root=PARTUUID=5452574f-02` to make it boot media agnostic.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

greyltc, 2024-02-26:
yes

obsy, 2024-02-27:
See https://github.com/openwrt/openwrt/pull/14520

Noltari, 2024-04-25:
Fixed by https://github.com/openwrt/openwrt/commit/fe0081eecf43bfd92ac68aa0f3ce7165aaddb4f2

closed 2024-04-25 by Noltari