openwrt/openwrt #14739: Xiaomi ax3600 - Cannot connect between local clients that are connected to same wifi radio

mmichalko, 2024-02-27:
### Describe the bug

Hi,

i tried solving this issue in [openwrt forum](https://forum.openwrt.org/t/xiaomi-ax3600-cannot-connect-between-local-clients-that-are-connected-to-same-wifi-radio/171869) some time ago, but no luck. 

Basically it is not possible to reach local clients connected to same wifi network. 

I tried to explicitly set Isolate clients option to false, but it did not help. I tried attended sysupgrade without keeping options and setting up everything again. 

Everytime at the beginning, the local clients are nicely reachable. But after some time, day or so, they stopped being reachable. I cannot ssh from my phone to my pc, if they are connected to same 5 Ghz radio. If i connect my phone to other radio on 2.4 GHz network, it suddenly works. I connect back to 5 GHz and it does not work again. If i ssh to router and run wifi command, it resets and connection with same wifi radio works again for a short while. It 

is frustrating, because i cannot use local network to stream videos to my TV etc, basically it renders the router unusable. I also dont have any complex settings after vanilla installation, just set up wifi, Adblock (tried turning it off) and some port forwards.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Just set up wifi network for 5 GHz radio, then other network for 2.4 Ghz radio. IoT radio is turned off. 
Then connect 2 devices to 5 GHz radio and try to reach 2nd device from 1st device, for example ssh to 2nd device.
If it works, good, try 1 day later. It will stop working.

### Actual behaviour

It is in the title and description, cannot reach any other device connected to same wifi radio after about a day until the router is restarted or wifi command is run.

### Expected behaviour

I want to reach my other devices on the local network.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-27:
Try to work around issue by passing sta to sta traffic over br-lan
```
option isolate1
```
https://openwrt.org/docs/guide-user/network/wifi/basic

mmichalko, 2024-02-28:
`config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/20000000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '36'
	option band '5g'
	option htmode 'VHT80'
	option disabled '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
	option multicast_to_unicast '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi'
	option channel 'auto'
	option band '5g'
	option htmode 'HE160'
	option country 'AT'
	option cell_density '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'Fastest internet'
	option encryption 'psk2'
	option key 'REMOVED'
	option multicast_to_unicast '0'
	option isolate '0'

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi+1'
	option channel 'auto'
	option band '2g'
	option htmode 'HE20'
	option country 'AT'
	option cell_density '0'

config wifi-iface 'wifinet4'
	option device 'radio2'
	option mode 'ap'
	option ssid 'Fastest internet 2400Mhz'
	option encryption 'psk2'
	option key 'REMOVED'
	option network 'lan'
	option multicast_to_unicast '0'
	option isolate '0'
`

`root@OpenWrt:~# sysctl -a 2> /dev/null  | grep arp
net.ipv4.conf.all.arp_accept = 0
net.ipv4.conf.all.arp_announce = 0
net.ipv4.conf.all.arp_filter = 0
net.ipv4.conf.all.arp_ignore = 1
net.ipv4.conf.all.arp_notify = 0
net.ipv4.conf.all.drop_gratuitous_arp = 0
net.ipv4.conf.all.proxy_arp = 0
net.ipv4.conf.all.proxy_arp_pvlan = 0
net.ipv4.conf.br-lan.arp_accept = 0
net.ipv4.conf.br-lan.arp_announce = 0
net.ipv4.conf.br-lan.arp_filter = 0
net.ipv4.conf.br-lan.arp_ignore = 1
net.ipv4.conf.br-lan.arp_notify = 0
net.ipv4.conf.br-lan.drop_gratuitous_arp = 0
net.ipv4.conf.br-lan.proxy_arp = 0
net.ipv4.conf.br-lan.proxy_arp_pvlan = 0
net.ipv4.conf.default.arp_accept = 0
net.ipv4.conf.default.arp_announce = 0
net.ipv4.conf.default.arp_filter = 0
net.ipv4.conf.default.arp_ignore = 1
net.ipv4.conf.default.arp_notify = 0
net.ipv4.conf.default.drop_gratuitous_arp = 0
net.ipv4.conf.default.proxy_arp = 0
net.ipv4.conf.default.proxy_arp_pvlan = 0
net.ipv4.conf.lan1.arp_accept = 0
net.ipv4.conf.lan1.arp_announce = 0
net.ipv4.conf.lan1.arp_filter = 0
net.ipv4.conf.lan1.arp_ignore = 0
net.ipv4.conf.lan1.arp_notify = 0
net.ipv4.conf.lan1.drop_gratuitous_arp = 0
net.ipv4.conf.lan1.proxy_arp = 0
net.ipv4.conf.lan1.proxy_arp_pvlan = 0
net.ipv4.conf.lan2.arp_accept = 0
net.ipv4.conf.lan2.arp_announce = 0
net.ipv4.conf.lan2.arp_filter = 0
net.ipv4.conf.lan2.arp_ignore = 1
net.ipv4.conf.lan2.arp_notify = 0
net.ipv4.conf.lan2.drop_gratuitous_arp = 0
net.ipv4.conf.lan2.proxy_arp = 0
net.ipv4.conf.lan2.proxy_arp_pvlan = 0
net.ipv4.conf.lan3.arp_accept = 0
net.ipv4.conf.lan3.arp_announce = 0
net.ipv4.conf.lan3.arp_filter = 0
net.ipv4.conf.lan3.arp_ignore = 1
net.ipv4.conf.lan3.arp_notify = 0
net.ipv4.conf.lan3.drop_gratuitous_arp = 0
net.ipv4.conf.lan3.proxy_arp = 0
net.ipv4.conf.lan3.proxy_arp_pvlan = 0
net.ipv4.conf.lo.arp_accept = 0
net.ipv4.conf.lo.arp_announce = 0
net.ipv4.conf.lo.arp_filter = 0
net.ipv4.conf.lo.arp_ignore = 1
net.ipv4.conf.lo.arp_notify = 0
net.ipv4.conf.lo.drop_gratuitous_arp = 0
net.ipv4.conf.lo.proxy_arp = 0
net.ipv4.conf.lo.proxy_arp_pvlan = 0
net.ipv4.conf.miireg.arp_accept = 0
net.ipv4.conf.miireg.arp_announce = 0
net.ipv4.conf.miireg.arp_filter = 0
net.ipv4.conf.miireg.arp_ignore = 1
net.ipv4.conf.miireg.arp_notify = 0
net.ipv4.conf.miireg.drop_gratuitous_arp = 0
net.ipv4.conf.miireg.proxy_arp = 0
net.ipv4.conf.miireg.proxy_arp_pvlan = 0
net.ipv4.conf.phy1-ap0.arp_accept = 0
net.ipv4.conf.phy1-ap0.arp_announce = 0
net.ipv4.conf.phy1-ap0.arp_filter = 0
net.ipv4.conf.phy1-ap0.arp_ignore = 1
net.ipv4.conf.phy1-ap0.arp_notify = 0
net.ipv4.conf.phy1-ap0.drop_gratuitous_arp = 0
net.ipv4.conf.phy1-ap0.proxy_arp = 0
net.ipv4.conf.phy1-ap0.proxy_arp_pvlan = 0
net.ipv4.conf.phy2-ap0.arp_accept = 0
net.ipv4.conf.phy2-ap0.arp_announce = 0
net.ipv4.conf.phy2-ap0.arp_filter = 0
net.ipv4.conf.phy2-ap0.arp_ignore = 1
net.ipv4.conf.phy2-ap0.arp_notify = 0
net.ipv4.conf.phy2-ap0.drop_gratuitous_arp = 0
net.ipv4.conf.phy2-ap0.proxy_arp = 0
net.ipv4.conf.phy2-ap0.proxy_arp_pvlan = 0
net.ipv4.conf.wan.arp_accept = 0
net.ipv4.conf.wan.arp_announce = 0
net.ipv4.conf.wan.arp_filter = 0
net.ipv4.conf.wan.arp_ignore = 1
net.ipv4.conf.wan.arp_notify = 0
net.ipv4.conf.wan.drop_gratuitous_arp = 0
net.ipv4.conf.wan.proxy_arp = 0
net.ipv4.conf.wan.proxy_arp_pvlan = 0
net.ipv4.conf.wlan0.arp_accept = 0
net.ipv4.conf.wlan0.arp_announce = 0
net.ipv4.conf.wlan0.arp_filter = 0
net.ipv4.conf.wlan0.arp_ignore = 1
net.ipv4.conf.wlan0.arp_notify = 0
net.ipv4.conf.wlan0.drop_gratuitous_arp = 0
net.ipv4.conf.wlan0.proxy_arp = 0
net.ipv4.conf.wlan0.proxy_arp_pvlan = 0
net.netfilter.nf_log.3 = nf_log_arp`

mmichalko, 2024-02-28:
> Try to work around issue by passing sta to sta traffic over br-lan
> 
> ```
> option isolate '1'
> ```
> 
> https://openwrt.org/docs/guide-user/network/wifi/basic

Not sure what you mean with "passing sta to sta traffic over br-lan". What does sta mean? How to do this? How is the traffic passed now?

I tried setting option isolate to 0, it is visible also in my pasted wireless config, as it should explicitly disable isolation, but it did not help.

mmichalko, 2024-02-28:
@danpawlik 

the settings 
```
option multicast_to_unicast '0'
option isolate '0'
```
I set, because i read, that it helped other people, but if it is there or not, it behaved the same and the wlan clients are not reachable.

brada4, 2024-02-28:
STA means wifi client.

danpawlik, 2024-02-28:
Could you enable [debugging](https://openwrt.org/docs/guide-developer/debugging) (just to make clear info) and share it with us?
In your case would be something like:
```
uci set wireless.default_radio0.log_level=1
uci set wireless.default_radio1.log_level=1
uci commit wireless
```
Or just add:
```
config wifi-iface 'default_radio0'
(...)
    option log_level '1'
```

mmichalko, 2024-03-05:
I enabled debugging. Should i share tcpdump while trying to connect from other client? 
I run 
```
ssh root@192.168.1.1 'grep -q mon0 /proc/net/dev || /usr/sbin/iw phy phy0 interface add mon0 type monitor;
    /sbin/ifconfig mon0 up; /usr/sbin/iwcap -i mon0 -s' > /tmp/wifi.pcap
```
then tried to ssh from 2.4 GHz client to 5 GHz client, which did not work. Then i checked this output file `tmp/wifi.pcap`, but it is empty. What should i do?

danpawlik, 2024-03-09:
@mmichalko hey, sorry for delay. I don't have this device currently at home, so I can not bring any help here, but if you can do a test: just try to add:
```
net.ipv4.conf.br-lan.arp_ignore = 0
net.ipv4.conf.lan1.arp_ignore = 0
net.ipv4.conf.lan2.arp_ignore = 0
net.ipv4.conf.lan3.arp_ignore = 0
net.ipv4.conf.phy1-ap0.arp_ignore = 0
net.ipv4.conf.phy2-ap0.arp_ignore = 0
```
into: `/etc/sysctl.conf` and check if that helps?

mmichalko, 2024-03-10:
@danpawlik  added all the lines, restarted the router. It does not work, i cannot reach clients right after restart.

danpawlik, 2024-03-10:
@mmichalko did you try snapshot image if there it works?

sadontsev, 2024-03-13:
Probably won't bring much help here, as I'm fairly new to OpenWRT and haven't understood everything that is going on.
But I've got AX3600 as well and are experiencing the same issue. 
SSH goes down and may become reachable again either by:

- turning wifi on/off on a device (ie laptop if local server is unreachable from a laptop)
- rebooting a device

brada4, 2024-03-13:
check connection from router to device.
 then `logread -e aa:bb:cc:dd:ee:ff` ie search lost device's mac latest activity. 
If device is still connec%able from router you need to enable isolation.

danpawlik, 2024-03-21:
Is it not something similar as here https://github.com/openwrt/openwrt/issues/9555?

danpawlik, 2024-03-21:
@mmichalko do you have mac address randomization enabled on your device for different network or its set to device mac address?