openwrt/openwrt #14741: login message contains newlines

orhid, 2024-02-27:
### Describe the bug

upon logging into openwrt via ssh there is a message that looks like this

```

BusyBox v1.36.1 (2023-11-14 13:38:11 UTC) built-in shell (ash)

```

i have already verified that it does not come from the `/etc/banner` file, nor any `motd` file. is there any way to remove those newlines and — ideally — the message itself from printing at every login? 

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy WR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

login to the router via ssh

### Actual behaviour

a message appears that displays the `busy box ...` line with two newlines above and one below

### Expected behaviour

no message is printed, or at least the `busy box ...` line without newlines is printed

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

orhid, 2024-02-27:
a dirty fix is to install another shell from a different source

ynezz, 2024-02-29:
> so i guess one would have to recompile busybox to fix this "feature"

This or send a patch upstream.

closed 2024-02-29 by ynezz