openwrt/openwrt #14744: commit permissions

systemcrash, 2024-02-27:
### Describe the bug

Would like to help with review and commits to the openwrt repo also so I'd like to request commit access to this repo.

Active in the luci repo, but need to nudge things along in openwrt and packages also quite frequently. 

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

all

### Device

-

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2024-02-28:
This is not a bug, please ask on the ML.

systemcrash, 2024-02-28:
Thanks, I saw that vote - I was under the impression that this is for new members (who maybe get other rights) and not just commit permissions.

ynezz, 2024-02-29:
> Would like to help with review and commits to the openwrt repo

BTW anyone do that already, either add your `Reviewed-by:` or `Tested-by:` tags as a comment on the specific pull request or mailing list patch. It's truly important and greatly assists committers by helping to prioritize and motivate actions.

> and packages also quite frequently.

Yep, [that is a correct approach](https://github.com/openwrt/packages/issues/23552) for that repository.

> so I'd like to request commit access to this repo.

This repository is a bit different and [general project rules](https://openwrt.org/rules) applies here 

>  6. Frequent contributors may become committers after a simple majority agreement among existing committers. Project members are free to suggest suitable people.

closed 2024-02-29 by ynezz

systemcrash, 2024-02-29:
Understood. Thanks Petr.