openwrt/openwrt #14749: " ath10k_pci: Unknown symbol pcie_capability_clear_and_set_word" since mac80211 updated to 6.6.15

noblemtw, 2024-02-29:
### Describe the bug

Since commit fb45887e85ca42c95f80fb9a3147d98b3edc7724 (mac80211: update to 6.6.15) fails to load for ath79 target (Specifically Devolo dLAN pro 1200+ WiFi ac) with the below error:

[Sat Feb 24 09:02:50 2024] ath10k_pci: Unknown symbol pcie_capability_clear_and_set_word (err -2)
[Sat Feb 24 09:02:50 2024] kmodloader: 1 module could not be probed
[Sat Feb 24 09:02:50 2024] kmodloader: - ath10k_pci - 0

This is using CONFIG_PACKAGE_kmod-ath10k, rather than the -ct variant 

### OpenWrt version

r25345-c6319de48b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Devolo dLAN pro 1200+ WiFi ac

### Image kind

Self-built image

### Steps to reproduce

Simply build from master with below diffconfig. Powerline packages are included from https://github.com/garyttirn/dlan-openwrt, although this unlikely relates to the issue

### Actual behaviour

Module does not load with below issue, therefore 5Ghz radio is not available

powerline2:> modprobe -v ath10k_pci
1 module could not be probed
- ath10k_pci

### Expected behaviour

Module loads and 5Gz works

### Additional info

Reverting  fb45887e85ca42c95f80fb9a3147d98b3edc7724 resolves the issue and system works correctly using kmod-ath10k - 5.15.148+6.5-2 rather than kmod-ath10k - 5.15.148+6.6.15-1

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_devolo_dlan-pro-1200plus-ac=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH10K_LEDS=y
# CONFIG_ATH9K_UBNTHSR is not set
CONFIG_BINARY_FOLDER="/home/openwrt/source/openwrt-generic/bin"
# CONFIG_BINUTILS_USE_VERSION_2_40 is not set
CONFIG_BINUTILS_USE_VERSION_2_42=y
CONFIG_BINUTILS_VERSION="2.42"
CONFIG_BINUTILS_VERSION_2_42=y
# CONFIG_BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT is not set
CONFIG_BUSYBOX_CONFIG_BUNZIP2=y
CONFIG_BUSYBOX_CONFIG_BZCAT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_BZIP2_DECOMPRESS=y
[... 245 lines cut ...]
CONFIG_USE_LTO=y
# CONFIG_USE_SSTRIP is not set
CONFIG_USE_STRIP=y
CONFIG_WPA_MBO_SUPPORT=y
# CONFIG_BUSYBOX_CONFIG_PIE is not set
# CONFIG_NGINX_HTTP_CACHE is not set
# CONFIG_NGINX_HTTP_GEO is not set
# CONFIG_NGINX_HTTP_LIMIT_CONN is not set
# CONFIG_NGINX_HTTP_LIMIT_REQ is not set
# CONFIG_NGINX_HTTP_MEMCACHED is not set
# CONFIG_NGINX_HTTP_UPSTREAM_HASH is not set
# CONFIG_NGINX_HTTP_UPSTREAM_IP_HASH is not set
# CONFIG_OPENSSH_LIBFIDO2 is not set
# CONFIG_OPENSSL_WITH_ERROR_MESSAGES is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
# CONFIG_PACKAGE_libcbor is not set
# CONFIG_PACKAGE_libfido2 is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

noblemtw, 2024-03-02:
I think I've found the issue...

_include/kernel-build.mk_  creates a list of symbols (_mod_symtab.txt_) in all modules located lib/modules in the stagingdir root, which are used to create a list of included symbols, _sym_include.txt_ and _sym_exclude.txt_ - _pcie_capability_clear_and_set_word_ is in the exclude list in my build environment and therefore can't be loaded by the module (I've no idea how this works as it is listed in /proc/kallsyms)

The issue appears to have been caused by building both ath10k variants, backports and -ct, with the latter being built as a module. The logic in kernel-build.mk is picking up the -ct variant modules from the staging directory, which don't use _pcie_capability_clear_and_set_word_, hence it ultimately being excluded 

If I only build the backport drivers then everything works perfectly although I think that really this is a workaround rather than a fix, the issue is likely going to affect any builds where multiple module variants are being built when those modules don't all include the same symbols (This may just be an edge case for ath10k and ath10k-ct, but it's not going to be ath79 specific)

robimarko, 2024-03-02:
Well, I think this pretty much boils down to both regular ath10k and ath10k-ct installing the same `ath10k_core.ko` and `ath10k_pci.ko` modules but since `ath10k-ct` is built after the mac80211/backports package it basically writes over the regular ath10k module and then since that symbol is not yet used in ath10k-ct (Since its still 6.4 based) it gets stripped

noblemtw, 2024-03-03:
That solves the issue, backports ath10k_pci.ko now loads even with the -ct variants selected as a module.