openwrt/openwrt #14756: Invalid values for hour can be set and saved in start / stop time of firewall rule

JustAnotherEndUser, 2024-02-29:
### Describe the bug

While editing a firewall rule I made a typo in the Start / Stop time area.  It allowed for an invalid hour entry to be entered and saved without generating an error.
[image]

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Steps to reproduce

I edited a different rule and verified it was reproducible.  I entered multiple different invalid values for the 'Hour' which were accepted when entered and saved.  Example of saved ( but not applied ) entry:
[image]

Through entry of multiple values I discovered that values 24-69 could be entered in the 'Hour' section of time entry, but values 70 or above would generate the expected error notification shown below.
[image]

### Actual behaviour

The firewall rules start / stop time area allows for entry and rule save despite containing invalid values for hour.

### Expected behaviour

I would expect the valid hours entry to be 0:00:00 through 23:59:59.  The time box allows the hour to be set (and saved) above 23: ( 24 -69 )  without error.

### Additional info

I saved but did not apply the rule changes.  I don't know what the outcome would be if a typo like this were actually mistakenly saved, beyond possibly the rule just not working. Even if minor, I just wanted to report this. Apologies if this is duplicate or already reported. I searched but didn't find any similar post. Thanks and regards.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-02-29:
You need to report it in ../luci

Rule will be rejected by fw4 as it never was there:
```
fw4 check
Section @rule[58] option 'start_time' specifies invalid value '25:61:61'
Section @rule[58] skipped due to invalid options
```
Will not get to nft -c failure that would keep firewall from restarting or starting at boot:
```
stdin:5:13-22: Error: wrong time format
                meta hour "25:61:61"
```

dannil, 2024-03-02:
I've created a fix at https://github.com/openwrt/luci/pull/6953 which solves this.