openwrt/openwrt #14758: umdns does not find any other devices until restart

NilsRo, 2024-02-29:
### Describe the bug

There is an issue with umdns that it does not find any devices until restart. There is no error message in the logs.
https://github.com/berlin-open-wireless-lab/DAWN/issues/237

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq806x/generic

### Device

ZyXEL NBG6817

### Image kind

Official downloaded image

### Steps to reproduce

Install dawn with umbus as dependency. After a while umdns will stops working...

### Actual behaviour

"ubus call umdns browse" will not show any devices until restart umdns

### Expected behaviour

It should keep running and should show all devices

### Additional info

device and target Independent, happens on all my OpenWRT devices like WAX206.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

edrikk, 2024-03-08:
I can confirm the same behaviour.  To be specific umdns restart is needed so the it provides info that DAWN is requesting (eg other APs).

HaraldNoack, 2024-03-13:
Hello,

I have the same problem with:
D-Link DAP-X1860 A1        OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-24.067.02086-146a8f1
TP-Link Archer C7 v5          OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-24.067.02086-146a8f1

I restart the devices every 6 hours by cron. 

How can i restart umdns by cron every 6 hours?

Thanks!

BigMaxTs, 2024-03-13:
> Hello,
>
> I have the same problem with: D-Link DAP-X1860 A1 OpenWrt 23.05.0 r23497-6637af95aa / LuCI openwrt-23.05 branch git-24.067.02086-146a8f1 TP-Link Archer C7 v5 OpenWrt 23.05.2 [...]

You basically do not need to restart the whole box, add this to cron (there is a menu for this in LuCi) and do a restart of umdns only. This workaround works for me for a week or so no issues - this restarts the service every 2 hours.

0 */2 * * * /etc/init.d/umdns restart

NilsRo, 2024-03-13:
Alternativ workaround for umdns should be to switch to broadcasts instead bonjour which is deprecated but still supported.

christf, 2024-03-13:
> I can confirm the same behaviour. To be specific umdns restart is needed so the it provides info that DAWN is requesting (eg other APs).

same here.

jonnycastaway, 2024-09-29:
Same here,
installed 3 APs with Openwrt 23.05.05 umdns and dawn. After every reboot and after some time radomly one or more APs have no entries in the umdns hosts list. I hve to restart the service and all is fine.
Using the workaround with 0 */2 * * * /etc/init.d/umdns restart in cron is fine but not the answer.

abrender, 2024-10-01:
Same here... after some undefined period of time umdns (2024.01.08~e91ed406-r1) stops responding to requests and only restarting the umdns service brings it back.   Setting up the cron job to restart the service is a bandaid that will probably work, but I'm hopeful that someone can actually root cause and fix the underlying issue.

christf, 2024-10-08:
Today I found this:

When umdns is not working, netstat shows no umdns process is listening on any port.

After /etc/init.d/umdns restart, netstat shows
```
udp        0      0 0.0.0.0:56551           0.0.0.0:*                           2747/umdns
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           2747/umdns
udp        0      0 :::56800                :::*                                2747/umdns
udp        0      0 :::5353                 :::*                                2747/umdns
```

After a while, dawn will recover.
So now we can run a watchdog that checks if umdns is listening on the ports as expected and if it isn't, restart it.

I was afraid to share this because of
[image]. After checking that the latest openwrt release is  using the latest umdns as per https://github.com/openwrt/mdnsd.

I am attempting to reproduce  this problem without ujail now.

bjornfor, 2024-11-29:
> Edit: I was not able to reproduce this after upgrading to openwrt 23.05.5

Does that mean this issue can be closed?

mattaw, 2024-12-10:
I just had this on 23.05.5 on two Netgear WAX218's, `ubus call umdns update && ubus call umdns browse` returned `{ }` on both until they had been restarted. I reported it on openwrt/mdnsd#16 against the openwrt component in the hopes of more visibility.

hexoxssaa, 2025-01-14:
> Today I found this:
> 
> When umdns is not working, netstat shows no umdns process is listening on any port.
> 
> After /etc/init.d/umdns restart, netstat shows
> 
> ```
> udp        0      0 0.0.0.0:56551           0.0.0.0:*                           2747/umdns
> udp        0      0 0.0.0.0:5353            0.0.0.0:*                           2747/umdns
> udp        0      0 :::56800                :::*                                2747/umdns
> udp        0      0 :::5353                 :::*                                2747/umdns
> ```
> 
> After a while, dawn will recover. So now we can run a watchdog that checks if umdns is listening on the ports as expected and if it isn't, restart it.
> 
> I was afraid to share this because of ![image](https://private-user-images.githubusercontent.com/4321652/374683175-fdbc9c87-1f5d-47d3-811d-49b96b07b110.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MTQ1MTksIm5iZiI6MTczNjgxNDIxOSwicGF0aCI6Ii80MzIxNjUyLzM3NDY4MzE3NS1mZGJjOWM4Ny0xZjVkLTQ3ZDMtODExZC00OWI5NmIwN2IxMTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDExNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMTRUMDAyMzM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTdjMmYyZjhkY2IyNDMzZWRmZTcxNGZhZTQ3YjM1ODU4ZmNiMDIxMTRhZTAxMDA1ZjUzMzMwNzQwNDJhNDdkZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.JodbpnCYt2nxmFbkG7LyN0j6TE8aquWRORw594emqlU). After checking that the latest openwrt release is using the latest umdns as per https://github.com/openwrt/mdnsd.
> 
> I am attempting to reproduce this problem without ujail now.
> 
> Edit: I was not able to reproduce this after upgrading to openwrt 23.05.5

same problem on my openwrt 23.05.5

hexoxssaa, 2025-01-14:
I found the solution from this link: [https://openwrt.org/docs/guide-user/network/wifi/dawn](https://openwrt.org/docs/guide-user/network/wifi/dawn)
mv /etc/seccomp/umdns.json /etc/seccomp/umdns.json.disable

just disable seccomp, then eveything works fine!

teeedubb, 2025-02-13:
I see that umdns has received an update in 24.10:

2024.09.17~fbaca4b6-r1 » 2025.02.10~c5ca22a7-r1

I will upgrade on both routers and report back if this fixes DAWN of 24.10

nbd168, 2025-03-13:
I pushed a fix to procd and serveral init scripts (including the umdns one), which handles ubus config apply after service crash properly now.

wryun, 2025-03-13:
@nbd168 , while you're looking at how procd starts these services, there's also a potentially related umdns issue where config file changes are not correctly applied because reload_service is overridden in the umdns procd init script. This mean it doesn't:
- re-evaluate the network interface triggers
- change the jail status

Not re-evaluating the network interface triggers means that any new interface that's added doesn't work correctly.

Unfortunately, the naive fix of renaming reload_service such that the 'normal' reload will run - which seems strictly better, as then procd will call service_started regardless (at least before your changes!) to do the set_config - does not seem to work correctly, at least on OpenWrt 23, because if the process is not restarted the triggers don't seem to work (the triggers appear in the service definitions accessible via ubus, but don't seem to take effect).

My apologies for not raising this as a separate bug yet; I haven't got around to replicating on the latest release.

nbd168, 2025-03-13:
@wryun would something like this help? https://nbd.name/p/07ddc4be

wryun, 2025-03-13:
> [@wryun](https://github.com/wryun) would something like this help? https://nbd.name/p/07ddc4be

Yes, we've trying to doing something like that, but the problem is that it doesn't establish the correct network interface triggers (at least in our experience with 23.05). Or rather, they appear to be set, but they don't work, so when you add a new network interface it doesn't react to changes on that interface.

nbd168, 2025-03-14:
Can you test if they work with those changes in 24.10? Maybe there is a bug in the older version of procd that got fixed along the way.

wryun, 2025-03-16:
Yes, I'll try to investigate if it's happening on 24.10 and raise if so. Thanks for fixing up the reload behaviour on master!