openwrt/openwrt #14759: WAN/Internet access not working after random times ASUS  RT-AX59U

tkso1997, 2024-03-01:
### Describe the bug

After several hours the connection to the attached modem is lost. Rebooting the router is necessary to get it working again, "/etc/init.d/network restart" is not sufficient.

> [25594.961784] ------------[ cut here ]------------
[25594.966412] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 4 timed out
[25594.973399] WARNING: CPU: 1 PID: 3609 at dev_watchdog+0x238/0x240
[25594.979483] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c compat crypto_safexcel cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ifb autofs4 sha1_generic seqiv md5 des_generic libdes authencesn authenc uas
[25594.979657]  usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common aquantia
[25595.077523] CPU: 1 PID: 3609 Comm: homebridge Not tainted 6.1.79 #0
[25595.083771] Hardware name: ASUS RT-AX59U (DT)
[25595.088110] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[25595.095050] pc : dev_watchdog+0x238/0x240
[25595.099044] lr : dev_watchdog+0x238/0x240
[25595.103036] sp : ffffffc00800be40
[25595.106334] x29: ffffffc00800be40 x28: ffffffc008b769c0 x27: ffffffc00800bf10
[25595.113450] x26: 00000000000000e0 x25: 0000000000000000 x24: dead000000000122
[25595.120567] x23: 0000000000000000 x22: ffffffc008b76000 x21: 0000000000000004
[25595.127683] x20: ffffff800080d000 x19: ffffff800080d4c0 x18: 0000000000000143
[25595.134800] x17: ffffffc01733b000 x16: ffffffc008008000 x15: ffffffc008b89ac0
[25595.141918] x14: 00000000000003c9 x13: 0000000000000143 x12: 00000000ffffffea
[25595.149037] x11: 00000000ffffefff x10: ffffffc008be1ac0 x9 : ffffffc008b89a68
[25595.156155] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
[25595.163272] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[25595.170389] x2 : ffffff801fe9e1d0 x1 : ffffffc01733b000 x0 : 000000000000003f
[25595.177508] Call trace:
[25595.179943]  dev_watchdog+0x238/0x240
[25595.183594]  call_timer_fn.constprop.0+0x20/0x80
[25595.188203]  __run_timers.part.0+0x208/0x284
[25595.192457]  run_timer_softirq+0x38/0x70
[25595.196363]  _stext+0x10c/0x278
[25595.199489]  ____do_softirq+0xc/0x14
[25595.203051]  call_on_irq_stack+0x24/0x40
[25595.206957]  do_softirq_own_stack+0x18/0x2c
[25595.211124]  __irq_exit_rcu+0xcc/0xd4
[25595.214775]  irq_exit_rcu+0xc/0x14
[25595.218163]  el1_interrupt+0x34/0x50
[25595.221725]  el1h_64_irq_handler+0x14/0x20
[25595.225806]  el1h_64_irq+0x68/0x6c
[25595.229191]  lzma_main+0x1ec/0x910
[25595.232582]  xz_dec_lzma2_run+0x6c0/0x880
[25595.236576]  xz_dec_run+0x190/0x904
[25595.240050]  squashfs_xz_uncompress+0xd4/0x210
[25595.244478]  squashfs_decompress+0x60/0xa4
[25595.248558]  squashfs_read_data+0xb8/0x450
[25595.252642]  squashfs_readahead+0x3f8/0x5b0
[25595.256809]  read_pages+0x50/0x24c
[25595.260197]  page_cache_ra_unbounded+0x114/0x1b4
[25595.264797]  page_cache_ra_order+0x40/0x50
[25595.268877]  filemap_fault+0x430/0x780
[25595.272613]  __do_fault+0x38/0x100
[25595.276003]  __handle_mm_fault+0x3ec/0xa24
[25595.280085]  handle_mm_fault+0xc0/0x240
[25595.283905]  do_page_fault+0x13c/0x310
[25595.287641]  do_translation_fault+0x80/0x90
[25595.291808]  do_mem_abort+0x40/0x90
[25595.295281]  el0_ia+0x44/0xb0
[25595.298235]  el0t_64_sync_handler+0xbc/0x120
[25595.302490]  el0t_64_sync+0x14c/0x150
[25595.306136] ---[ end trace 0000000000000000 ]---
[25595.310764] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[25595.497746] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[25596.031757] mtk_soc_eth 15100000.ethernet: warm reset failed
[25596.050697] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[25596.059091] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[75853.448294] mt7530-mdio mdio-bus:1f wan: Link is Down
[75856.577756] mt7530-mdio mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[76727.650416] mt7530-mdio mdio-bus:1f wan: Link is Down
[76731.810136] mt7530-mdio mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off

### OpenWrt version

r25345-c6319de48b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS RT-AX59U

### Image kind

Self-built image

### Steps to reproduce

Wait till it happens.

### Actual behaviour

After several hours internet access via WAN is not working any longer.

### Expected behaviour

Working internet access.

### Additional info

Attached is a Docsis Cable Modem which works with other OpenWRT routers including this one (with kernel 5.15)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-01:
Please format dmesg with three backticks \`\`\`

do you by chance have firewall offload enabled? disable it.
```
And from /etc/config add network and firewall, deleting passwords/usernames/public ips from the files. 
```
Are last 2 message lines from your attemt to recover? Did ut work? What you did?

tkso1997, 2024-03-01:
> Please format dmesg with three backticks ```

done!

> do you by chance have firewall offload enabled? disable it.

firewall offload is and was deactivated

> ```
> And from /etc/config add network and firewall, deleting passwords/usernames/public ips from the files. 
> ```

```
root@OpenWrt:~# cat /etc/config/network 

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'xxxx:xxxx:xxxx::/xx'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.31.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option device 'wan'
	option proto 'dhcp'

config interface 'wan6'
	option device 'wan'
	option proto 'dhcpv6'
```

```
root@OpenWrt:~# cat /etc/config/firewall 

config defaults
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option synflood_protect '1'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'
```
> Are last 2 message lines from your attemt to recover? Did ut work? What you did?

I guess the lines are the result of "/etc/init.d/network restart" but that did not bring the WAN up again. Only solution is rebooting the router.

brada4, 2024-03-01:
You see eth0 also restarted? Do you know if LAN (the ethernet , not wifi) was usable?
Nothing wrong with config shown. Backtrace is similar to firewall offload on slow platforms, though seems loosely co-related only.

tkso1997, 2024-03-02:
> You see eth0 also restarted? Do you know if LAN (the ethernet , not wifi) was usable? Nothing wrong with config shown. Backtrace is similar to firewall offload on slow platforms, though seems loosely co-related only.

Only wifi was usable, ethernet was completely broken.
I am trying again and report back.  Any suggestions what I should look for/could try out to provide more helpful informations?

brada4, 2024-03-02:
Thanks for info. And sorry for jumping in wrong pond.

brada4, 2024-03-02:
Yoy can try disabling pause/flow control on network ports, that was another suspect in stalling adapter:
target/linux/mediatek/dts/mt7986a-asus-rt-ax59u.dts

tkso1997, 2024-03-03:
The problem did not occur agin after switching flow control off by using "ethtool -K eth0 rx off tx off". Should we report this somewhere else or just modify the mt7986a-asus-rt-ax59u.dts and accept that flow control is making trouble?

brada4, 2024-03-03:
Does not work that way. Imagebuilder deos not integrate DTS changed in kernel, you need to rebuild the kernel.
`-k tx off rx off` disabled checksumming offloads and likely all other network acceleration in card along. 
If pause disablement is supported it is via `-A/-a tx off rx off auto on`

tkso1997, 2024-03-05:
Issue seems to be gone when using linux 6.6. Overall network seems to be much snappier with the 6.6 kernel.

palmdm, 2026-09-23:
Hi everybody. As I understand it, the problem has been fixed for 2026, at least on OpenWRT 25 everything should be stable. We were just trying to raise it, because I'm going to buy a router at home.

brada4, 2026-09-23:
You understand too much. rx-gro-list is still default.