openwrt/openwrt #14760: Offlining a core on ipq806x you are unable to bring the core back online

GXTX, 2024-03-01:
### Describe the bug

After offlining a core on ipq806x you are unable to bring the core back online.

### OpenWrt version

r23711+66-cdc907abed

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

R7800

### Image kind

Self-built image

### Steps to reproduce

echo 0 > /sys/devices/system/cpu/cpu1/online
echo 1 > /sys/devices/system/cpu/cpu1/online

### Actual behaviour

```
root@R7800:~# echo 0 > /sys/devices/system/cpu/cpu1/online <-- successfully takes core offline
root@R7800:~# echo 1 > /sys/devices/system/cpu/cpu1/online
ash: write error: I/O error
```

dmesg:
```[261430.588864] CPU1: failed to come online```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-01:
Please add diffconfig and describe +66 patches you have in source tree.

robimarko, 2024-03-01:
Can I ask why are you even trying to do this?

GXTX, 2024-03-02:
> Can I ask why are you even trying to do this?

It's a feature of Linux (and Windows for that matter), and was just playing around with it. Ideally a Linux based OS should support it. It could also have some real world uses for low power requirements -- by disabling cores during known hours of inactivity.

> describe +66 patches you have in source tree

Enabled NSS cores using community patches from forum.

brada4, 2024-03-02:
None of your postings explain how non-default cpu hotplug control option suddenly enabled itself.

GXTX, 2024-03-02:
Thanks!

Made me go and check my colo'ed box, and indeed there appears to be a kernel config for it.
```
CONFIG_HOTPLUG_CPU=y
```

I suppose I was so used to distros just having this enabled by default.
Will close and might report back if I decide to recompile with that enabled.

closed 2024-03-02 by GXTX