openwrt/openwrt #14767: bpi-r4 doesn't recognize all PCI devices

mmisiewicz, 2024-03-02:
### Describe the bug

Compared to the [Banana pi-provided image](https://wiki.banana-pi.org/Banana_Pi_BPI-R4#Release_image) which is based on kernel 5.4.246, the most recent snapshot of openwrt fails to detect all the PCI devices on. I'm specifically trying to get a T99W175 b-key modem working. On the sinovoip provided build, the modem is detected, but not on snapshot openwrt. 

There are some suggestive errors in the kernel log:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd090]
[    0.000000] Linux version 6.1.79 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r25361-a86ff06d2e) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Fri Mar  1 01:42:36 2024
[    0.000000] Machine model: Bananapi BPI-R4
...
[    0.042510] mtk-xsphy soc:xphy@11e10000: failed to get ref_clk(id-1)
[    0.043034] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    0.043054] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    0.043065] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    0.043078] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    0.043174] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    0.043185] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    0.043194] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    0.043204] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    0.043223] /soc/pcie@11290000: Failed to get clk index: 0 ret: -517
[    0.043232] mtk-pcie-gen3 11290000.pcie: failed to get clocks
[    0.043294] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    0.043304] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    0.043313] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    0.043322] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    0.043337] /soc/pcie@11300000: Failed to get clk index: 0 ret: -517
[    0.043345] mtk-pcie-gen3 11300000.pcie: failed to get clocks
[    0.043418] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    0.043428] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    0.043436] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    0.043445] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    0.043461] /soc/pcie@11310000: Failed to get clk index: 0 ret: -517
...
[    2.264470] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    2.271809] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    2.278074] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    2.285724] mtk-msdc 11230000.mmc: Final PAD_DS_TUNE: 0x13014
[    2.286502] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
...
[    2.637311] mtk-pcie-gen3 11280000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x0)
[    2.646106] mtk-pcie-gen3: probe of 11280000.pcie failed with error -110
[    2.653141] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    2.660455] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    2.666719] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    2.675154] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    3.017310] mtk-pcie-gen3 11290000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.026086] mtk-pcie-gen3: probe of 11290000.pcie failed with error -110
[    3.032937] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    3.040244] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    3.046503] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    3.054938] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    3.397313] mtk-pcie-gen3 11300000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.406088] mtk-pcie-gen3: probe of 11300000.pcie failed with error -110
[    3.412945] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    3.420253] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    3.426511] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    3.434942] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    3.707436] mtk-pcie-gen3 11310000.pcie: set IO trans window[0]: cpu_addr = 0x38000000, pci_addr = 0x38000000, size = 0x200000
[    3.718816] mtk-pcie-gen3 11310000.pcie: set MEM trans window[1]: cpu_addr = 0x38200000, pci_addr = 0x38200000, size = 0x200000
[    3.730281] mtk-pcie-gen3 11310000.pcie: set MEM trans window[2]: cpu_addr = 0x38400000, pci_addr = 0x38400000, size = 0x400000
[    3.741743] mtk-pcie-gen3 11310000.pcie: set MEM trans window[3]: cpu_addr = 0x38800000, pci_addr = 0x38800000, size = 0x800000
[    3.753206] mtk-pcie-gen3 11310000.pcie: set MEM trans window[4]: cpu_addr = 0x39000000, pci_addr = 0x39000000, size = 0x1000000
[    3.764754] mtk-pcie-gen3 11310000.pcie: set MEM trans window[5]: cpu_addr = 0x3a000000, pci_addr = 0x3a000000, size = 0x2000000
[    3.776303] mtk-pcie-gen3 11310000.pcie: set MEM trans window[6]: cpu_addr = 0x3c000000, pci_addr = 0x3c000000, size = 0x4000000
[    3.787965] mtk-pcie-gen3 11310000.pcie: PCI host bridge to bus 0001:00
[    3.794568] pci_bus 0001:00: root bus resource [bus 00-ff]
[    3.800048] pci_bus 0001:00: root bus resource [io  0x600000-0x7fffff] (bus address [0x38000000-0x381fffff])
[    3.809863] pci_bus 0001:00: root bus resource [mem 0x38200000-0x3fffffff]
[    3.816724] pci_bus 0001:00: scanning bus
[... 116 lines cut ...]
mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

Put a PCIe card in the b key M2 slot, boot up. 

### Actual behaviour

Card not detected

### Expected behaviour

Card is detected (and then drivers load) 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mmisiewicz, 2024-03-02:
full boot log attached 
[r25361-a86ff06d2e boot.txt](https://github.com/openwrt/openwrt/files/14467454/r25361-a86ff06d2e.boot.txt)

mmisiewicz, 2024-03-02:
full boot log from bpi stock 
[bpi-stock.txt](https://github.com/openwrt/openwrt/files/14467456/bpi-stock.txt)

mmisiewicz, 2024-03-02:
Reading the lspci output between the bpi and snapshot images, it seems that one of the pci controllers is not being detected under 6.1 kernel. Specifically this one:

```
0003:00:00.0 PCI bridge: MEDIATEK Corp. Device 7988 (rev 01) (prog-if 00 [Normal decode])
```

[lspci snapshot.txt](https://github.com/openwrt/openwrt/files/14467802/lspci.snapshot.txt)
[lspci bpi image.txt](https://github.com/openwrt/openwrt/files/14467803/lspci.bpi.image.txt)

mmisiewicz, 2024-03-02:
Using the command `dtc -s -I fs /proc/device-tree -O dts` I have diffed the device trees between the sinovoip image and the snapshot device tree. Looking in the PCI section there are some discrepancies. On the sinovoip:
```
	pcie@11280000 {
		#address-cells = <0x03>;
		#interrupt-cells = <0x01>;
		#size-cells = <0x02>;
		bus-range = <0x00 0xff>;
		clock-names = "pl_250m\0tl_26m\0peri_26m\0top_133m";
		clocks = <0x0d 0x59 0x0d 0x55 0x0d 0x61 0x0d 0x5d>;
		compatible = "mediatek,mt7988-pcie\0mediatek,mt7986-pcie";
		device_type = "pci";
		interrupt-map = <0x00 0x00 0x00 0x01 0x25 0x00 0x00 0x00 0x00 0x02 0x25 0x01 0x00 0x00 0x00 0x03 0x25 0x02 0x00 0x00 0x00 0x04 0x25 0x03>;
		interrupt-map-mask = <0x00 0x00 0x00 0x07>;
		interrupts = <0x00 0xaa 0x04>;
		linux,pci-domain = <0x03>;
		phy-names = "pcie-phy";
		phys = <0x24 0x02>;
		pinctrl-0 = <0x26>;
		pinctrl-names = "default";
		ranges = <0x81000000 0x00 0x20000000 0x00 0x20000000 0x00 0x200000 0x82000000 0x00 0x20200000 0x00 0x20200000 0x00 0x7e00000>;
		reg = <0x00 0x11280000 0x00 0x2000>;
		reg-names = "pcie-mac";
		status = "okay";

		interrupt-controller {
			#address-cells = <0x00>;
			#interrupt-cells = <0x01>;
			interrupt-controller;
			phandle = <0x25>;
		};
	};
```

On Snapshot:
```
pcie@11280000 {
			#address-cells = <0x03>;
			#interrupt-cells = <0x01>;
			#size-cells = <0x02>;
			bus-range = <0x00 0xff>;
			clock-names = "pl_250m\0tl_26m\0peri_26m\0top_133m";
			clocks = <0x09 0x5d 0x09 0x59 0x09 0x15 0x09 0x61>;
			compatible = "mediatek,mt7988-pcie\0mediatek,mt7986-pcie\0mediatek,mt8192-pcie";
			device_type = "pci";
			interrupt-map = <0x00 0x00 0x00 0x01 0x1e 0x00 0x00 0x00 0x00 0x02 0x1e 0x01 0x00 0x00 0x00 0x03 0x1e 0x02 0x00 0x00 0x00 0x04 0x1e 0x03>;
			interrupt-map-mask = <0x00 0x00 0x00 0x07>;
			interrupts = <0x00 0xaa 0x04>;
			linux,pci-domain = <0x03>;
			phandle = <0x7c>;
			phy-names = "pcie-phy";
			phys = <0x19 0x02>;
			pinctrl-0 = <0x1d>;
			pinctrl-names = "default";
			ranges = <0x81000000 0x00 0x20000000 0x00 0x20000000 0x00 0x200000 0x82000000 0x00 0x20200000 0x00 0x20200000 0x00 0x7e00000>;
			reg = <0x00 0x11280000 0x00 0x2000>;
			reg-names = "pcie-mac";
			status = "okay";

			interrupt-controller {
				#address-cells = <0x00>;
				#interrupt-cells = <0x01>;
				interrupt-controller;
				phandle = <0x1e>;
			};
		};
```

Some of the lines that changed seem like they might be the problem. On the sinovoip image:
```
		clock-names = "pl_250m\0tl_26m\0peri_26m\0top_133m";
		clocks = <0x0d 0x59 0x0d 0x55 0x0d 0x61 0x0d 0x5d>;
		compatible = "mediatek,mt7988-pcie\0mediatek,mt7986-pcie";
...
		interrupt-map = <0x00 0x00 0x00 0x01 0x25 0x00 0x00 0x00 0x00 0x02 0x25 0x01 0x00 0x00 0x00 0x03 0x25 0x02 0x00 0x00 0x00 0x04 0x25 0x03>;
...
		phys = <0x24 0x02>;
		pinctrl-0 = <0x26>;
...
			phandle = <0x25>;

```

these change to the following on snapshot:
``` 
			clocks = <0x09 0x5d 0x09 0x59 0x09 0x15 0x09 0x61>;
			compatible = "mediatek,mt7988-pcie\0mediatek,mt7986-pcie\0mediatek,mt8192-pcie";
...
			interrupt-map = <0x00 0x00 0x00 0x01 0x1e 0x00 0x00 0x00 0x00 0x02 0x1e 0x01 0x00 0x00 0x00 0x03 0x1e 0x02 0x00 0x00 0x00 0x04 0x1e 0x03>;
...
			phys = <0x19 0x02>;
			pinctrl-0 = <0x1d>;
...
				phandle = <0x1e>;

```

So the clocks seem different on the latest snapshot. That might explain the kernel log errors and why the card is not being detected.

renanqts, 2024-05-03:
Seems like it is the same issue as mine.

https://github.com/openwrt/openwrt/pull/14140#issuecomment-2093691202

frank-w, 2024-05-05:
if it helps in my repo all pci work for me (basicly same code daniel has in his WIP tree)

https://github.com/frank-w/BPI-Router-Linux/blob/6.9-rc/arch/arm64/boot/dts/mediatek/mt7988a.dtsi#L891

https://github.com/frank-w/BPI-Router-Linux/blob/6.9-rc/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4.dts

but maybe driver is different in 6.1

renanqts, 2024-05-08:
> if it helps in my repo all pci work for me (basicly same code daniel has in his WIP tree)
>
> https://github.com/frank-w/BPI-Router-Linux/blob/6.9-rc/arch/arm64/boot/dts/mediatek/mt7988a.dtsi#L891 [...]

it does not seems to be for R4, did you try it out?

renanqts, 2024-05-14:
@frank-w did you try in which kernel version?

frank-w, 2024-05-14:
At least with 6.9...the version i linked. I have the wifi7 card in both mpci slots,an nvme in key-m and one in keyb...all 4 cards were recognized

renanqts, 2024-05-14:
> At least with 6.9...the version i linked. I have the wifi7 card in both mpci slots,an nvme in key-m and one in keyb...all 4 cards were recognized

that is super weird, I also built 6.9 yesterday, same error

frank-w, 2024-05-14:
Which cards have you inserted?

renanqts, 2024-05-14:
> Which cards have you inserted?

I tried with `Qualcomm Atheros AR242x / AR542x Wireless Network Adapter` and `MEDIATEK Corp. MT7921 802.11ax PCI Express Wireless Network Adapter`, same error in `MEDIATEK Corp. Device 7988` for both

```
0001:00:00.0 PCI bridge: MEDIATEK Corp. Device 7988 (rev 01) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 113
	Memory at 38400000 (64-bit, non-prefetchable) [size=32K]
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: [disabled] [32-bit]
	Memory behind bridge: [disabled] [32-bit]
	Prefetchable memory behind bridge: 38200000-383fffff [size=2M] [32-bit]
	Capabilities: [80] Express Root Port (Slot-), IntMsgNum 0
	Capabilities: [e0] MSI: Enable+ Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting
	Capabilities: [300] Secondary PCI Express
	Kernel driver in use: pcieport
lspci: Unable to load libkmod resources: error -2
```

frank-w, 2024-05-15:
Bpi boards seem to be incompatible with atheros cards...there are several discussions in bpi-forum.

Not sure if r64+ all missing tx-capacitors,resistors for power or need additional reset at initialization.

renanqts, 2024-05-15:
> Bpi boards seem to be incompatible with atheros cards...there are several discussions in bpi-forum.
> 
> Not sure if r64+ all missing tx-capacitors,resistors for power or need additional reset at initialization.

But there is one from MEDIATEK, it does not work either.

renanqts, 2024-05-15:
JFY, I got my `MEDIATEK Corp. MT7921` working.

P-Light, 2024-05-30:
Have the same problem on bpi-r4 with mt7922, openwrt snapshot:
```
...
[    0.031289] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.049667] jitterentropy: Initialization failed with host not compliant with requirements: 9
[    0.050602] mtk-xsphy soc:xphy@11e10000: failed to get ref_clk(id-1)
[    0.051129] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    0.051148] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    0.051160] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    0.051173] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    0.051273] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    0.051284] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    0.051293] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    0.051303] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    0.051327] /soc/pcie@11290000: Failed to get clk index: 0 ret: -517
[    0.051335] mtk-pcie-gen3 11290000.pcie: failed to get clocks
[    0.051398] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    0.051408] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    0.051417] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    0.051426] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    0.051442] /soc/pcie@11300000: Failed to get clk index: 0 ret: -517
[    0.051449] mtk-pcie-gen3 11300000.pcie: failed to get clocks
[    0.051521] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    0.051531] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    0.051540] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    0.051548] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    0.051563] /soc/pcie@11310000: Failed to get clk index: 0 ret: -517
[    0.051570] mtk-pcie-gen3 11310000.pcie: failed to get clocks
[    0.055961] Serial: 8250/16550 driver, 3 ports, IRQ sharing disabled
...
[    2.397085]  mmcblk0: p1 p2 p3 p4 p5 p6 p7
[    2.661624] mtk-pcie-gen3 11280000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    2.670428] mtk-pcie-gen3: probe of 11280000.pcie failed with error -110
[    2.677460] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    2.684774] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    2.691039] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    2.699473] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    3.041625] mtk-pcie-gen3 11290000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.050407] mtk-pcie-gen3: probe of 11290000.pcie failed with error -110
[    3.057274] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
...
[    3.663847] mtk-pcie-gen3 11300000.pcie: msi#0x0 address_hi 0x0 address_lo 0x11300c00 data 0
[    3.672285] mtk-pcie-gen3 11300000.pcie: msi#0x1 address_hi 0x0 address_lo 0x11300c00 data 1
[    3.680710] mtk-pcie-gen3 11300000.pcie: msi#0x2 address_hi 0x0 address_lo 0x11300c00 data 2
[    3.689138] mtk-pcie-gen3 11300000.pcie: msi#0x3 address_hi 0x0 address_lo 0x11300c00 data 3
[    3.697565] mtk-pcie-gen3 11300000.pcie: msi#0x4 address_hi 0x0 address_lo 0x11300c00 data 4
[    3.705991] mtk-pcie-gen3 11300000.pcie: msi#0x5 address_hi 0x0 address_lo 0x11300c00 data 5
[    3.714418] mtk-pcie-gen3 11300000.pcie: msi#0x6 address_hi 0x0 address_lo 0x11300c00 data 6
[    3.722843] mtk-pcie-gen3 11300000.pcie: msi#0x7 address_hi 0x0 address_lo 0x11300c00 data 7
[    3.731267] mtk-pcie-gen3 11300000.pcie: msi#0x8 address_hi 0x0 address_lo 0x11300c00 data 8
[    3.739694] mtk-pcie-gen3 11300000.pcie: msi#0x9 address_hi 0x0 address_lo 0x11300c00 data 9
[    3.748120] mtk-pcie-gen3 11300000.pcie: msi#0xa address_hi 0x0 address_lo 0x11300c00 data 10
[    3.756634] mtk-pcie-gen3 11300000.pcie: msi#0xb address_hi 0x0 address_lo 0x11300c00 data 11
[    3.765147] mtk-pcie-gen3 11300000.pcie: msi#0xc address_hi 0x0 address_lo 0x11300c00 data 12
[    3.773660] mtk-pcie-gen3 11300000.pcie: msi#0xd address_hi 0x0 address_lo 0x11300c00 data 13
[    3.782172] mtk-pcie-gen3 11300000.pcie: msi#0xe address_hi 0x0 address_lo 0x11300c00 data 14
[    3.790682] mtk-pcie-gen3 11300000.pcie: msi#0xf address_hi 0x0 address_lo 0x11300c00 data 15
[    3.799198] mtk-pcie-gen3 11300000.pcie: msi#0x10 address_hi 0x0 address_lo 0x11300c00 data 16
[    3.807798] mtk-pcie-gen3 11300000.pcie: msi#0x11 address_hi 0x0 address_lo 0x11300c00 data 17
[    3.816398] mtk-pcie-gen3 11300000.pcie: msi#0x12 address_hi 0x0 address_lo 0x11300c00 data 18
[    3.824998] mtk-pcie-gen3 11300000.pcie: msi#0x13 address_hi 0x0 address_lo 0x11300c00 data 19
[    3.833598] mtk-pcie-gen3 11300000.pcie: msi#0x14 address_hi 0x0 address_lo 0x11300c00 data 20
[    3.842197] mtk-pcie-gen3 11300000.pcie: msi#0x15 address_hi 0x0 address_lo 0x11300c00 data 21
[    3.850795] mtk-pcie-gen3 11300000.pcie: msi#0x16 address_hi 0x0 address_lo 0x11300c00 data 22
[    3.859395] mtk-pcie-gen3 11300000.pcie: msi#0x17 address_hi 0x0 address_lo 0x11300c00 data 23
[    3.867993] mtk-pcie-gen3 11300000.pcie: msi#0x18 address_hi 0x0 address_lo 0x11300c00 data 24
[... cut ...]
[    4.074241] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    4.421624] mtk-pcie-gen3 11310000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    4.430412] mtk-pcie-gen3: probe of 11310000.pcie failed with error -110
[    4.437938] FIT: Detected U-Boot 2024.01-OpenWrt-r26450-6510eb3b5d
...
```

```
root@OpenWrt:~# lspci 
00:00.0 PCI bridge: MEDIATEK Corp. Device 7988 (rev 01)
01:00.0 Network controller: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter
```

12 V switched off

P-Light, 2024-06-04:
@frank-w hello!
Could you show how use [your repo](https://github.com/openwrt/openwrt/issues/14767#issuecomment-2094721713) with openwrt snapshot? Do I need to compile it myself or it possible add your repo url in feeds file? Thank you!

frank-w, 2024-06-04:
I do not use openwrt,so i do not know if and how my kernel source can beused there. But my kernel source does not contain patches for this pcie problem.

ptpt52, 2024-08-28:
same issue on latest openwrt build.

ptpt52, 2024-08-28:
ping @dangowrt

james-hitech, 2024-09-14:
same issue see on openwrt main branch: 
Linux version 6.6.50 (sw@dev) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r27352-47370eb9ba) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Fri Sep 13 20:17:35 2024

and here is the log when probing pcie:
---
`[    0.053351] mtk-xsphy soc:xphy@11e10000: failed to get ref_clk(id-1)
[    0.053876] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    0.053893] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    0.053905] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    0.053918] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    0.054011] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    0.054023] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    0.054033] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    0.054043] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    0.054061] /soc/pcie@11290000: Failed to get clk index: 0 ret: -517
[    0.054070] mtk-pcie-gen3 11290000.pcie: failed to get clocks
[    0.054132] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    0.054143] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    0.054152] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    0.054161] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    0.054178] /soc/pcie@11300000: Failed to get clk index: 0 ret: -517
[    0.054185] mtk-pcie-gen3 11300000.pcie: failed to get clocks
[    0.054258] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    0.054269] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    0.054278] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    0.054287] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    0.054303] /soc/pcie@11310000: Failed to get clk index: 0 ret: -517
[    0.054310] mtk-pcie-gen3 11310000.pcie: failed to get clocks
[    0.058563] Serial: 8250/16550 driver, 3 ports, IRQ sharing disabled
[    0.059361] printk: console [ttyS0] disabled
[    0.079625] 11000000.serial: ttyS0 at MMIO 0x11000000 (irq = 99, base_baud = 2500000) is a ST16650V2
[    0.079660] printk: console [ttyS0] enabled
[    0.992132] 11000100.serial: ttyS1 at MMIO 0x11000100 (irq = 100, base_baud = 2500000) is a ST16650V2
[    1.022015] 11000200.serial: ttyS2 at MMIO 0x11000200 (irq = 101, base_baud = 2500000) is a ST16650V2
[    1.033885] loop: module loaded
[    1.038316] spi-nand spi0.0: Winbond SPI NAND was found.
[    1.043663] spi-nand spi0.0: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 64
[    1.051896] 2 fixed-partitions partitions found on MTD device spi0.0
[    1.058261] OF: Bad cell count for /soc/spi@11007000/spi_nand@0/partitions
[    1.065135] OF: Bad cell count for /soc/spi@11007000/spi_nand@0/partitions
[    1.072129] Creating 2 MTD partitions on "spi0.0":
[    1.076915] 0x000000000000-0x000000200000 : "bl2"
[    1.083028] 0x000000200000-0x000008000000 : "ubi"
[    1.161031] ubi0: default fastmap pool size: 50
[    1.165578] ubi0: default fastmap WL pool size: 25
[    1.170358] ubi0: attaching mtd1
[    1.989704] ubi0: scanning is finished
[    2.003136] ubi0: attached mtd1 (name "ubi", size 126 MiB)
[    2.008618] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[    2.015486] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[    2.022264] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[    2.029212] ubi0: good PEBs: 1008, bad PEBs: 0, corrupted PEBs: 0
[    2.035294] ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
[    2.042504] ubi0: max/mean erase counter: 2/0, WL threshold: 4096, image sequence number: 1697192243
[    2.051621] ubi0: available PEBs: 105, total reserved PEBs: 903, PEBs reserved for bad PEB handling: 20
[    2.061007] ubi0: background thread "ubi_bgt0d" started, PID 243
[    2.067672] block ubiblock0_1: created from ubi0:1(rootfs)
[    2.073161] ubiblock: device ubiblock0_1 (rootfs) set to be root filesystem
[    2.223596] i2c_dev: i2c /dev entries driver
[    2.230836] mtk-wdt 1001c000.watchdog: Watchdog enabled (timeout=31 sec, nowayout=0)
[    2.238888] mtk-cpufreq mtk-cpufreq: failed to initialize dvfs info for cpu0
[    2.246400] mtk-msdc 11230000.mmc: Got CD GPIO
[    2.246514] NET: Registered PF_INET6 protocol family
[    2.256239] Segment Routing with IPv6
[    2.259904] In-situ OAM (IOAM) with IPv6
[    2.263868] NET: Registered PF_PACKET protocol family
[    2.268924] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    2.282066] 8021q: 802.1Q VLAN Support v1.8
[    2.298787] phy phy-soc:xphy@11e10000.3: type_sw - reg 0x218, index 0
[    2.306019] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[... 8 lines cut ...]
[    2.672200] mtk-pcie-gen3 11280000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    2.681007] mtk-pcie-gen3: probe of 11280000.pcie failed with error -110
[    2.688066] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    2.695385] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    2.701649] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[... cut ...]
[    3.469829] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    3.812195] mtk-pcie-gen3 11310000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.820973] mtk-pcie-gen3: probe of 11310000.pcie failed with error -110
[    3.828295] FIT: Detected U-Boot 2024.07-OpenWrt-r27352-47370eb9ba
[    3.834473] FIT: Selected configuration: "config-mt7988a-bananapi-bpi-r4" (OpenWrt bananapi_bpi-r4)`

it seems this issue pending almost half year , does any final solutions about this issue ?
@dangowrt  @frank-w

ejohb, 2024-11-18:
I seem to have a very similar problem on a R4 with the [BE14 WiFi Module](https://docs.banana-pi.org/en/BPI-R4/BananaPi_BPI-R4-NIC-BE14) installed

```
root@router:~# dmesg|grep OpenW
[    0.000000] Linux version 6.6.60 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28062-be3e7a84bd) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Nov 12 02:31:06 2024
```

```
root@router:~# dmesg|grep pci
[    0.000000] Kernel command line: console=ttyS0,115200n1 pci=pcie_bus_perf root=/dev/fit0 rootwait
[    0.051714] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    0.051734] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    0.051745] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    0.051758] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    0.051848] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    0.051858] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    0.051867] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    0.051876] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    0.051894] /soc/pcie@11290000: Failed to get clk index: 0 ret: -517
[    0.051903] mtk-pcie-gen3 11290000.pcie: failed to get clocks
[    0.051967] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    0.051976] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    0.051985] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    0.051994] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    0.052010] /soc/pcie@11300000: Failed to get clk index: 0 ret: -517
[    0.052018] mtk-pcie-gen3 11300000.pcie: failed to get clocks
[    0.052091] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    0.052101] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    0.052110] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    0.052119] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    0.052135] /soc/pcie@11310000: Failed to get clk index: 0 ret: -517
[    0.052142] mtk-pcie-gen3 11310000.pcie: failed to get clocks
[    2.412262] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    2.419606] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    2.425884] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    2.434342] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    2.782550] mtk-pcie-gen3 11280000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    2.791349] mtk-pcie-gen3: probe of 11280000.pcie failed with error -110
[    2.798403] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    2.805718] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    2.811983] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    2.820416] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    3.162568] mtk-pcie-gen3 11290000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.171350] mtk-pcie-gen3: probe of 11290000.pcie failed with error -110
[    3.178223] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    3.185533] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    3.191793] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    3.200226] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    3.542559] mtk-pcie-gen3 11300000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.551338] mtk-pcie-gen3: probe of 11300000.pcie failed with error -110
[    3.558206] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    3.565517] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    3.571775] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    3.580206] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    3.922557] mtk-pcie-gen3 11310000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.931336] mtk-pcie-gen3: probe of 11310000.pcie failed with error -110
```

I get no `lspci` output whatsoever:

```
root@router:~# lspci -vvv
root@router:~# 
```

Note: I tried installing an m.2 storage card into one of the spare m.2 slots and that _did_ show up in `lspci`.

frank-w, 2024-11-18:
Have you enabled the sw4 to supply 12v to the be14...(Switch off for other cards)

matt1606, 2024-11-20:
Hi,
On my own example I can confirm that the pcie ports are working fine with https://github.com/frank-w/BPI-Router-Linux/tree/6.11-main when I plug a not usable at all `Broadcom BCM943228HMB` HP OEM card, but `Qualcomm Atheros AR9462 Wireless Network Adapter (rev 01)` manufactured by Lite-On Communications Inc doesn't show up.
That should be on bpi-r4 hw warning list for any potential buyers!

Regards, Mateusz

ejohb, 2024-11-21:
> Have you enabled the sw4 to supply 12v to the be14?...(Switch off for other cards)

That was the issue for me. All working now. Many thanks.

0x5826, 2024-12-03:
same issue, B-key M2 PCIE Channel don't work.

yanosz, 2024-12-10:
Also observing an empty lspci output on an bpi-r4. OpenWRT-version is: 24.10.0-rc2, r28161-ea17e958b9

najdanovicivan, 2024-12-26:
I have 2 issues with NVMe drives on 24.10.0-rc3

**Drive 1**

In key m slot I have Kingston snv3s/2000g disk which is not detected at all the issue is
[    3.812436] mtk-pcie-gen3 11290000.pcie: PCIe link down, current LTSSM state: polling.compliance (0x3000003

The disk is not detected even with the official Debian image. The I've tested M key slot with Samsung PM981 256GB drive and it works without Issue.  I'm wondering if 2TB NV3 drive can even work on single PCIe lane.

**Drive 2**

In B key slot I have SK hynix BC501 drive which is detected but not working properly it cannot be mounted or formatted. 
I'm getting this errors
[    6.052184] nvme nvme0: failed to set APST feature (4)
[    6.120186] nvme nvme0: 4/0/0 default/read/poll queues
[    6.132333]  nvme0n1: p1
[   15.567438] EXT4-fs (nvme0n1p1): ext4_check_descriptors: Block bitmap for group 0 overlaps superblock
[   15.576670] EXT4-fs (nvme0n1p1): group descriptors corrupted!

I've tested the same drive with official Debian image and it works without any issues. 

lscpi 
```
0003:00:00.0 PCI bridge: MEDIATEK Corp. Device 7988 (rev 01)
0003:01:00.0 Non-Volatile memory controller: SK hynix BC501 NVMe Solid State Drive
```

dmesg | grep pci
```
[    0.000000] Kernel command line: console=ttyS0,115200n1 pci=pcie_bus_perf root=/dev/fit0 rootwait acpi=off
[    0.052461] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    0.052480] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    0.052491] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    0.052504] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    0.052597] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    0.052607] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    0.052616] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    0.052626] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    0.052644] /soc/pcie@11290000: Failed to get clk index: 0 ret: -517
[    0.052653] mtk-pcie-gen3 11290000.pcie: failed to get clocks
[    0.052715] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    0.052724] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    0.052733] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    0.052743] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    0.052759] /soc/pcie@11300000: Failed to get clk index: 0 ret: -517
[    0.052766] mtk-pcie-gen3 11300000.pcie: failed to get clocks
[    0.052839] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    0.052849] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    0.052857] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    0.052866] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    0.052882] /soc/pcie@11310000: Failed to get clk index: 0 ret: -517
[    0.052890] mtk-pcie-gen3 11310000.pcie: failed to get clocks
[    2.442997] mtk-pcie-gen3 11280000.pcie: host bridge /soc/pcie@11280000 ranges:
[    2.450313] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    2.456580] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    2.465014] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    2.739815] mtk-pcie-gen3 11280000.pcie: set IO trans window[0]: cpu_addr = 0x20000000, pci_addr = 0x20000000, size = 0x200000
[    2.751204] mtk-pcie-gen3 11280000.pcie: set MEM trans window[1]: cpu_addr = 0x20200000, pci_addr = 0x20200000, size = 0x200000
[    2.762668] mtk-pcie-gen3 11280000.pcie: set MEM trans window[2]: cpu_addr = 0x20400000, pci_addr = 0x20400000, size = 0x400000
[    2.774132] mtk-pcie-gen3 11280000.pcie: set MEM trans window[3]: cpu_addr = 0x20800000, pci_addr = 0x20800000, size = 0x800000
[    2.785596] mtk-pcie-gen3 11280000.pcie: set MEM trans window[4]: cpu_addr = 0x21000000, pci_addr = 0x21000000, size = 0x1000000
[    2.797146] mtk-pcie-gen3 11280000.pcie: set MEM trans window[5]: cpu_addr = 0x22000000, pci_addr = 0x22000000, size = 0x2000000
[    2.808695] mtk-pcie-gen3 11280000.pcie: set MEM trans window[6]: cpu_addr = 0x24000000, pci_addr = 0x24000000, size = 0x4000000
[    2.820514] mtk-pcie-gen3 11280000.pcie: PCI host bridge to bus 0003:00
[    2.827126] pci_bus 0003:00: root bus resource [bus 00-ff]
[    2.832606] pci_bus 0003:00: root bus resource [io  0x0000-0x1fffff] (bus address [0x20000000-0x201fffff])
[    2.842245] pci_bus 0003:00: root bus resource [mem 0x20200000-0x27ffffff]
[    2.849113] pci_bus 0003:00: scanning bus
[    2.853136] pci 0003:00:00.0: [14c3:7988] type 01 class 0x060400
[    2.859143] pci 0003:00:00.0: reg 0x10: [mem 0x00000000-0x00007fff 64bit]
[... 108 lines cut ...]
[    5.879234] nvme nvme0: pci function 0003:01:00.0
[    5.883951] nvme 0003:01:00.0: enabling device (0000 -> 0002)
[    5.889697] nvme 0003:01:00.0: enabling bus mastering
[    5.903286] nvme 0003:01:00.0: save config 0x00: 0x13271c5c
[    5.908848] nvme 0003:01:00.0: save config 0x04: 0x00100406
[    5.914419] nvme 0003:01:00.0: save config 0x08: 0x01080200
[    5.919981] nvme 0003:01:00.0: save config 0x0c: 0x00000000
[    5.925545] nvme 0003:01:00.0: save config 0x10: 0x20200004
[    5.931106] nvme 0003:01:00.0: save config 0x14: 0x00000000
[... cut ...]
[    5.975609] nvme 0003:01:00.0: save config 0x34: 0x00000080
[    5.981170] nvme 0003:01:00.0: save config 0x38: 0x00000000
[    5.986732] nvme 0003:01:00.0: save config 0x3c: 0x00000173
[    6.052184] nvme nvme0: failed to set APST feature (4)
[    6.120186] nvme nvme0: 4/0/0 default/read/poll queues
[    6.132333]  nvme0n1: p1
[   15.567438] EXT4-fs (nvme0n1p1): ext4_check_descriptors: Block bitmap for group 0 overlaps superblock
[   15.576670] EXT4-fs (nvme0n1p1): group descriptors corrupted!
```

najdanovicivan, 2024-12-26:
I was able to make the SK hynix BC501 drive work

The issue happens because of pci=pcie_bus_perf flag. When this is set I'll get the error 
nvme nvme0: failed to set APST feature (4)

When I try mounting the partition the data will get corrupted. 

Removing the pci=pcie_bus_perf resolves the APST issues and the drive gets mounted properly

barannikovav, 2025-10-02:
Got same problem on latest snapshot
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd090]
[    0.000000] Linux version 6.12.49 (a@a) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 14.3.0 r31257-9bb6f2568e) 14.5
[    0.000000] Machine model: Banana Pi BPI-R4 (2x SFP+)
[    0.000000] Memory limited to 2048MB
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004304ffff (320 KiB) nomap non-reusable secmon@43000000
[    0.000000] OF: reserved mem: 0x0000000047cc0000..0x0000000047dbffff (1024 KiB) nomap non-reusable wmcpu-reserved@47cc0000
[    0.000000] OF: reserved mem: 0x000000004f600000..0x000000004f63ffff (256 KiB) nomap non-reusable wo-emi@4f600000
[    0.000000] OF: reserved mem: 0x000000004f640000..0x000000004f67ffff (256 KiB) nomap non-reusable wo-emi@4f640000
[    0.000000] OF: reserved mem: 0x000000004f680000..0x000000004f6bffff (256 KiB) nomap non-reusable wo-emi@4f680000
[    0.000000] OF: reserved mem: 0x000000004f700000..0x000000004fefffff (8192 KiB) nomap non-reusable wo-data@4f700000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004304ffff]
[    0.000000]   node   0: [mem 0x0000000043050000-0x0000000047cbffff]
[    0.000000]   node   0: [mem 0x0000000047cc0000-0x0000000047dbffff]
[    0.000000]   node   0: [mem 0x0000000047dc0000-0x000000004f5fffff]
[    0.000000]   node   0: [mem 0x000000004f600000-0x000000004f6bffff]
[    0.000000]   node   0: [mem 0x000000004f6c0000-0x000000004f6fffff]
[    0.000000]   node   0: [mem 0x000000004f700000-0x000000004fefffff]
[    0.000000]   node   0: [mem 0x000000004ff00000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.4
[    0.000000] percpu: Embedded 20 pages/cpu s42520 r8192 d31208 u81920
[    0.000000] pcpu-alloc: s42520 r8192 d31208 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyS0,115200n1 pci=pcie_bus_perf root=/dev/fit0 rootwait ubi.block=0,fit mem=2048M
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 524288
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 2MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000bd600000-0x00000000bd800000] (2MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 416 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: GICD_CTRL.DS=0, SCR_EL3.FIQ=0
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000065] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000071] pid_max: default: 32768 minimum: 301
[    0.002141] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[... 99 lines cut ...]
[    1.190674] Creating 2 MTD partitions on "spi0.0":
[    1.195457] 0x000000000000-0x000000200000 : "bl2"
[    1.201520] 0x000000200000-0x000008000000 : "ubi"
[    1.276084] ubi0: default fastmap pool size: 50
[    1.280617] ubi0: default fastmap WL pool size: 25
[    1.285396] ubi0: attaching mtd1
[    1.704407] ubi0: scanning is finished
[    1.712688] ubi0: attached mtd1 (name "ubi", size 126 MiB)
[    1.718170] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[... cut ...]
[    1.782379] OF: Bad cell count for /soc/spi@11007000/flash@0/partitions
[    1.789292] block ubiblock0_4: created from ubi0:4(fit)
[    1.911604] i2c_dev: i2c /dev entries driver
[    1.916319] /soc/i2c@11003000/rt5190a@64: Fixed dependency cycle(s) with /soc/i2c@11003000/rt5190a@64/regulators/buck1
[    1.928071] mtk-lvts-thermal 1100a000.lvts: golden temp=60
[    1.934967] mtk-wdt 1001c000.watchdog: Watchdog enabled (timeout=31 sec, nowayout=0)
[    1.947269] mtk-ccifreq cci: failed to add devfreq device: -517
```