openwrt/openwrt #14771: OpenWrt does not have firmware for Intel AX201 WiFi card

patrakov, 2024-03-02:
### Describe the bug

There is no package containing firmware for Intel AX201 WiFi card. It is not the same as for AX200.

### OpenWrt version

r25366-7f13b9f8be

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

LENOVO 82MA

### Image kind

Official downloaded image

### Actual behaviour

Even after installing all `*iwl*` packages, the correct firmware is not there:

```
[ 5265.223581] Intel(R) Wireless WiFi driver for Linux
[ 5265.229409] iwlwifi 0000:00:14.3: Detected crf-id 0x3617, cnv-id 0x20000302 wfpm id 0x80000000
[ 5265.233226] iwlwifi 0000:00:14.3: PCI dev a0f0/0074, rev=0x351, rfid=0x10a100
[ 5265.236708] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-77.ucode failed with error -2
[ 5265.241003] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-QuZ-a0-hr-b0-77.ucode
[ 5265.246678] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-76.ucode failed with error -2
[ 5265.250893] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-QuZ-a0-hr-b0-76.ucode
...
[ 5265.503113] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-50.ucode failed with error -2
[ 5265.507338] iwlwifi 0000:00:14.3: Falling back to sysfs fallback for: iwlwifi-QuZ-a0-hr-b0-50.ucode
[ 5265.512689] iwlwifi 0000:00:14.3: no suitable firmware found!
[ 5265.515619] iwlwifi 0000:00:14.3: minimum version required: iwlwifi-QuZ-a0-hr-b0-50
[ 5265.519125] iwlwifi 0000:00:14.3: maximum version supported: iwlwifi-QuZ-a0-hr-b0-77
[ 5265.522681] iwlwifi 0000:00:14.3: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```

### Expected behaviour

The firmware should load.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-02:
[image]
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/

pkgadd, 2024-03-02:
As it's a tad difficult to download the binary, as-is in binary form, from the gitweb interface (but possible), it may be easier to `git clone https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/` the whole repo and to copy out the necessary files.

If you do use gitweb, make sure that you get the binary file - and not just a html file.

patrakov, 2024-03-03:
Downloading the correct file from git and copying it to OpenWrt works. Still, it's a feature request to have it packaged :)

ynezz, 2024-03-04:
> I guess I have to retest with 23.02.5, too, in order to get the correct filename. I will do it later.

Additional testing doesn't hurt, but the filename and maximum supported API version 77 should be the same.

closed 2024-03-24 by openwrt-bot

MaciejLisCK, 2024-05-24:
> Downloading the correct file (77) from git and copying it to OpenWrt works. Still, it's a feature request to have it packaged :)
> 
> I guess I have to retest with 23.02.5, too, in order to get the correct filename. I will do it later.

For me, downloading version 77, get error of maximum version not supported.
It does work with version 72,

`cd /lib/firmware`
`wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/iwlwifi-QuZ-a0-hr-b0-72.ucode`

brada4, 2024-05-24:
@MaciejLisCK And your OpenWRT version is....
And your dmesg section says.....

commit 8db83d4c (2024-03-24): linux-firmware: package Intel AX201 firmware

Alexander reported following:

 iwlwifi 0000:00:14.3: Detected crf-id 0x3617, cnv-id 0x20000302 wfpm id 0x80000000
 iwlwifi 0000:00:14.3: PCI dev a0f0/0074, rev=0x351, rfid=0x10a100
 iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-77.ucode failed with error -2

It seems, that as of the current date, the highest firmware API version
supported by Linux 6.8-rc7 is still 77.

Closes: #14771
PR #14782 (2024-03-24): linux-firmware: package Intel AX201 firmware
Alexander reported following:

```
 iwlwifi 0000:00:14.3: Detected crf-id 0x3617, cnv-id 0x20000302 wfpm id 0x80000000
 iwlwifi 0000:00:14.3: PCI dev a0f0/0074, rev=0x351, rfid=0x10a100
 iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-77.ucode failed with error -2
```

It seems, that as of the current date, the highest firmware API version supported by Linux 6.8-rc7 is still 77.

Closes: #14771