openwrt/openwrt #14773: `sysupgrade -b -` does not work anymore

rany2, 2024-03-02:
### Describe the bug

It simply creates a file called `-` instead of printing to stdout, this broke my backup scripts.

### OpenWrt version

336a531c15e7fa5f8a42a7b9f2662c249aafce89

### OpenWrt release

336a531c15e7fa5f8a42a7b9f2662c249aafce89

### OpenWrt target/subtarget

N/A

### Device

N/A

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-03-02:
@rmilecki I've taken a look and I think this commit is to blame, I've linked the line in specific: https://github.com/openwrt/openwrt/commit/e36cc530927c4294c1510d7cdf89f0b9739e6cde#diff-cc3419ad054e045f4f6a07d3b634c844190512ef3779d61f5f188ca5fc439f5eR266

csharper2005, 2024-03-03:
SNAPSHOT: "Generate archive" action in Luci produces zero length backup.tar.gz. @rany2 Is this the same problem?

rany2, 2024-03-03:
@csharper2005 Yep, Luci calls cgi-backup which also pipes to stdout: https://github.com/openwrt/cgi-io/blob/901b0f0463c9d16a8cf5b9ed37118d8484bc9176/main.c#L652-L653

closed 2024-03-04 by openwrt-bot

tones111, 2024-03-04:
@jow- 
I manually applied the changes to my r25381 snapshot and I'm able to generate backups now.  Thanks for the quick fix!

commit 6f6406a1 (2024-03-04): base-files: sysupgrade: fix streaming backup archives to stdout

Due to previous refactoring in sysupgrade, writing backup archives to
stdout became impossible since the hardcoded gzip output redirection
did not account for the `-` special case filename.

Fix this issue by substituting `-` with `/proc/self/fd/1` in the tar
archive output path variable.

Also remove a redundant `rm -f` of the target file path that occurs
before the file could've possibly been written.

Fixes: #14773
Fixes: https://github.com/openwrt/luci/issues/6961
Fixes: e36cc53092 ("base-files: sysupgrade: use tar helper to  [... cut]