openwrt/openwrt #14783: gettext-full: known security bug

zpc0, 2024-03-04:
### Describe the bug

According to [Fedora](https://src.fedoraproject.org/rpms/gettext/c/1b17af5), gettext libtextstyle has a known security bug. At least, Fedora and ArchLinux disables libtextstyle builds.

### OpenWrt version

fbb942a

### OpenWrt release

snapshot

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-04:
Could you fill issue template with existing router and version tags and tag maintainer and last commiters?

brada4, 2024-03-05:
you are not only dramatic but also wrong.
https://src.fedoraproject.org/rpms/gettext//blob/rawhide/f/gettext.spec#_454

zpc0, 2024-03-06:
> but also wrong. https://src.fedoraproject.org/rpms/gettext//blob/rawhide/f/gettext.spec#_454

`libtextstyle` still uses `libcroco` which has a security bug.
Please search "CVE-2020-12825", then check source code carefully.
I have no idea whether this bug "actually" affects OpenWrt people or not, but it is a fact that many distributions have patched this bug or disabled `libtextstyle` build completely.

I don't know why Fedora package maintainer restored `libtextstyle`, but this is just RawHide only. Even beta-stage Fedora 40 has no `libtextstyle`.

brada4, 2024-03-06:
Seems only dovecot depends on target gettext-full package, rest use it only on build host.
Just disable it , and check you can build semanage here and run dovecot from packages....

neheb, 2024-03-09:
that target dependency might be wrong: https://github.com/openwrt/packages/commit/96145db78af6afa27d8fa857d2af384b1c623259

neheb, 2024-03-09:
actually it might not even be needed...

```
> find staging_dir/ -name iconv.m4 -exec md5sum '{}' \;
c5515337a3d16c0eabc21d8c8ddd0909  staging_dir/host/share/aclocal/iconv.m4
c5515337a3d16c0eabc21d8c8ddd0909  staging_dir/host/share/gnulib/m4/iconv.m4
88e17b0d896baf9a03994015ac79ba9d  staging_dir/target-aarch64_generic_musl/usr/share/aclocal/iconv.m4
88e17b0d896baf9a03994015ac79ba9d  staging_dir/hostpkg/share/aclocal/iconv.m4
```

brada4, 2024-03-09:
@neheb seems useless file is included in the package adding cve.