openwrt/openwrt #14792: eapol_test (wpa_supplicant 2.10) + Openssl v3.0.8 cannot authenticate to 802.1x network

VimuVR3, 2024-03-05:
### Describe the bug

I used eapol_test provided with wpa_supplicant 2.10 + OpenSSLv3.0.8 to test connecting to an 802.1x network, I got the following results.

PEAP+MSCHAPV2 got failed due to an error code returned from SSL.

PEAP+GTC got passed

I want to use PEAP+MSCHAPV2 authentication with openssl3.0.8 for windows NPS servers where GTC is not supported (only MSCHAPV2 is supported by default). And while using freeradius too, same issue has occured. I used eapol_test (wpa_supplicant2.10) + OpenSSLv1.1.1t where it can authenticate to 802.1x using PEAP+MSCHAPV2

**Steps to reproduce:**

Authenticate to the 802.1x network using eapol_test and select the encryption mode PEAP + MSCHAPV2 with OpenSSL v3.0.8

**Observed behavior:**

Connect timeout

**Expected behavior:**

Connect OK

**Log snippet using eapol_test utility to test 802.1x authentication provided in wpa_supplicant:**

> EAP-MSCHAPV2: Generating Challenge Response
Get randomness: len=16 entropy=0
random from os_get_random - hexdump(len=16): 77 b5 40 38 12 e0 da 75 3c 96 41 67 9a 40 6a f5
random_mix_pool - hexdump(len=20): 0d b9 b1 bf 70 7c bd fa 8b 8c 0a 46 d8 96 87 a4 8e 89 0d 7d
random from internal pool - hexdump(len=16): 52 c7 66 0a bf 85 ed d3 d8 c1 5b 8c 5d 36 f0 8e
mixed random - hexdump(len=16): 25 72 26 32 ad 65 37 a6 e4 57 1a eb c7 76 9a 7b
MSCHAPV2: Identity - hexdump_ascii(len=5):
61 64 6d 69 6e admin
MSCHAPV2: Username - hexdump_ascii(len=5):
61 64 6d 69 6e admin
MSCHAPV2: auth_challenge - hexdump(len=16): 3e 04 b8 c6 6b 23 3d 40 cb bf 55 7b e4 b2 85 d9
MSCHAPV2: peer_challenge - hexdump(len=16): 25 72 26 32 ad 65 37 a6 e4 57 1a eb c7 76 9a 7b
MSCHAPV2: username - hexdump_ascii(len=5):
61 64 6d 69 6e admin
MSCHAPV2: password - hexdump_ascii(len=8):
70 61 73 73 77 6f 72 64 password
OpenSSL: EVP_DigestInit_ex failed: error:0308010C:digital envelope routines::unsupported
EAP-MSCHAPV2: Failed to derive response
EAP: method process -> ignore=FALSE methodState=MAY_CONT decision=FAIL eapRespData=0
EAP: EAP entering state SEND_RESPONSE
EAP: No eapRespData available
EAP: EAP entering state IDLE
EAPOL test timed out
EAPOL: EAP key not available
EAPOL: EAP Session-Id not available
WPA: Clear old PMK and PTK
EAP: deinitialize previously used EAP method (25, PEAP) at EAP deinit
MPPE keys OK: 0 mismatch: 1
FAILURE

**Error code:**
OpenSSL: EVP_DigestInit_ex failed: error:0308010C:digital envelope routines::unsupported
EAP-MSCHAPV2: Failed to derive response

### OpenWrt version

f544226+r49254

### OpenWrt release

Chaos Calmer

### OpenWrt target/subtarget

ipq/ipq60xx

### Device

Qualcomm Technologies, Inc. IPQ6018/AP-CP03-C1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castiel652, 2024-03-05:
> Chaos Calmer
That's an ancient release and your board is not official supported.

brada4, 2024-03-05:
You are using vendor sdk aka unsupported fork. You have to return where you got it from.

closed 2024-03-05 by ynezz