openwrt/openwrt #14801: 23.05.x Update breaks WAN - RT-AX53U/1800U

radiohead-lab, 2024-03-05:
### Describe the bug

I am opening as a bug as per the advise of Developer in OpenWrt forum. The problem is still exist and we didn't get answer/clarification is that a real issue or already resolved on 23.05.2 release.

Issue is reported in the following thread and noted on ASUS RT-AX53U hw description page as well, but let me summarize it:
https://forum.openwrt.org/t/23-05-0-update-breaks-wan-rt-ax53u-1800u/174225

According to the observations WAN interface is not working on 23.05.0 and possible on latest release (.2) as well. Many cases the issue come up after couple of reboots. However people reported that it's working fine via Snapshot versions, and no issues with WAN either on (.2) latest release. It's quiet confusing, and this is a very popular hardware, used by many people. 
Is that possible to verify and confirm by OpenWrt developers that the issue is still exist/fixed or not an issue at all?

I can confirm that it is working fine on 22.03.6 latest old-stable release perfectly.

Many people are waiting for the outcome if you check the topic in the above mentioned forum entry.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Hard to reproduce, but according to observations after couple of reboots WAN interface (and WLAN either) was gone.

### Actual behaviour

Same as described in reproduction step: after couple of reboots WAN interface (and WLAN either) was gone, but it is working fine on 22.03.6 latest old-stable release perfectly.

### Expected behaviour

WAN (and WLAN) interface must be up and operational all the time.

### Additional info

Please try to test it on a real hardware and confirm.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-05:
There was a big red banner when flashing that said to wipe configuration. You often times burn intentionally walking fire.
New soc switch layout, reset conf after botched upgrade and it is golden again.

brada4, 2024-03-06:
> Many people are

tag them all, a% least 3

radiohead-lab, 2024-03-06:
> ou often times burn intentionally walking fire.

Sorry, but I don't get you. Maybe its my fault. Can you clarify it? 
Is this still an issue? If yes is that possible to add the device specification wiki how to workaround it if any?
If not an issue just because of new hardware then how to upgrade proper way to avoid this issue?
Really sorry for my ignorant, its my fault but can you please give a detailed explanation?

radiohead-lab, 2024-03-06:
> > Many people are
> 
> tag them all, at least 3

Many people in the forum I mean. I cannot tag them here, but you can find 3 at least or more in the forum:
https://forum.openwrt.org/t/23-05-0-update-breaks-wan-rt-ax53u-1800u/174225/27

brada4, 2024-03-06:
2 people complained, 3rd who complained reset conf and it started to work. Rest of crowd just believed them. Simply wan port is moved to separate netcard out of switch and old ethernet config is no longer valid. There is a warning to reset config when upgrading. Just get on with 23.05.2 , you will need to restore individual confs manually taking those in config backup as a reference.

radiohead-lab, 2024-03-06:
Ok, got it. Thanks for your explanation. It's ok and if this is the solution then we can consider this issue as fixed with this workaround.
Can someone modify the wiki page as you suggested and remove WAN port issue note?
https://openwrt.org/toh/asus/rt-ax53u
I guess it's quiet confusing in this case if there is no issue actually just config files must be restored manually.

brada4, 2024-03-06:
There is no workaround, it is pretty well documented that you have to reset configuration sometimes on upgrade especially if told so.

radiohead-lab, 2024-03-06:
OK, in this case can you remove the note from Wiki page? I can comment the thread and we can close this issue.

brada4, 2024-03-06:
You need to contact forum user inserting that line.

radiohead-lab, 2024-03-06:
OK, let me contact with him and if modified then we can close this issue.
Thanks for your help. I will update this ticket.

brada4, 2024-03-06:
You need to contact individual in forums, really not a problem that code change would solve.
These devices got improvement, and likely ad-hoc edits in wiki and endless "i dont care to read upgrade warnings" threads.
https://github.com/openwrt/openwrt/commit/f1c9afd801380a05a91d979b475c76cc0a67caae

brada4, 2024-03-06:
There is another wiki-related bug, you need to start w 22.03.6 or 23.05.2 due to select devices shipped with uninitialized flash.
like: https://github.com/openwrt/openwrt/issues/12674

radiohead-lab, 2024-03-06:
@brada4 
make sense! it should be mentioned as a Warning! too on Wiki page...however its TFTP method, but also upgrade related.
we can ask @humaita to do additional favor to add this issue as a Warning!

@humaita: can you please add as a Warning! too this issue:
https://github.com/openwrt/openwrt/issues/12674
thanks in advance.

brada4, 2024-03-06:
You know, if it was upfront 22..6 and 23..2 in the page unlikely anyone would land with serial cable.

radiohead-lab, 2024-03-06:
Yeah, anyway we can have a note with that TFTP installation issue as you suggested. At least everyone can see it...as it is not yet closed, still an open issue.

brada4, 2024-03-06:
It is not TFTP-only issue, same error initialising overlay at first boot would force to recover with good version in openwrt recovery mode as minimum.

radiohead-lab, 2024-03-06:
Ok, @humaita can edit the page at least with the warning.
...but back to this thread. What do you think @brada4 can we close the ticket?

brada4, 2024-03-06:
I think it is worth to have working release links in page. Yes, thanks for raising heads at inaccurate wiki ;-)

radiohead-lab, 2024-03-06:
> I think it is worth to have working release links in page. Yes, thanks for raising heads at inaccurate wiki ;-)

Ok, you are right. Let's ask @humaita too to modify release links. I will update the separate forum thread with that request too.

radiohead-lab, 2024-03-06:
OK, so as we concluded we can close this issue. However no additional code changes are needed but let's summarize what process needs to be followed during upgrade to 23.05.2 (commented by @brada4):
- _Simply wan port is moved to separate netcard out of switch and old ethernet config is no longer valid. There is a warning to reset config when upgrading. Just get on with 23.05.2 , you will need to restore individual confs manually taking those in config backup as a reference._

Wiki page !Warning! was modified by @humaita, thanks for your support:
https://openwrt.org/toh/asus/rt-ax53u

Closing this ticket.

closed 2024-03-06 by radiohead-lab

brada4, 2024-03-06:
Would be nice to update copy/paste sections to 22..**6** for initial install, that initializes flash correctly and one can downgrade if they are really into it. Recovery is possible with putty, not a big deal, but would save lots of stress in support channels.

radiohead-lab, 2024-03-06:
> Would be nice to update copy/paste sections to 22..**6** for initial install, that initializes flash correctly and one can downgrade if they are really into it. Recovery is possible with putty, not a big deal, but would save lots of stress in support channels.

ok, thanks. let me share with @humaita this one too

Meister1593, 2024-04-26:
I just upgraded from `22.03.5, r20134-5f15225c1e` to `23.05.3, r23809-234f1a2efa` through [sysupgrade image](https://mirror-03.infra.openwrt.org/releases/23.05.3/targets/ramips/mt7621/openwrt-23.05.3-ramips-mt7621-asus_rt-ax53u-squashfs-sysupgrade.bin) and did not require any configuration resets, WAN works fine.

brada4, 2024-04-26:
@Meister1593 seems never customized bridge migrates just fine, ie wan@dsa lan1@dsa safely changes to wan and lan1@eth0

Meister1593, 2024-04-26:
> @Meister1593 seems never customized bridge migrates just fine, ie wan@dsa lan1@dsa safely changes to wan and lan1@eth0

if that's what it is, then i guess it's good?
[image]

brada4, 2024-04-26:
Check `ip link` in command line and /etc/config/network -> there should be no eth0 or dsa as bridge port, picture is correct according to colour codes.

Meister1593, 2024-04-26:
> Check `ip link` in command line and /etc/config/network -> there should be no eth0 or dsa as bridge port, picture is correct according to colour codes.

`ip link`: lo, eth0, wan, lan1@eth0, lan2@eth0, lan3@eth0, br-lan, phy1-ap0, phy0-ap0
`/etc/config/network` (interfaces): loopback, lan, wan, wan6

Meister1593, 2024-04-26:
The only mention of bridge is on `br-lan`

brada4, 2024-04-26:
Nice :-) People complaining just overcustomized their installation.

mailinglists35, 2024-05-01:
done using the gui upgrade. but noticed this while diff'ing `ip link` before and after: under 22.03.x devices have CoDel support, while in 23.05.x this is missing. had no idea what codel is until this upgrade. is this intentional or has it been removed by mistake?

````
$ diff before.txt after.txt 
4c4
< 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc fq_codel state UP qlen 1000
---
> 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
````
wikipedia says "it is designed to overcome [bufferbloat](https://en.wikipedia.org/wiki/Bufferbloat) in [networking hardware](https://en.wikipedia.org/wiki/Networking_hardware), such as [routers](https://en.wikipedia.org/wiki/Router_(computing)), by setting limits on the delay [network packets](https://en.wikipedia.org/wiki/Network_packet) experience as they pass through [buffers](https://en.wikipedia.org/wiki/Buffer_(telecommunication)) in this equipment. CoDel aims to improve on the overall performance of the [random early detection](https://en.wikipedia.org/wiki/Random_early_detection) (RED) algorithm by addressing some of its fundamental misconceptions, as perceived by Jacobson, and by being easier to manage."

brada4, 2024-05-01:
fq_codel is default qdisc and attaches to individual queues of multiqueue adapter. just like pfifo_fast does on upstream kernel.

mailinglists35, 2024-05-07:
yes but on 23.05.x I'm only seeing it on pppoe device, no longer on physical devices like I saw on 22.03.x
hence my asking is this some sort of deliberate choice? not that I would care too much about it, just an offtopic spinoff talk that I noticed when doing a diff between ip link output. also seeing `mq` instead that was not in 22.03

````
root@OpenWrt:~# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
3: wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
4: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
5: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
6: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
9: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
10: phy1-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
11: phy1-ap1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
12: phy0-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
13: pppoe-wan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN qlen 3
    link/ppp 
14: phy0-ap1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether anon brd ff:ff:ff:ff:ff:ff
````

on x86_64 ubuntu with kernel 6.5 I see fc_codel for physical and ppp interfaces (except bridges), while on oracle linux x86_64 with kernel 5.15 I see mq instead of fc_codel

staples1347, 2026-03-16:
> There was a big red banner when flashing that said to wipe configuration. You often times burn intentionally walking fire. New soc switch layout, reset conf after botched upgrade and it is golden again.

I didn't see any warning about this in either the release notes or when I ran sysupgrade when updating to 23.05.  Was this warning only shown in the web ui upgrade or was it only triggered for specific configs?  I didn't have to reset my config either.