openwrt/openwrt #14816: SHA256 checksums do not match images for Espressobin v7

paulmsander, 2024-03-07:
### Describe the bug

I downloaded the 23.05.2 images for the Espressobin v7 from the downloads.openwrt.org web site using Firefox on a Mac.  When I ran the "shasum -a 256 -b" command on these images, the checksums did not match those shown on the download site.  I also downloaded the factory image for the TP-LINK Archer C7, and the sum for that matched, so it does not seem to be a problem with the Mac.  Here are the checksums computed on my Mac:

4bb1eb338e4c0ea10c18a1277802740f59b1b9d7cf8c875ca4a49b003f4f149e *openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
f6b8a7b5e6c0abfbf2620bce4e7bf4511cc7c17fd6a59350c6f61d0e5f704dc1 *openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
cd191b250adbe5be1f13b9eebcb027c1496316e85d1f936b10a9db7798a7c4bc *openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img

Again, the first checksum in this list matches the download page, the rest do not.

### OpenWrt version

n/a - image not installed, version 23.05.2 for Espressobin v7

### OpenWrt release

n/a - image not installed, version 23.05.2 for Espressobin v7

### OpenWrt target/subtarget

n/a - image not installed, version 23.05.2 for Espressobin v7

### Device

n/a - image not installed, Espressobin v7, mvebu-cortexa53

### Image kind

Official downloaded image

### Steps to reproduce

1. Download the image(s) from the downloads.openwrt.org web site using Firefox on a Mac:
[Download OpenWrt firmware for your device](https://firmware-selector.openwrt.org/?version=23.05.2&target=mvebu%2Fcortexa53&id=globalscale_espressobin-v7-emmc)   

2. Run the "shasum -a 256 -b" command and specify the name of the image(s) on the command line.
3. Compare the output of the shasum command to the SHA-256 checksum shown on the download page.

### Actual behaviour

The checksums shown on the download page don't match for the downloaded Espressobin v7 images as computed by the "shasum -a 256 -b" command on the Mac.

### Expected behaviour

The checksums for the Espressobin v7 images should match.

### Additional info

Don't have any helf to pul.

### Diffconfig

```text
n/a - This was a production image.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

paulmsander, 2024-03-07:
Hi,

Thank you very much for your very prompt response!

I've attached a screenshot of the download page.  Looking at your output, I see a .gz file extension that I don't see in my filesystem.  Perhaps Safari did some unauthorized decompression.  I gzipped the files and recomputed the checksums, and they did not match.  Here is my output:

***@***.*** OpenWRT % for f in *; do echo "FILE $f:"; ( set -x; file $f; shasum -a 256 -b $f; gzip -c $f | shasum -a 256 -b; ); echo; done
FILE openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin:
+-zsh:106> file openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin: data
+-zsh:106> shasum -a 256 -b openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
4bb1eb338e4c0ea10c18a1277802740f59b1b9d7cf8c875ca4a49b003f4f149e *openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
+-zsh:106> gzip -c openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
+-zsh:106> shasum -a 256 -b
c009e63ee49199e1fb652dcf2999c937a4a58d1a504dfe7d85dd42eed17e7e03 *-

FILE openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img:
+-zsh:106> file openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img: DOS/MBR boot sector; partition 1 : ID=0x83, active, start-CHS (0x2,0,33), end-CHS (0x23,0,48), startsector 2048, 33280 sectors; partition 2 : ID=0x83, start-CHS (0x24,9,10), end-CHS (0xf8,6,6), startsector 36864, 213504 sectors
+-zsh:106> shasum -a 256 -b openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
f6b8a7b5e6c0abfbf2620bce4e7bf4511cc7c17fd6a59350c6f61d0e5f704dc1 *openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
+-zsh:106> gzip -c openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
+-zsh:106> shasum -a 256 -b
59fa8f99b649c0ab173d7688ac8ef11647d1a24ef576487b836838fdcd6f931d *-

FILE openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img:
+-zsh:106> file openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img: DOS/MBR boot sector; partition 1 : ID=0x83, active, start-CHS (0x2,0,33), end-CHS (0x23,0,48), startsector 2048, 33280 sectors; partition 2 : ID=0x83, start-CHS (0x24,9,10), end-CHS (0xf8,6,6), startsector 36864, 213504 sectors
+-zsh:106> shasum -a 256 -b openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
cd191b250adbe5be1f13b9eebcb027c1496316e85d1f936b10a9db7798a7c4bc *openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
+-zsh:106> gzip -c openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
+-zsh:106> shasum -a 256 -b
30803f13a6a460f4e323347f1272d013087c4103ead2b97bb88ade7d078c7973 *-

***@***.*** OpenWRT %

I'll try again with a different browser to see if I can retrieve the images without decompression, then re-check the checksums.

Paul

> 
> 
> I just pulled https://downloads.openwrt.org/releases/23.05.2/targets/mvebu/cortexa53/openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img.gz and get the checksum 363cd76ea9a46c244c1bbc7b7c1984c9b8f246df289453e8501b98ec280be0eb, the same as reported on the download directory page here: https://downloads.openwrt.org/releases/23.05.2/targets/mvebu/cortexa53/
> Can you open the directory page on your end and see what checksums are reported there?

—
Start-up companies use state-of-the-art tools because they’re not old enough to use something else.

Paul Sander
***@***.*** ***@***.***>

paulmsander, 2024-03-07:
Hi,

I tried downloading the images using Firefox, and it did not decompress the files.  The checksums also match.

Thank you for your time!

Paul

> 
> Hi,
> 
> Thank you very much for your very prompt response!
> 
> I've attached a screenshot of the download page.  Looking at your output, I see a .gz file extension that I don't see in my filesystem.  Perhaps Safari did some unauthorized decompression.  I gzipped the files and recomputed the checksums, and they did not match.  Here is my output:
> 
> FILE openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin:
> +-zsh:106> file openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
> openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin: data
> +-zsh:106> shasum -a 256 -b openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
> 4bb1eb338e4c0ea10c18a1277802740f59b1b9d7cf8c875ca4a49b003f4f149e *openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
> +-zsh:106> gzip -c openwrt-23.05.2-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin
> +-zsh:106> shasum -a 256 -b
> c009e63ee49199e1fb652dcf2999c937a4a58d1a504dfe7d85dd42eed17e7e03 *-
> 
> FILE openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img:
> +-zsh:106> file openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
> openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img: DOS/MBR boot sector; partition 1 : ID=0x83, active, start-CHS (0x2,0,33), end-CHS (0x23,0,48), startsector 2048, 33280 sectors; partition 2 : ID=0x83, start-CHS (0x24,9,10), end-CHS (0xf8,6,6), startsector 36864, 213504 sectors
> +-zsh:106> shasum -a 256 -b openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
> f6b8a7b5e6c0abfbf2620bce4e7bf4511cc7c17fd6a59350c6f61d0e5f704dc1 *openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
> +-zsh:106> gzip -c openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-ext4-sdcard.img
> +-zsh:106> shasum -a 256 -b
> 59fa8f99b649c0ab173d7688ac8ef11647d1a24ef576487b836838fdcd6f931d *-
> 
> FILE openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img:
> +-zsh:106> file openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
> openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img: DOS/MBR boot sector; partition 1 : ID=0x83, active, start-CHS (0x2,0,33), end-CHS (0x23,0,48), startsector 2048, 33280 sectors; partition 2 : ID=0x83, start-CHS (0x24,9,10), end-CHS (0xf8,6,6), startsector 36864, 213504 sectors
> +-zsh:106> shasum -a 256 -b openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
> cd191b250adbe5be1f13b9eebcb027c1496316e85d1f936b10a9db7798a7c4bc *openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
> +-zsh:106> gzip -c openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img
> +-zsh:106> shasum -a 256 -b
> 30803f13a6a460f4e323347f1272d013087c4103ead2b97bb88ade7d078c7973 *-
> 
> 
> 
> I'll try again with a different browser to see if I can retrieve the images without decompression, then re-check the checksums.
> 
> Paul
> 
> 
>> 
>> 
>> I just pulled https://downloads.openwrt.org/releases/23.05.2/targets/mvebu/cortexa53/openwrt-23.05.2-mvebu-cortexa53-globalscale_espressobin-v7-emmc-squashfs-sdcard.img.gz and get the checksum 363cd76ea9a46c244c1bbc7b7c1984c9b8f246df289453e8501b98ec280be0eb, the same as reported on the download directory page here: https://downloads.openwrt.org/releases/23.05.2/targets/mvebu/cortexa53/
>> Can you open the directory page on your end and see what checksums are reported there?
> 
> Start-up companies use state-of-the-art tools because they’re not old enough to use something else.
> 
> Paul Sander
> 
> <Screenshot 2024-03-07 at 12.07.30 AM.png>

—
A woodworker can never have too many vises.

Paul Sander
***@***.***

brada4, 2024-03-07:
If you uncompress file you get different checksum?

jow-, 2024-03-07:
I vaguely recall similar incidents in the past, Safari tends to silently decompress .gz compressed files and does it in a way that is not immediately obvious. Maybe we can tune the server side configuration to deliver image files as `application/octet-stream` to force browsers to treat them as opaque binary files.

brada4, 2024-03-07:
https://macreports.com/how-to-download-zip-files-without-unzipping/