openwrt/openwrt #14823: Bananapi BPI-R4 kernel panic on SPI1 usage

KungfuPancake, 2024-03-07:
### Describe the bug

The board crashes with a kernel panic as soon as the SPI device is being used. Binding the device works, but sending data does not.

### OpenWrt version

r25404-e247763617

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

Add the following definition to `mt7988a-bananapi-bpi-r4.dts`

```
&spi1 {
        pinctrl-names = "default";
        pinctrl-0 = <&spi1_pins>;
        status = "okay";
        
        spidev@0 {   
                compatible = "linux,spidev";
                reg = <0>;
                #address-cells = <1>;
                #size-cells = <0>;
                spi-max-frequency = <500000>;
                status = "okay";
        };
};
```

After booting, register and bind the SPI device to spidev:

```
echo "spidev" > /sys/bus/spi/devices/spi1.0/driver_override
echo "spi1.0" > /sys/bus/spi/drivers/spidev/bind
```

Run a simple device test:

```
opkg update
opkg install spidev_test
spidev-test -D /dev/spidev1.0
```

### Actual behaviour

A kernel panic occurs. Here's mine:

```
[  130.687392] SError Interrupt on CPU3, code 0x00000000bf000002 -- SError
[  130.687404] CPU: 3 PID: 3275 Comm: spidev_test Not tainted 6.1.80 #0
[  130.687410] Hardware name: Bananapi BPI-R4 (DT)
[  130.687412] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  130.687416] pc : mtk_spi_set_cs+0x54/0x7c
[  130.687427] lr : spi_set_cs+0x88/0x1e0
[  130.687433] sp : ffffffc00c643cb0
[  130.687435] x29: ffffffc00c643cb0 x28: ffffff80c0d30000 x27: 0000000000000000
[  130.687442] x26: 0000000000420084 x25: ffffff80c3e40358 x24: ffffff80c4678da0
[  130.687447] x23: ffffff80c0a20400 x22: 0000000000000000 x21: 0000000000000000
[  130.687452] x20: 0000000000000000 x19: ffffff80c0a20400 x18: 0000000000000000
[  130.687456] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[  130.687460] x14: 0000000000000000 x13: 0000000258891970 x12: ffffff80c0a06a80
[  130.687464] x11: 0000000000000065 x10: 00000000000002b8 x9 : 0000000000000001
[  130.687468] x8 : 0000000000895440 x7 : 0000000002e4ecfa x6 : 0000000000000023
[  130.687472] x5 : 0000000000000000 x4 : ffffff80c094c800 x3 : ffffffc008dfd018
[  130.687476] x2 : 0000000000000000 x1 : 0000000000000001 x0 : ffffff80c094cd80
[  130.687482] Kernel panic - not syncing: Asynchronous SError Interrupt
[  130.687484] SMP: stopping secondary CPUs
[  130.687488] Kernel Offset: disabled
[  130.687489] CPU features: 0x00000,00020004,0000400b
[  130.687492] Memory Limit: none
```

### Expected behaviour

The SPI test runs successfully.

### Additional info

I initially encountered this problem while trying to use the SPI interface to attach a display. Using spidev is the easiest way to trigger the kernel panic.

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_kmod-spi-dev=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_luci-theme-openwrt-2020=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

KungfuPancake, 2024-03-08:
The panic also happens when using a 6.6 kernel (https://github.com/dangowrt/openwrt/tree/mediatek-6.6)

KungfuPancake, 2024-03-10:
I'm having a hard time debugging this myself, as I don't have any experience doing so. Can anyone point me to some explanation how to get started on that? Right now I can't even decrypt the trace to get to the offending code line.

KungfuPancake, 2024-03-16:
I've played with the GPIO a bit and the relevant pins seem to be assigned correctly:

```
root@OpenWrt:/# gpiomon gpiochip0 28
[  378.618844] mt7988-pinctrl 1001f000.pinctrl: pin SPI1_CSB already requested by 11008000.spi; cannot claim for pinctrl_moore:540
[  378.630349] mt7988-pinctrl 1001f000.pinctrl: pin-28 (pinctrl_moore:540) status -22
gpiomon: error waiting for events: Invalid argument
root@OpenWrt:/# gpiomon gpiochip0 29
[  401.685575] mt7988-pinctrl 1001f000.pinctrl: pin SPI1_MISO already requested by 11008000.spi; cannot claim for pinctrl_moore:541
[  401.697161] mt7988-pinctrl 1001f000.pinctrl: pin-29 (pinctrl_moore:541) status -22
gpiomon: error waiting for events: Invalid argument
root@OpenWrt:/# gpiomon gpiochip0 30
[  404.275294] mt7988-pinctrl 1001f000.pinctrl: pin SPI1_MOSI already requested by 11008000.spi; cannot claim for pinctrl_moore:542
[  404.286891] mt7988-pinctrl 1001f000.pinctrl: pin-30 (pinctrl_moore:542) status -22
gpiomon: error waiting for events: Invalid argument
root@OpenWrt:/# gpiomon gpiochip0 31
[  409.719736] mt7988-pinctrl 1001f000.pinctrl: pin SPI1_CLK already requested by 11008000.spi; cannot claim for pinctrl_moore:543
[  409.731249] mt7988-pinctrl 1001f000.pinctrl: pin-31 (pinctrl_moore:543) status -22
gpiomon: error waiting for events: Invalid argument
```

Everything looks like it should work, but I'm still running into the kernel panic. I'm giving up for now and will postpone the project.

KungfuPancake, 2024-04-01:
Some more information: I tested the display with the bit bang SPI implementation which works, but is extremely slow. Just to poke around a bit I changed to the hardware SPI and fbcon, which to my surprise showed a graphical console on the TFT, up until some point at the end of the boot process where presumably a panic occurs and the board resets. If I don't use fbcon but try and fill the framebuffer from userspace I get kernel panics again.
So at some point during boot something happens that renders SPI1 unusable.
Next step would be to get simultaneous output on UART and try to pinpoint where the bad magic happens.

KungfuPancake, 2024-04-01:
Use the following bootargs to get both fbcon and serial console:
`console=ttyS0,115200n1 pci=pcie_bus_perf root=/dev/fit0 earlycon console=tty0 fbcon=map:0`

SPI related kernel panic again.

```
[    4.714261] device eth0 entered promiscuous mode
[    4.719163] DSA: tree 0 setup
[    4.723189] UBI: auto-attach mtd1
[    4.726769] ubi: mtd1 is already attached to ubi0
[    4.731736] UBI error: cannot attach mtd1
[    4.738568] VFS: Mounted root (squashfs filesystem) readonly on device 259:1.
[    4.746235] Freeing unused kernel memory: 448K
[    4.819680] Run /sbin/init as init process
[    4.824035]   with arguments:
[    4.827255]     /sbin/init
[    4.830093]   with environment:
[    4.833488]     HOME=/
[    4.835972]     TERM=linux
[    4.932656] init: Console is alive
[    4.936555] init: - watchdog -
[    4.959628] fb_ili9341 spi1.0: SPI transfer timed out
[    4.965106] SError Interrupt on CPU1, code 0x00000000bf000002 -- SError
[    4.965113] CPU: 1 PID: 461 Comm: kworker/1:7 Not tainted 6.1.82 #0
[    4.965119] Hardware name: Bananapi BPI-R4 (DT)
[    4.965122] Workqueue: events fb_deferred_io_work
[    4.965136] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    4.965141] pc : mtk_spi_reset+0x20/0x64
[    4.965148] lr : mtk_spi_set_cs+0x70/0x7c
[    4.965151] sp : ffffffc00a2aba10
[    4.965153] x29: ffffffc00a2aba10 x28: ffffff80ff780a38 x27: ffffff80c0a42340
[    4.965158] x26: 0000000000000001 x25: 0000003cf937f300 x24: 0000003cf937f3e0
[    4.965163] x23: 0000000000000001 x22: ffffffc00a2abbb0 x21: 0000000000000000
[    4.965168] x20: 0000000000000000 x19: ffffff80c0ddbc00 x18: 0000000000000102
[    4.965172] x17: 0000000000000000 x16: 0000000000000000 x15: ffffffc008b59d40
[    4.965177] x14: 0000000000000306 x13: 0000000000000102 x12: 00000000ffffffea
[    4.965181] x11: 00000000ffffefff x10: ffffffc008bb1d40 x9 : ffffffc008b59ce8
[    4.965186] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
[    4.965190] x5 : 0000000000000fff x4 : ffffff80c0957800 x3 : ffffffc008ddd018
[    4.965194] x2 : ffffffc008ddd000 x1 : 0000000000000000 x0 : ffffff80c0957d80
[    4.965200] Kernel panic - not syncing: Asynchronous SError Interrupt
[    4.965202] SMP: stopping secondary CPUs
[    4.965205] Kernel Offset: disabled
[    4.965207] CPU features: 0x00000,00020004,0000400b
[    4.965210] Memory Limit: none
PANIC at PC : 0x000000004300490c
```

onlyfly34, 2024-05-27:
Hi, I also found a bug about SPI1 dtsi settings. But I'm not sure if it's related to your bug or not.
The bug I found out is that we should change [SPI pll](https://github.com/openwrt/openwrt/blob/f7f8099aa3ae551df14b3b2bac332a80e861a690/target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a.dtsi#L794) from **CLK_TOP_SPI_SEL** to **CLK_TOP_SPIM_MST_SEL**

1. Which SPI device do you use? (Max SPI bus clock is 500KHz?)
2. If your SPI device's max clock rate is 500KHz, then I suggest you to change parent-clk to CLK_TOP_XTAL. Then SPI1 node in dtsi will look like this:
```
		spi1: spi@11008000 {
			compatible = "mediatek,ipm-spi-single", "mediatek,spi-ipm";
			reg = <0 0x11008000 0 0x100>;
			interrupts = <GIC_SPI 141 IRQ_TYPE_LEVEL_HIGH>;
			clocks = <&topckgen CLK_TOP_XTAL>,
				 <&topckgen CLK_TOP_SPIM_MST_SEL>,
				 <&infracfg CLK_INFRA_104M_SPI1>,
				 <&infracfg CLK_INFRA_66M_SPI1_HCK>;
			clock-names = "parent-clk", "sel-clk", "spi-clk",
				      "spi-hclk";
			#address-cells = <1>;
			#size-cells = <0>;
			pinctrl-names = "default";
			pinctrl-0 = <&spi1_pins>;
			status = "disabled";
		};
```

BRs,
Sky.Huang

KungfuPancake, 2024-05-31:
Thank you for your input and suggestions! I will try this as soon as I'm back home, which will be end of next week. I'll let you know!

KungfuPancake, 2024-06-14:
I finally found time to test this. With a max bus frequency of 8MHz and a sel-clk value of CLK_TOP_SPIM_MST_SEL the kernel still panics with the same issue :(

KungfuPancake, 2024-07-13:
It seems I'm not the only one running into this:

https://lists.openwall.net/netdev/2024/04/07/61
https://forum.openwrt.org/t/banana-bpi-r4-spi/193295/11

phuongnam-code, 2024-10-21:
Guys, can you tell me when BPI-R4 will have official v23 version?

dangowrt, 2024-11-03:
Should be fixed by 2b173ab730897f53539cab39e87e2b4638cb482a
Please try and report

KungfuPancake, 2024-11-03:
Thank you, I will try this sometime today

KungfuPancake, 2024-11-03:
```
root@OpenWrt:~# spidev_test -s 8000000 -D /dev/spidev1.0 -v
spi mode: 0x0
bits per word: 8
max speed: 8000000 Hz (8000 kHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  |......@.........................|
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  |......@.........................|
```

It seems to be working now! Thank you very much, @dangowrt!

closed 2024-11-03 by KungfuPancake