openwrt/openwrt #14824: iPhone connection freeze in standby after some time

trunneml, 2024-03-07:
### Describe the bug

On my Banana Pi R3 running 23.05.2 my two iPhones are loosing internet access on the 5 GHz band after some time. This occurs multiple times on a day, when the phone is in standby. The wifi signal is still good and the connection is shown on the router and the phone. But the phone can't access any thing (100% package loose) and is not pingable. Turning Wifi off and on helps and it is working again for some hours.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

1. Connect iPhone to 5Ghz
2. wait without using the phone
3. after sometime the try to use the internet on the phone

### Actual behaviour

Pages are not loading on the phone

### Expected behaviour

Network is working on the phone

### Additional info

Wifi-Chip is the internal MT7975P

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

khanjui, 2024-03-07:
option dtim_period '3'
option wpa_group_rekey '86400'
option disassoc_low_ack '0'

Try these.

brada4, 2024-03-07:
Are your network parameters set up to their expectations?
https://support.apple.com/en-us/HT202068 ?

trunneml, 2024-03-09:
Hi,

I will try wpa_group_rekey. The rest I have already tried.

brada4, 2024-03-09:
Problem seems widespread on apples own forums. Even their store routers implicated.
Among original ideas there is disabling wifi location service on the phone.

Do you have wpa2 / wap3 mix / wpa3 identical accesspoints in all bands ( examine config file sections) and no tkip/wpa1/legacy rates?

trunneml, 2024-03-10:
> Do you have wpa2 / wap3 mix / wpa3 identical accesspoints in all bands ( examine config file sections) and no tkip/wpa1/legacy rates?

WPA2/WPA3 mix. Density is at high. The SSID for the iPhones is only in the 5 GHz Band. 
There is another 2,4 GHz SSID configured for my IoT Devices.

trunneml, 2024-03-10:
> option dtim_period '3'
> 
> option wpa_group_rekey '86400'
> 
> option disassoc_low_ack '0'
> 
> 
> 
> Try these.

wpa_group_rekey '86400' seems to help no connection loose since Friday. 😎

I will close the issue, when it holds the complete weekend.

trunneml, 2024-03-10:
> https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n1870 disable tkip as was the default recommrnded in their guide and effect will be same. You can make separate ap for 20 years old wifi clients requiring tkip

I don't know what you trying to say.
I don't use TKIP. Only WPA2 and WPA3 is in use. The default for wpa_group_rekey in OpenWRT is 600 even for CCMP. I changed that setting to 86400 as @khanjui recommended. Since then no more disconnects.

brada4, 2024-03-10:
The default displayed in luci in absence of parameter is not indicative of actual hostapd default, you need to enable hostapd debug log to catch it re-keying diverging from upstream default and causing interoperability woes.

khanjui, 2024-03-10:
> The default displayed in luci in absence of parameter is not indicative of actual hostapd default, you need to enable hostapd debug log to catch it re-keying diverging from upstream default and causing interoperability woes.

https://forum.openwrt.org/t/does-openwrt-set-wpa-strict-rekey-1/123275/2

brada4, 2024-03-10:
I dont get 10min rekey in wpa1+2 aes wpa23+roam and wpa3 accesspoints with wpad-mbedtls (non default)
I set roaming pair with WPA2+aes and I can trigger strict rekey by clicking both clients off to other band, but no regular messages.
 
https://openwrt.org/docs/guide-developer/debugging#logging_hostapd_behaviour -> set to level 1
You will see result (or not) in 10 minutes.

brada4, 2024-03-14:
Will not go anywhere without trace log.

trunneml, 2024-03-14:
> Will not go anywhere without trace log.

 sadly there was to much logging activated and the disconnect was already pruned. ‚Logread -f‘ didn’t show the timespan anymore. 
 
I try to geht the trace log when it happens next time.

trunneml, 2024-03-14:
Here is the syslog output from the last 3 hours. I had no problems in this time period.
```
Thu Mar 14 10:46:45 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: start SAE authentication (RX commit, status=126 (SAE_HASH_TO_ELEMENT))
Thu Mar 14 10:46:45 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: SAE authentication (RX confirm, status=0 (SUCCESS))
Thu Mar 14 10:46:45 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-AUTHENTICATE.indication(b0:8c:75:ec:15:e7, unknown)
Thu Mar 14 10:46:45 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 0 notification
Thu Mar 14 10:46:45 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 1
Thu Mar 14 10:46:45 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 2
Thu Mar 14 10:46:46 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: authentication OK (open system)
Thu Mar 14 10:46:46 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 0 notification
Thu Mar 14 10:46:46 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-AUTHENTICATE.indication(b0:8c:75:ec:15:e7, OPEN_SYSTEM)
Thu Mar 14 10:46:46 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: authenticated
Thu Mar 14 10:46:46 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 3
Thu Mar 14 10:46:46 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 4
Thu Mar 14 10:46:46 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 5
Thu Mar 14 10:46:46 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query timed out
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: authentication OK (open system)
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 0 notification
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-AUTHENTICATE.indication(b0:8c:75:ec:15:e7, OPEN_SYSTEM)
Thu Mar 14 10:46:47 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: authenticated
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association OK (aid 5)
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 8 notification
Thu Mar 14 10:46:47 2024 daemon.notice hostapd: phy1-ap0: STA-OPMODE-MAX-BW-CHANGED b0:8c:75:ec:15:e7 40
Thu Mar 14 10:46:47 2024 daemon.notice hostapd: phy1-ap0: STA-OPMODE-N_SS-CHANGED b0:8c:75:ec:15:e7 2
Thu Mar 14 10:46:47 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: associated (aid 5)
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-ASSOCIATE.indication(b0:8c:75:ec:15:e7)
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-DELETEKEYS.request(b0:8c:75:ec:15:e7)
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: binding station to interface 'phy1-ap0'
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 1 notification
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 4 notification
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: sending 1/4 msg of 4-Way Handshake
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: received EAPOL-Key frame (2/4 Pairwise)
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: sending 3/4 msg of 4-Way Handshake
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: received EAPOL-Key frame (4/4 Pairwise)
Thu Mar 14 10:46:47 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.1X: authorizing port
Thu Mar 14 10:46:47 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 RADIUS: starting accounting session E3F782176C5125C4
Thu Mar 14 10:46:47 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: pairwise key handshake completed (RSN)
Thu Mar 14 10:46:47 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED b0:8c:75:ec:15:e7
Thu Mar 14 10:46:47 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.70 b0:8c:75:ec:15:e7
Thu Mar 14 10:46:47 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.70 b0:8c:75:ec:15:e7 Desis-iPhone
Thu Mar 14 12:05:59 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.55 00:a0:de:a4:98:d9
Thu Mar 14 12:05:59 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.55 00:a0:de:a4:98:d9 AV475
Thu Mar 14 12:18:56 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED b0:8c:75:ec:15:e7
Thu Mar 14 12:18:56 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 2 notification
Thu Mar 14 12:18:56 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.1X: unauthorizing port
Thu Mar 14 12:18:56 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: disassociated
[... cut ...]
Thu Mar 14 12:18:57 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 RADIUS: starting accounting session 8D9BEFF6511CA9CB
Thu Mar 14 12:18:57 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: pairwise key handshake completed (RSN)
Thu Mar 14 12:18:57 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED b0:8c:75:ec:15:e7
Thu Mar 14 12:18:57 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.70 b0:8c:75:ec:15:e7
Thu Mar 14 12:18:57 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.70 b0:8c:75:ec:15:e7 Desis-iPhone
```

trunneml, 2024-03-14:
Should I also activate some logging in the mt76 kernel module or is hostapd debug enough for the first step?

brada4, 2024-03-14:
Not really, it will log fatal driver conditions anyway, this seems more like hostapd logic vs apple supplicant's
You can install hostapd-utils for "official view" at non-working client, though iwinfo may already fill the bill.

trunneml, 2024-03-14:
This time the iphone lost the connection will using it and reconnected immediately:

```
Thu Mar 14 17:17:04 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: start SAE authentication (RX commit, status=126 (SAE_HASH_TO_ELEMENT))
Thu Mar 14 17:17:04 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: SAE authentication (RX confirm, status=0 (SUCCESS))
Thu Mar 14 17:17:04 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-AUTHENTICATE.indication(b0:8c:75:ec:15:e7, unknown)
Thu Mar 14 17:17:04 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 0 notification
Thu Mar 14 17:17:04 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 1
Thu Mar 14 17:17:05 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 2
Thu Mar 14 17:17:05 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 3
Thu Mar 14 17:17:05 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 4
Thu Mar 14 17:17:05 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query attempt 5
Thu Mar 14 17:17:05 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association SA Query timed out
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: authentication OK (open system)
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 0 notification
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-AUTHENTICATE.indication(b0:8c:75:ec:15:e7, OPEN_SYSTEM)
Thu Mar 14 17:17:06 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: authenticated
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: association OK (aid 3)
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 8 notification
Thu Mar 14 17:17:06 2024 daemon.notice hostapd: phy1-ap0: STA-OPMODE-MAX-BW-CHANGED b0:8c:75:ec:15:e7 40
Thu Mar 14 17:17:06 2024 daemon.notice hostapd: phy1-ap0: STA-OPMODE-N_SS-CHANGED b0:8c:75:ec:15:e7 2
Thu Mar 14 17:17:06 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: associated (aid 3)
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-ASSOCIATE.indication(b0:8c:75:ec:15:e7)
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 MLME: MLME-DELETEKEYS.request(b0:8c:75:ec:15:e7)
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.11: binding station to interface 'phy1-ap0'
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 1 notification
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: event 4 notification
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: sending 1/4 msg of 4-Way Handshake
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: received EAPOL-Key frame (2/4 Pairwise)
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: sending 3/4 msg of 4-Way Handshake
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: received EAPOL-Key frame (4/4 Pairwise)
Thu Mar 14 17:17:06 2024 daemon.debug hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 IEEE 802.1X: authorizing port
Thu Mar 14 17:17:06 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 RADIUS: starting accounting session 3F5C9C932B81E2C3
Thu Mar 14 17:17:06 2024 daemon.info hostapd: phy1-ap0: STA b0:8c:75:ec:15:e7 WPA: pairwise key handshake completed (RSN)
Thu Mar 14 17:17:06 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED b0:8c:75:ec:15:e7
```

brada4, 2024-03-14:
Signal strength changed... You need to reset various network aspects of iphone, changing signal strength is normal workings of wifi.

brada4, 2024-03-14:
Feels this piece about location service:
https://krebsonsecurity.com/2019/12/the-iphone-11-pros-location-data-puzzler/

rany2, 2024-03-14:
When this happens, what is the output of `tail /sys/kernel/debug/ieee80211/*/aql_pending`?

trunneml, 2024-03-18:
And it happend again:

```
root@OpenWrt:~# tail /sys/kernel/debug/ieee80211/*/aql_pending
==> /sys/kernel/debug/ieee80211/phy0/aql_pending <==
AC     AQL pending
VO     0 us
VI     0 us
BE     0 us
BK     0 us
total  0 us

==> /sys/kernel/debug/ieee80211/phy1/aql_pending <==
AC     AQL pending
VO     0 us
VI     0 us
BE     0 us
BK     0 us
total  0 us
```

[image]

```
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-ASSOCIATE.indication(0c:19:f8:a1:64:f3)
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-DELETEKEYS.request(0c:19:f8:a1:64:f3)
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: binding station to interface 'phy1-ap0'
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: event 1 notification
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: event 4 notification
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: sending 1/4 msg of 4-Way Handshake
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: received EAPOL-Key frame (2/4 Pairwise)
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: sending 3/4 msg of 4-Way Handshake
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: received EAPOL-Key frame (4/4 Pairwise)
Sun Mar 17 21:28:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.1X: authorizing port
Sun Mar 17 21:28:23 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 RADIUS: starting accounting session BA8CFFD395B3CF3E
Sun Mar 17 21:28:23 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: pairwise key handshake completed (RSN)
Sun Mar 17 21:28:23 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 0c:19:f8:a1:64:f3
Sun Mar 17 21:32:36 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 08:71:90:b7:d8:f1
Sun Mar 17 21:32:36 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: event 3 notification
Sun Mar 17 21:32:36 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.1X: unauthorizing port
Sun Mar 17 21:32:36 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.11: deauthenticated
Sun Mar 17 21:32:36 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 MLME: MLME-DEAUTHENTICATE.indication(08:71:90:b7:d8:f1, 1)
Sun Mar 17 21:32:36 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 MLME: MLME-DELETEKEYS.request(08:71:90:b7:d8:f1)
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.11: authentication OK (open system)
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 MLME: MLME-AUTHENTICATE.indication(08:71:90:b7:d8:f1, OPEN_SYSTEM)
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 MLME: MLME-DELETEKEYS.request(08:71:90:b7:d8:f1)
Sun Mar 17 21:33:02 2024 daemon.info hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.11: authenticated
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.11: association OK (aid 6)
Sun Mar 17 21:33:02 2024 daemon.info hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.11: associated (aid 6)
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 MLME: MLME-REASSOCIATE.indication(08:71:90:b7:d8:f1)
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 MLME: MLME-DELETEKEYS.request(08:71:90:b7:d8:f1)
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.11: binding station to interface 'phy1-ap0'
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: event 1 notification
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: start authentication
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.1X: unauthorizing port
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: sending 1/4 msg of 4-Way Handshake
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: received EAPOL-Key frame (2/4 Pairwise)
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: sending 3/4 msg of 4-Way Handshake
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: received EAPOL-Key frame (4/4 Pairwise)
Sun Mar 17 21:33:02 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 08:71:90:b7:d8:f1 auth_alg=open
Sun Mar 17 21:33:02 2024 daemon.debug hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 IEEE 802.1X: authorizing port
Sun Mar 17 21:33:02 2024 daemon.info hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 RADIUS: starting accounting session 1244B1BD73677421
Sun Mar 17 21:33:02 2024 daemon.info hostapd: phy1-ap0: STA 08:71:90:b7:d8:f1 WPA: pairwise key handshake completed (RSN)
Sun Mar 17 21:33:02 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 08:71:90:b7:d8:f1
[...]
Sun Mar 17 21:34:13 2024 daemon.err uwsgi[3617]: *** has_emperor mode detected (fd: 7) ***
Sun Mar 17 21:34:13 2024 daemon.err uwsgi[3617]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-webui.ini
[... cut ...]
Mon Mar 18 07:41:21 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 0c:19:f8:a1:64:f3 auth_alg=sae
Mon Mar 18 07:41:21 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.1X: authorizing port
Mon Mar 18 07:41:21 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 RADIUS: starting accounting session ED2B9E37C2A90F20
Mon Mar 18 07:41:21 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: pairwise key handshake completed (RSN)
Mon Mar 18 07:41:21 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 0c:19:f8:a1:64:f3
```

brada4, 2024-03-18:
Is it filtered or iphone does not do dhcp request at all?

trunneml, 2024-03-18:
I did not filter it. But dnsmasq logging has no facility set in the default config.

trunneml, 2024-03-18:
What I think is interesting is that the working iphone has the HE-MCS and HE-NSS attributes in the screenshot. The not working iphone not. So I don't think the problem is dhcp related. Anyway I activated the dhcp logging to syslog daemon.

brada4, 2024-03-18:
It is there in first log batch.

trunneml, 2024-03-18:
Than, I have no Idea why there is no dhcp log.

trunneml, 2024-03-18:
`root@OpenWrt:~# logread -e 0c:19:f8:a1:64:f3`
```
Fri Mar 15 13:01:22 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 0c:19:f8:a1:64:f3
Fri Mar 15 13:01:22 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: disassociated due to inactivity
Fri Mar 15 13:01:22 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-DISASSOCIATE.indication(0c:19:f8:a1:64:f3, 4)
Fri Mar 15 13:01:22 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-DELETEKEYS.request(0c:19:f8:a1:64:f3)
Fri Mar 15 13:01:23 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Mar 15 13:01:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-DEAUTHENTICATE.indication(0c:19:f8:a1:64:f3, 2)
Fri Mar 15 13:01:23 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-DELETEKEYS.request(0c:19:f8:a1:64:f3)
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: start SAE authentication (RX commit, status=126 (SAE_HASH_TO_ELEMENT))
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: SAE authentication (RX confirm, status=0 (SUCCESS))
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-AUTHENTICATE.indication(0c:19:f8:a1:64:f3, unknown)
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-DELETEKEYS.request(0c:19:f8:a1:64:f3)
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: association OK (aid 2)
Fri Mar 15 14:08:39 2024 daemon.notice hostapd: phy1-ap0: STA-OPMODE-MAX-BW-CHANGED 0c:19:f8:a1:64:f3 40
Fri Mar 15 14:08:39 2024 daemon.notice hostapd: phy1-ap0: STA-OPMODE-N_SS-CHANGED 0c:19:f8:a1:64:f3 2
Fri Mar 15 14:08:39 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: associated (aid 2)
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-ASSOCIATE.indication(0c:19:f8:a1:64:f3)
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-DELETEKEYS.request(0c:19:f8:a1:64:f3)
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: binding station to interface 'phy1-ap0'
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: event 1 notification
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: start authentication
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.1X: unauthorizing port
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: sending 1/4 msg of 4-Way Handshake
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: received EAPOL-Key frame (2/4 Pairwise)
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: sending 3/4 msg of 4-Way Handshake
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: received EAPOL-Key frame (4/4 Pairwise)
Fri Mar 15 14:08:39 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 0c:19:f8:a1:64:f3 auth_alg=sae
Fri Mar 15 14:08:39 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.1X: authorizing port
Fri Mar 15 14:08:39 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 RADIUS: starting accounting session D5934F590BD66F63
Fri Mar 15 14:08:39 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: pairwise key handshake completed (RSN)
Fri Mar 15 14:08:39 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 0c:19:f8:a1:64:f3
Fri Mar 15 16:02:04 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: sending 1/2 msg of Group Key Handshake
Fri Mar 15 16:02:05 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: EAPOL-Key timeout
Fri Mar 15 16:02:05 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: sending 1/2 msg of Group Key Handshake
Fri Mar 15 16:02:05 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: received EAPOL-Key frame (2/2 Group)
Fri Mar 15 16:02:05 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: group key handshake completed (RSN)
Fri Mar 15 16:02:05 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: received EAPOL-Key 2/2 Group with unexpected replay counter
Fri Mar 15 17:04:43 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: start SAE authentication (RX commit, status=126 (SAE_HASH_TO_ELEMENT))
Fri Mar 15 17:04:43 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: SAE authentication (RX confirm, status=0 (SUCCESS))
Fri Mar 15 17:04:43 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 MLME: MLME-AUTHENTICATE.indication(0c:19:f8:a1:64:f3, unknown)
Fri Mar 15 17:04:43 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: event 0 notification
Fri Mar 15 17:04:43 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: association SA Query attempt 1
Fri Mar 15 17:04:43 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.11: association SA Query attempt 2
[... cut ...]
Mon Mar 18 07:41:21 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 0c:19:f8:a1:64:f3 auth_alg=sae
Mon Mar 18 07:41:21 2024 daemon.debug hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 IEEE 802.1X: authorizing port
Mon Mar 18 07:41:21 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 RADIUS: starting accounting session ED2B9E37C2A90F20
Mon Mar 18 07:41:21 2024 daemon.info hostapd: phy1-ap0: STA 0c:19:f8:a1:64:f3 WPA: pairwise key handshake completed (RSN)
Mon Mar 18 07:41:21 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 0c:19:f8:a1:64:f3
```

brada4, 2024-03-18:
Worth somehow marking time spots where phone seems stuck.

trunneml, 2024-03-18:
I split the log. And added a not when the last push notification was received.

brada4, 2024-03-18:
Do you still have low ack deauth disabled?

trunneml, 2024-03-18:
Not in this try. For testing I reactivated it after the last disconnect. In the disconnect before I increased DTIM to 4.

```
config wifi-iface 'default_radio1'
        option device 'radio1'
        option mode 'ap'
        option ssid '***'
        option encryption 'sae'
        option key '***'
        option network 'lan'
        option max_inactivity '7200'
        option time_advertisement '2'
        option time_zone 'CET-1CEST,M3.5.0,M10.5.0/3'
        option dtim_period '4'
        option wpa_group_rekey '86400'
```

brada4, 2024-03-18:
what if you remove all custom options below `network 'lan'` ?
namely worth wifi-pinging idle client more often than every 2h

trunneml, 2024-03-18:
That was the starting point, where the iPhone lost the connection multiple times a day.

brada4, 2024-03-18:
disable apple wifi location.  Found in apple forums too. 
Nothing in openwrt stops communicating with iphone and iphone responds to kerpalives. No signal to ap  it cannot communicate anymore.

trunneml, 2024-03-18:
I can try it, but the location service is a needed feature on my phone, so this is not a solution for me. And many other routers (Fritzbox and other routers running openwrt) do not have this problem. How can I see when the wpa group rekeying is running in the logs?

trunneml, 2024-03-18:
> rekey is written in log...

How does this look like? Can you point me to the line in my provided logs?

> it is the new cloud+triangulation location service,

I created this config via luci:

[image]

trunneml, 2024-03-18:
I also had the disconnection issue without this setting. I also activated all checkboxes because 802.11v was working with my fritzbox. But it didn't help.

brada4, 2024-03-18:
This may send STA to sleep in un-coordinated manner. So far logs look "normal" up to the point you say it does not work anymore. Debug level 0 will log every packet (dont even think posting them), namely confirm empty idle packets get sent and get acked by iphone....

trunneml, 2024-03-18:
Question: in the time frame where the iphone lost its connection the second SSID `phy1-ap1` DEAUTH an other client and there where no more clients connected to this second SSID. Is it possible that the second SSID on the same radio is a problem? Is there a known bug?

Mon Mar 18 06:36:52 2024 daemon.notice hostapd: phy1-ap1: AP-STA-DISCONNECTED 48:a5:e7:06:ba:67
Mon Mar 18 06:36:52 2024 daemon.info hostapd: phy1-ap1: STA 48:a5:e7:06:ba:67 IEEE 802.11: disassociated
Mon Mar 18 06:36:53 2024 daemon.info hostapd: phy1-ap1: STA 48:a5:e7:06:ba:67 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

brada4, 2024-03-18:
> aql_pending

hostapd is supposed to clean up pending packets to that other station, could be a bug there.
Can you try more predictable situation - just disconnect second to last client while iphone is sleeping? Maybe it is easy to trigger repeatably.

trunneml, 2024-03-18:
I will try that, the next days.

trunneml, 2024-03-21:
New information when that problem occurs `tcpdump` shows messages from the iphone. Mosty DNS and IP6 neighbor solicitation.

```
13:22:17.431386 IP (tos 0x0, ttl 64, id 18739, offset 0, flags [DF], proto UDP (17), length 124)
    192.168.1.1.53 > 192.168.1.70.63557: 19527 0/1/0 (96)
13:22:17.433705 IP (tos 0x0, ttl 64, id 29105, offset 0, flags [none], proto UDP (17), length 57)
    192.168.1.70.53183 > 192.168.1.1.53: 33776+ AAAA? openwrt.org. (29)
13:22:17.433705 IP (tos 0x0, ttl 64, id 54214, offset 0, flags [none], proto UDP (17), length 57)
    192.168.1.70.63373 > 192.168.1.1.53: 51319+ A? openwrt.org. (29)
13:22:17.434177 IP (tos 0x0, ttl 64, id 18740, offset 0, flags [DF], proto UDP (17), length 85)
    192.168.1.1.53 > 192.168.1.70.53183: 33776 1/0/0 openwrt.org. AAAA 2a03:b0c0:3:d0::1a51:c001 (57)
13:22:17.434232 IP (tos 0x0, ttl 64, id 18741, offset 0, flags [DF], proto UDP (17), length 73)
    192.168.1.1.53 > 192.168.1.70.63373: 51319 1/0/0 openwrt.org. A 64.226.122.113 (45)
```

But it looks like that the iphone do not receive the answer packages.

trunneml, 2024-03-21:
Another detail. The two iphones are the only Wifi6 devices (802.11ax) all others on that SSID are Wifi5 (802.11ac).

brada4, 2024-03-21:
Can you ping phone from ap or connected device when it goes dark?

trunneml, 2024-03-21:
> Can you ping phone from ap or connected device when it goes dark?

No

trunneml, 2024-03-21:
I think this is a driver problem. I created an issue here: https://github.com/openwrt/mt76/issues/867

Delphius7, 2024-05-15:
@trunneml: Did you find a solution for this issue?

I have the same behavior with 2 iPhones and a GL-MT6000 router about every 3-4 days. The phones cannot be pinged and cannot ping anything on the network. Disable/Enable wifi solves the issue temporarily.
I am using the latest snapshot with kernel 6.6 but had the issue for weeks now even with kernel 5.

Still trying out the options you mentioned above but so far also unsuccessful.

trunneml, 2024-05-26:
As my iPhones are the only AX devices I own, I just turned of AX in OpenWRT. So the 5GHz Channel is running in AC mode. Since then no more problems.

peterneutron, 2024-06-07:
@trunneml I have the exact same issue using multiple Linksys MR8300 (ipq40xx, OpenWrt 23.05.3) with an iPhone 15 Pro which is an AX device but the APs are AC only. I already tried all recommended tips (DTIM, timeouts, rekeying, AQL, ANI etc.....) to no avail. This seems strictly limited to the 5 Ghz band.

Delphius7, 2024-09-12:
It seems like I found a workaround that works for me.
I have installed the custom build of pesa1234 (based on r27280-aa1c1b6e29) on my MT6000 and for almost 3 weeks i had no issues anymore. Before this I had the issues with 2 iPhones 3-5x per day.
https://github.com/pesa1234/MT6000_cust_build
https://forum.openwrt.org/t/mt6000-custom-build-with-luci-and-some-optimization-kernel-6-6-x/185241

I am not sure what exactly in this build solves the issue and I would be happy to have it solved in a clean snapshot build as well. Maybe someone with more knowledge can figure this out?

peterneutron, 2024-09-12:
Seems I forgot to unsubscribe to this issue, solved it by switching the whole network to UniFi hardware 😬. But deep down in my heart neckbeards still reign supreme 🥰

dlubawy, 2025-03-20:
I'm not using OpenWRT and it doesn't look like you're using VLANs, but this might be related to a bug I noticed with hostapd PMK caching and is seen in your logs `PMKID found from PMKSA cache eap_type=0 vlan=0`. I think devices getting authenticated through the cache are actually having their VLAN ID set to 0 instead of their actual VLAN. So they maintain a connection to the WiFi interface but packet forwarding breaks entirely as the station is communicating on an invalid VLAN now. 

Since Apple devices need PMK caching for the 4-way handshake on 5G, I've set a scheduled script to run `pmksa_flush` through `hostapd_cli` every 5 mins and it seems to help. Disabling the PMK cache altogether with 802.11r enabled and `disassoc_low_ack=1` on 5G seemed to work too as the Apple devices would fail the handshake, switch to 2.4G, auth ok, and then bss_transition kicked in to auth on 5G. But that 2nd option was more flakey.

olituks, 2025-05-18:
I am using 2 Google WiFi (Gale) devices that have been flashed and configured as Access Points (AP) with OpenWRT version 24.10.

All devices connected to these APs (on either 2.4 GHz or 5 GHz), except for one specific iPhone, connect and maintain a stable internet connection without issues.

For this iPhone, the behavior is different. Once the device goes into standby mode, it loses its internet connectivity. However, it remains shown as connected to the WiFi network (and the specific AP) in the OpenWRT interface and on the device itself.

I found a temporary workaround by manually configuring a public DNS server (8.8.8.8) directly in the WiFi settings for that specific SSID on the iPhone. With this configuration, the iPhone maintains a stable internet connection even in standby mode.

However, this workaround is not ideal as it bypasses the local network's DNS settings, including my AdGuard Home server, thus negating their purpose.

I am seeking a cleaner solution for the iPhone to correctly use the DNS settings provided by the network (either via DHCP from the OpenWRT AP or another DHCP server on the network).

Any suggestions on the potential cause for this behavior or specific OpenWRT configurations I should check/modify to resolve this issue?

brada4, 2025-05-18:
Please add ios version tags, ie whether it is old or new

olituks, 2025-05-18:
iOS 18.5 (22F76)

brada4, 2025-05-18:
Thats like 3 identical phones and one glitching?

olituks, 2025-05-19:
Hello, no. In fact, on my network, I have 4 Android devices and one iPhone. 
The DNS issue is only with the iPhone.
To have the iPhone fully working, I should set up manual DNS for my network SSID, with the internal DNS server IP plus a public one.