openwrt/openwrt #14832: Device Crash / reboot when a mobile device roam between 2.4 & 5 GHz WLAN with same SSID

momothefox, 2024-03-08:
### Describe the bug

**Hardware Configuration:**
Hanyang Digitech Co., Ltd.
MSIP-CMM-HYD-HYC-G920
CJ-Hello HYC-G920

SoC : MediaTek MT7621AT
RAM : 256M (SK hynix H5TQ2G63FFR)
FLASH : 16MB (Winbond W25Q128BV)
WiFi : MediaTek MT7602EN bgn 2SS
WiFi : MediaTek MT7612EN nac 2SS

```
		bdinfo: partition@100000 {
			label = "bdinfo";
			reg = <0x100000 0x40000>;
			read-only;
		};

```

```
&pcie0 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&bdinfo 0x0>;
	};
};

```

**Software Configuration:**
Previously running custom builds of OpenWRT (versions 22.03.0 and 23.05.2)
Currently running a compiled snapshot from 9-2-2024

**Observations:**

**Reboot Issue:**
Devices experienced multiple reboots on versions 22.03.0 and 23.05.2.
Reboots have decreased with the current compiled snapshot compared to previous versions.

**Roaming Trigger:**
Roaming between 2.4GHz and 5GHz WLANs occurs just before crashes, with a device transitioning between the two frequencies.
Both WLAN interfaces have the same SSID, leading to potential conflicts during roaming.

**Log Analysis:**
Logs indicate crashes occurring shortly after a device roams between WLANs.
Logs from the previous crashes are no longer present in the current snapshot from 9-2-2024.

**Configuration:**
**WiFi Configuration:**
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'HT20'
	option band '2g'
	option channel '1'
	option country 'US'
	option cell_density '3'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option encryption 'none'
	option ssid 'WiFi'

[... 245 lines cut ...]
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_TARGET_PREINIT_DISABLE_FAILSAFE=y
CONFIG_USE_LLVM_BUILD=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/snapshots"
CONFIG_VERSION_SUPPORT_URL=""
# CONFIG_IB is not set
# CONFIG_MAKE_TOOLCHAIN is not set
# CONFIG_SDK is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-08:
You should not doubt usability of roaming between bands. Most movile phones will keep sockets connected while roaming to extend range like considering aps with same name same 'network'

Subjective not causal suggestion is to have either xt/ipt xor nf/nft packet processing modules for firewall, ie iptables-nft can program xt modules via xt_bridge. Seems you use -legacy, thus try a clean installation and remove all nf/nft modules no not have nft flow offload mixing in. They dont help iptables-legacy at all.

momothefox, 2024-03-09:
@brada4 
thank you for your contribution, iptables-zz-legacy is for coovachilli to work.
and i have other devices that uses different wireless driver never faced this issue.
such as DW02-412H, running same configuration in build and every thing. never had reboot because of roaming.

i collected large amount of logs and analyze it. 
it is a certain behavior from certain mobiles.
 
and as my little little knowledge it is related to either mt76 or mt7621 or both.

brada4, 2024-03-09:
There is no explicit nft flow package selected, but one shows in backt$ace. i suspect something else pulled it as a dependency and added some fancy default nftables rule.
really check resulting packageson the target and uninstalling kmod-nf-* kmod-nft-* try to figure out what ipt/xt components need to be selected to satisfy whatever dependency of that other package.

momothefox, 2024-03-10:
@brada4 
Yes you are correct, i checked back then and found packages that depends on nft but i cannot recall them now.
in 2 days i will be back and test what you refer to.
i did not notice this before now 

```
kernel: [ 5692.616944] Modules linked in: pppoe ppp_async nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt76_connac_lib mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ip_tables cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat tun nls_iso8859_6 nls_cp864 sha512_generic seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic

```

but still i don't understand why ath79 device works with the same setup with no problem

brada4, 2024-03-10:
Come back if you discover excess dependencies or conflicts.

trunneml, 2024-03-18:
I needed this patch. https://github.com/openwrt/openwrt/pull/14083
without it mt76 crashes.

momothefox, 2024-03-18:
@trunneml i compiled my own snapshot and crashes are much less now, but still happening
@brada4  offered an approach for removing nft totally but there are dependencies i cannot remove them all, i removed what i could  

```
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_hanyang_hyc-g920=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_COOVACHILLI_JSONINTERFACE=y
CONFIG_COOVACHILLI_MINIPORTAL=y
CONFIG_COOVACHILLI_NOSSL=y
CONFIG_COOVACHILLI_REDIR=y
CONFIG_HASERL_shell_lua=y
CONFIG_HASERL_shell_luac=y
CONFIG_HASERL_with_lua=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IMAGEOPT=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="builddomain"
CONFIG_KERNEL_BUILD_USER="buildhost"
CONFIG_KERNEL_WERROR=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
# CONFIG_PACKAGE_arptables-nft is not set
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_coova-chilli=y
CONFIG_PACKAGE_curl=y
# CONFIG_PACKAGE_ebtables-nft is not set
CONFIG_PACKAGE_firewall=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_haserl=y
CONFIG_PACKAGE_ip-tiny=y
# CONFIG_PACKAGE_ip6tables-extra is not set
# CONFIG_PACKAGE_ip6tables-mod-nat is not set
# CONFIG_PACKAGE_ip6tables-nft is not set
# CONFIG_PACKAGE_ip6tables-zz-legacy is not set
# CONFIG_PACKAGE_iptables-mod-checksum is not set
# CONFIG_PACKAGE_iptables-mod-cluster is not set
# CONFIG_PACKAGE_iptables-mod-clusterip is not set
# CONFIG_PACKAGE_iptables-mod-conntrack-extra is not set
# CONFIG_PACKAGE_iptables-mod-extra is not set
# CONFIG_PACKAGE_iptables-mod-filter is not set
# CONFIG_PACKAGE_iptables-mod-hashlimit is not set
# CONFIG_PACKAGE_iptables-mod-ipopt is not set
# CONFIG_PACKAGE_iptables-mod-iprange is not set
[... 193 lines cut ...]
CONFIG_VERSION_CODE=""
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_DIST="WitWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/snapshots"
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_WOLFSSL_HAS_NO_HW=y
# CONFIG_IB is not set
# CONFIG_MAKE_TOOLCHAIN is not set
CONFIG_PACKAGE_jansson=y
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libnftnl=y
# CONFIG_SDK is not set

```

i will dispatch this build for testing and i will back with feedback.

brada4, 2024-03-18:
Can you identify who pulled nf-conntrack module? It should be like same nf-conntrack or xt_conntrack via xt_bridge, wrong to have both required at once.

momothefox, 2024-03-18:
@brada4 

[image]

[image]

brada4, 2024-03-18:
You had modules from kmod-nft-offload in backtrace....

momothefox, 2024-03-22:
@brada4 
Same thing . reboots occurs 
in some devices lowering transmission power made reboots to stop or be be much less than before.
i opened an issue [https://github.com/openwrt/mt76/issues/858](url)
before i open this issue here, but i got no feedback at all.

momothefox, 2024-04-14:
> Can you identify who pulled nf-conntrack module? It should be like same nf-conntrack or xt_conntrack via xt_bridge, wrong to have both required at once.
> 
> You fount it in mt76 driver, so not essential to dig why you have 2 conntracks at once.

[image]

brada4, 2024-04-14:
Worth leaving log timestamps over at mt76 report....
You should carefully move to nftables via iptables-nft. 
run `nft list ruleset | grep xt` and add missing nftables modules as per map:
https://wiki.nftables.org/wiki-nftables/index.php/Supported_features_compared_to_xtables
Normally there should be no mix of nft and xt modules in same kernel.

momothefox, 2024-04-14:
can coova work without iptables ? all this because i have to use old ipt for coova.

brada4, 2024-04-14:
No idea about that. You need all working filtering modules in one or another side of x_tables bridge.