openwrt/openwrt #14837: Why there no ipq60xx builds?

YaeMashiro, 2024-03-09:
### Describe the bug

https://downloads.openwrt.org/snapshots/targets/qualcommax/
https://buildbot.openwrt.org/images/#/builders
[image]

### OpenWrt version

snapshots

### OpenWrt release

snapshots

### OpenWrt target/subtarget

qualcommax/ipq6000

### Device

axt1800

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2024-03-09:
https://github.com/openwrt/openwrt/commit/23deb4ac90e16277e9d779856e77d0ed22142bc5

> WiFi support still has to be handled and correctly fix hence this is currently marked as source-only to have a solid base to progress on correct support of this and hope Upstream QUIC publish newers ath11k drivers for this SoC.

YaeMashiro, 2024-03-09:
> [23deb4a](https://github.com/openwrt/openwrt/commit/23deb4ac90e16277e9d779856e77d0ed22142bc5)
> 
> > WiFi support still has to be handled and correctly fix hence this is currently marked as source-only to have a solid base to progress on correct support of this and hope Upstream QUIC publish newers ath11k drivers for this SoC.

Thx

closed 2024-03-09 by YaeMashiro