openwrt/openwrt #14841: PF_RING-8.0.0 error: invalid application of 'sizeof' to incomplete type 'char[]'

hexsen929, 2024-03-10:
### Describe the bug

`make[2]: Entering directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/feeds/packages/libs/libpfring'
rm -f /home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/.built
touch /home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/.built_check
make -C "/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.150" KERNEL_DIR="/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.150" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" M="/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel" EXTRA_CFLAGS=" -I/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel" modules
make[3]: Entering directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.150'
  CC [M]  /home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel/pf_ring.o
/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel/pf_ring.c: In function 'ring_bind':
/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel/pf_ring.c:5608:19: error: invalid application of 'sizeof' to incomplete type 'char[]'
 5608 |   char name[sizeof(sa->sa_data)+1];
      |                   ^
/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel/pf_ring.c:5620:35: error: invalid application of 'sizeof' to incomplete type 'char[]'
 5620 |   memcpy(name, sa->sa_data, sizeof(sa->sa_data));
      |                                   ^
/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel/pf_ring.c:5608:8: error: unused variable 'name' [-Werror=unused-variable]
 5608 |   char name[sizeof(sa->sa_data)+1];
      |        ^~~~
/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel/pf_ring.c:5628:1: error: control reaches end of non-void function [-Werror=return-type]
 5628 | }
      | ^
cc1: all warnings being treated as errors
make[4]: *** [scripts/Makefile.build:289: /home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel/pf_ring.o] Error 1
make[3]: *** [Makefile:1908: /home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/kernel] Error 2
make[3]: Leaving directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.150'
make[2]: *** [Makefile:92: /home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.0.0/.built] Error 2
make[2]: Leaving directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/feeds/packages/libs/libpfring'
time: package/feeds/packages/libpfring/compile#1.03#0.19#1.20
    ERROR: package/feeds/packages/libpfring failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/libpfring/compile] Error 1
make[1]: Leaving directory '/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt'
make: *** [/home/runner/work/OpenWrt-K/OpenWrt-K/workspace/openwrt/include/toplevel.mk:232: package/compile] Error 2`

### OpenWrt version

openwrt-23.05

### OpenWrt release

openwrt-23.05

### OpenWrt target/subtarget

openwrt-23.05

### Device

debian12 x86

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hexsen929, 2024-03-10:
> Invalid Target/Subtarget reported. `openwrt-23.05` Is this from a supported device?

from the Branches openwrt-23.05

remittor, 2024-03-10:
I confirm the presence of the same error in snapshots: https://github.com/openwrt/openwrt/pull/14766#issuecomment-1985119265

Error log: https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/

kmod founded: https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/kmods/6.1.79-1-623c9e94575bfb1c74c4f7e8e8d139f9/kmod-pf-ring_6.1.79+8.4.0-1_aarch64_cortex-a53.ipk

But `kmod-pf-ring` for 6.1.80 not founded.

JimMatthew, 2024-03-10:
I am having the same issue when trying to build from main x86/64 with my custom config.

brada4, 2024-03-11:
Upgrade to openwrt.
You are using this fork and concealing the version tags:
https://github.com/chenmozhijin/OpenWrt-K/releases
Would be more productive to report to them, then if they indicate so fill the report against clean Openwrt in ../packages

JimMatthew, 2024-03-11:
> Upgrade to openwrt. You are using this fork and concealing the version tags: https://github.com/chenmozhijin/OpenWrt-K/releases Would be more productive to report to them, then if they indicate so fill the report against clean Openwrt in ../packages

this is happening in openwrt too, see the error logs below
https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/libpfring/

brada4, 2024-03-11:
Yes, you need to update libpfring **package** for new kernel. It has changed upstream in meantime to 8.6.1.

brada4, 2024-03-11:
It compiles without fortified headers, if you need functionality.

JimMatthew, 2024-03-11:
Thanks for the help. 
I tried to update the makefile but it looks like it is failing to apply one of the patches. I will keep messing around with this, I don't have much experience with this but ill see if I can figure it out. Not too urgent, my current build is running great 

Applying /home/jam/openwrt/feeds/packages/libs/libpfring/patches/0001-fix-cross-compiling.patch using plaintext:

patching file userland/configure
Hunk #1 succeeded at 3873 (offset 5 lines).
Hunk #2 succeeded at 3885 (offset 5 lines).
Hunk #3 succeeded at 3899 (offset 5 lines).

Applying /home/jam/openwrt/feeds/packages/libs/libpfring/patches/002-implement-probabilistic-sampling.patch using plaintext:
patching file kernel/linux/pf_ring.h
Hunk #1 FAILED at 1310.
1 out of 1 hunk FAILED -- saving rejects to file kernel/linux/pf_ring.h.rej
patching file kernel/pf_ring.c
Hunk #1 succeeded at 3716 (offset 21 lines).
Hunk #2 FAILED at 3825.
Hunk #3 FAILED at 4181.
Hunk #4 FAILED at 7977.
3 out of 4 hunks FAILED -- saving rejects to file kernel/pf_ring.c.rej
Patch failed!  Please fix /home/jam/openwrt/feeds/packages/libs/libpfring/patches/002-implement-probabilistic-sampling.patch!
make[2]: *** [Makefile:97: /home/jam/openwrt/build_dir/target-x86_64_musl/linux-x86_64/PF_RING-8.6.1/.prepared_890351f362080501a400c2ed7fb9c86f_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[2]: Leaving directory '/home/jam/openwrt/feeds/packages/libs/libpfring'

closed 2024-03-12 by hexsen929