openwrt/openwrt #14843: XDR6088 has serious performance issues, possibly caused by the driver

pupboss, 2024-03-10:
### Describe the bug

The six ports are: `eth1, lan5, lan1, lan2, lan3, lan4` respectively. And I use lan4 as the WAN port.

When eth1 my nas download and upload simultaneously at around 50MB/s, there is almost no speed for ALL other devices for intranet comminication.

[image]

[image]

Above is the iperf result between my server and my laptop. Same result either I use cable or wifi. My server's performance is sufficient and will not become a bottleneck.

Moreover if I run a iperf server on openwrt instead, the speedtest works like normal:

[image]

In earlier research, I found that `eth1` and the two WiFi `radio` are CPU direct pass-through ports, while the other four gigabit ports `(lan1-lan4)` and one 2.5G port `(lan5)` are expanded.

By right the wifi devices to eth1 should be very fast, and there is no bandwidth limitation like other ports.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

TP-Link TL-XDR6088

### Image kind

Official downloaded image

### Steps to reproduce

1. lan4 as WAN, and eth1 connect to NAS
2. Let the NAS eat all bandwidth
3. Access NAS from intranet (either wifi or any cable port)

### Actual behaviour

Download speed less than 100Mbps, and upload speed less than 300Mbps

But if I stop the NAS data transmission, speed back to normal. At least 800Mbps for both direction.

### Expected behaviour

NAS ethernet speed is 2.5Gbps RX + 2.5Gbps TX, assume the transmission to WAN eating 500Mbps RX + 500Mbps TX.

There are still around 2Gbps RX and 2Gbpx TX bandwidth left.

### Additional info

I did some search before installing openwrt, seems the driver for MT chipset is not optimal in openwrt.

Moreover, I ever run the same test with TPLINK official firmware, there was no such issue, below I provide some evidences.

[image]

from the picture you can see the NAS uploading at 600Mbps for many days, however in that period intranet devices were not impacted. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pupboss, 2024-03-10:
One more round of testing, testing environment as below:
1. NAS and WAN still running
2. My lap top start a iperf3 server with both wifi (192.168.10.9) and cable (192.168.10.33)
3. My phone speedtest with my laptop

[image]

Above shows that with cable connection, my phone to my laptop usually got disruption

[image]

Laptop server via wifi performs much better

[image]

Above as the control group, around 1.4Gbps speed when speedtest between phone and router (192.168.10.1)

---

I believe the screenshots above are sufficient to prove certain things. For example, communication between Wi-Fi devices under the same Wi-Fi radio seems relatively stable. Also, accessing wired devices from Wi-Fi devices does indeed often experience interruptions, for either my NAS with high load and laptop with no workload.

Additionally, my Wi-Fi signal coverage is very good, with no interference in the surroundings. The third screenshot proves that it can reach up to 1.4Gbps at maximum.

brada4, 2024-03-10:
You need to enable kernel threads and detailed CPU in htop.
If all NAPI interrupts fire on one CPU you need to enable packet steering (or alternatively irqbalance on ARM cpu)
There is a gigabit port to host backing 4x lan ports. If you pump something from NAS the remaining bandwidth is for internet.

brada4, 2024-03-11:
Check /proc/interrupts whether you get multiple ethernet interrupts, if so irqbalance may be better than steering.
Can you get more than 600/600 directly connected to NAS box? Maybe it just screams for gigabit port?

pupboss, 2024-03-11:
```
root@TL-XDR6088:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
 11:    6381248    5955064    5756896    6028400     GICv3  30 Level     arch_timer
 14:          0          0          0          0     GICv3 142 Level     wdt_bark
 25:          0          0          0          0   mt-eint   9 Edge      keys
 26:          0          0          0          0   mt-eint  10 Edge      keys
 27:          0          0          0          0   mt-eint  11 Edge      keys
 82:          7          0          0          0   mt-eint  66 Level     mt7530
117:          0          0          0          0     GICv3 148 Level     10320000.crypto
118:          0          0          0          0     GICv3 149 Level     10320000.crypto
119:          0          0          0          0     GICv3 150 Level     10320000.crypto
120:          0          0          0          0     GICv3 151 Level     10320000.crypto
121:         14          0          0          0     GICv3 155 Level     ttyS0
122:      74373          0          0          0     GICv3 172 Level     1100a000.spi
123:          0          0          0          0     GICv3 205 Level     xhci-hcd:usb1
130:  518735265          0          0          0     GICv3 229 Level     15100000.ethernet
131:  301232840          0          0          0     GICv3 230 Level     15100000.ethernet
133:   43970612          0          0          0     GICv3 245 Level     mt7915e
137:          0          0          0          0    mt7530   0 Edge      mt7530-0:00
138:          0          0          0          2    mt7530   1 Edge      mt7530-0:01
139:          0          0          0          0    mt7530   2 Edge      mt7530-0:02
140:          0          0          0          5    mt7530   3 Edge      mt7530-0:03
IPI0:    252092     284364    1002896     319454       Rescheduling interrupts
IPI1:  82591244  311936802  291519495  284864460       Function call interrupts
IPI2:         0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
IPI4:         0          0          0          0       Timer broadcast interrupts
IPI5:         0          0          0          0       IRQ work interrupts
IPI6:         0          0          0          0       CPU wake-up interrupts
```

Thanks Andrew, I am not familiar with the proc/interrupts, could you please advice

One more thing to mention, to test the more than 600/600 stream I can only do it two weeks later back from business travel because I am now at the airport :D. If irq performs better I can install it remotely as I configured VPN to homelab access.

brada4, 2024-03-11:
irqbalance is supposed to route these on different CPU cores on supported system, you will need steering anyway since there are only two of these. 
```
130:  518735265          0          0          0     GICv3 229 Level     15100000.ethernet
131:  301232840          0          0          0     GICv3 230 Level     15100000.ethernet
```
Since now load is evenly balanced it may be just no-op in the end performance-wise even if it works out.

brada4, 2024-03-11:
@dangowrt are you able to set steering default for this device with OP-d clear indication network performance is sub-par by default?

pupboss, 2024-03-11:
Okay got it. I havn't installed irqbalance yet. And I won't try installing irqbalance in the near future as steering is good enough, the performance is very close to what it was when idle.

Something to share:

```
[ ID][Role] Interval           Transfer     Bitrate
[  5][TX-C]   0.00-1.01   sec   242 MBytes  2.02 Gbits/sec
[  7][RX-C]   0.00-1.01   sec  74.0 MBytes   618 Mbits/sec
[  5][TX-C]   9.01-10.01  sec   251 MBytes  2.11 Gbits/sec
[  7][RX-C]   9.01-10.01  sec  74.9 MBytes   628 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.01  sec  2.39 GBytes  2.05 Gbits/sec                  sender
[  7][RX-C]   0.00-10.01  sec   740 MBytes   620 Mbits/sec    0             sender
```

This is the speedtest result a few weeks ago with official firmware and with no load on the router. Can see that the maximum tx rate between the two 2.5G ports (eth1 and lan5) was around 2.7Gbps. (lan5 is extended by MT7531A via a HSGMII lane from MT7531)

Moreover, if I run iperf3 on the router, the bandwidth between lan5 (MT7531A) and CPU is around 4Gbps

```
[ ID][Role] Interval           Transfer     Bitrate
[  5][TX-C]   6.00-7.01   sec   231 MBytes  1.94 Gbits/sec
[  7][RX-C]   6.00-7.01   sec   276 MBytes  2.31 Gbits/sec
[  5][TX-C]   7.01-8.00   sec   241 MBytes  2.03 Gbits/sec
[  7][RX-C]   7.01-8.00   sec   275 MBytes  2.31 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.01  sec  2.41 GBytes  2.07 Gbits/sec                  sender
[  7][RX-C]   0.00-10.01  sec  2.29 GBytes  1.96 Gbits/sec    3             sender
```

brada4, 2024-03-11:
Glad to hear. No need to experiment further.