openwrt/openwrt #14845: nftables issue with rate limiting

hayzamjs, 2024-03-10:
### Describe the bug

I have this table present in /usr/share/nftables.d/ruleset-post/0-dqos.nft (I also tried placing it in /usr/share/nftables.d/ruleset-pre/0-dqos.nft`

```
table bridge dqos
flush table bridge dqos

table bridge dqos {
                chain download {
                        type filter hook postrouting priority 0; policy accept;
                        ether daddr 98:0D:AF:45:29:05 accept
                        ether daddr 3C:7C:3F:23:5C:50 accept
                        limit rate over 3125 kbytes/second drop
                }

                chain upload {
                        type filter hook prerouting priority 0; policy accept;
                        ether saddr 98:0D:AF:45:29:05 accept
                        ether saddr 3C:7C:3F:23:5C:50 accept
                        limit rate over 3125 kbytes/second drop
                }
}
```

What I want to achieve is for 3C:7C:3F:23:5C:50 and 98:0D:AF:45:29:05 to be immune from the rate limit limit rate over 3125 kbytes/second drop but for some reason every client on the network including the "whitelisted" ones is rate limited to 25 mbits/second (or 3125 kbytes/second).

### OpenWrt version

r25007-c8d6594bbb

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Banana Pi R3

### Image kind

Self-built image

### Actual behaviour

Mac addresses that are "whitelisted" are also being rate limited

### Expected behaviour

Mac addresses that are "whitelisted" **should not** be rate limited

### Additional info

My LAN device and interface:

```
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'sfp2'
        list ports 'lan1'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        list ipaddr '192.168.16.1/24'
        list dns '192.168.16.1'
        option ip6assign '64'
        list ip6class 'local'
        list ip6class 'wan6'
```

One of the clients that I've allowed is on WiFi and the other is connected via an ethernet cable, both are rate limited along with other clients on the network

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-10:
put `ether Xaddr counter` after `rate counter drop `
you can add counters before accept actions too. 
Better place would be forward
then saddr {a,b} daddr {a,b} accept.

brada4, 2024-03-10:
Note that bridge firewall pulls all traffic via cpu port, so you quarter total bw...

hayzamjs, 2024-03-11:
> Note that bridge firewall pulls all traffic via cpu port, so you quarter total bw...

Can you provide a source for this information? 

```
table bridge dqos
flush table bridge dqos

table bridge dqos {
                chain download {
                        type filter hook postrouting priority 0; policy accept;
                        ether daddr 98:0D:AF:45:29:05 accept
                        ether daddr 3C:7C:3F:23:5C:50 limit rate over 18750 kbytes/second drop
                        #limit rate over 3125 kbytes/second drop
                }

                chain upload {
                        type filter hook prerouting priority 0; policy accept;
                        ether saddr 98:0D:AF:45:29:05 accept
                        ether saddr 3C:7C:3F:23:5C:50 limit rate over 18750 kbytes/second drop
                        #limit rate over 3125 kbytes/second drop
                }
}
```

The rule above gives me 150 mbits/second on the device with the mac `3C:7C:3F:23:5C:50` (my total is 300 mbits/second) so a quarter would be 75?

> put `ether Xaddr counter` after `rate counter drop ` you can add counters before accept actions too. Better place would be forward then saddr {a,b} daddr {a,b} accept.

I'm not exactly sure what you meant here but I tried:

```
table bridge dqos
flush table bridge dqos

table bridge dqos {
                chain download {
                        type filter hook postrouting priority 0; policy accept;
                        ether daddr { 98:0D:AF:45:29:05, 3C:7C:3F:23:5C:50 } counter accept
                        limit rate over 3125 kbytes/second counter drop
                }

                chain upload {
                        type filter hook prerouting priority 0; policy accept;
                        ether saddr { 98:0D:AF:45:29:05, 3C:7C:3F:23:5C:50 } counter accept
                        limit rate over 3125 kbytes/second counter drop
                }
}
```

After loading ^ into the firewall, I run a speed test on both the "whitelisted" MACs and I am still rate limited to 25 mbits/second and the counters look like this:

```
table bridge dqos {
        chain download {
                type filter hook postrouting priority 0; policy accept;
                ether daddr { 3c:7c:3f:23:5c:50, 98:0d:af:45:29:05 } counter packets 69946 bytes 59355104 accept
                limit rate over 3125 kbytes/second counter packets 2639 bytes 3951421 drop
        }

        chain upload {
                type filter hook prerouting priority 0; policy accept;
                ether saddr { 3c:7c:3f:23:5c:50, 98:0d:af:45:29:05 } counter packets 56539 bytes 60205676 accept
                limit rate over 3125 kbytes/second counter packets 4678 bytes 6996892 drop
        }
}
```

brada4, 2024-03-11:
I mean to count spills:
```
table bridge dqos {
        chain download {
                type filter hook forward priority 0; policy accept;
                ether daddr { 3c:7c:3f:23:5c:50, 98:0d:af:45:29:05 } saddr { 3c:7c:3f:23:5c:50, 98:0d:af:45:29:05 } counter accept
                ether daddr { 3c:7c:3f:23:5c:50, 98:0d:af:45:29:05 } saddr { 3c:7c:3f:23:5c:50, 98:0d:af:45:29:05 } counter
                limit rate over 3125 kbytes/second counter drop
                counter
        }
```

hayzamjs, 2024-03-11:
```
table bridge dqos {
        chain download {
                type filter hook forward priority 0; policy accept;
                ether daddr 3c:7c:3f:23:5c:50 ether saddr 3c:7c:3f:23:5c:50 counter packets 0 bytes 0 accept
                ether daddr 3c:7c:3f:23:5c:50 ether saddr 3c:7c:3f:23:5c:50 counter packets 0 bytes 0
                limit rate over 3125 kbytes/second counter packets 0 bytes 0 drop
                counter packets 3 bytes 2119
        }

        chain upload {
                type filter hook prerouting priority 0; policy accept;
                ether daddr 3c:7c:3f:23:5c:50 ether saddr 3c:7c:3f:23:5c:50 counter packets 0 bytes 0 accept
                ether daddr 3c:7c:3f:23:5c:50 ether saddr 3c:7c:3f:23:5c:50 counter packets 0 bytes 0
                limit rate over 3125 kbytes/second counter packets 1534 bytes 2166695 drop
                counter packets 26839 bytes 22069300
        }
}
```

My hunch is that it never really gets past the upload prerouting chain?

jow-, 2024-03-11:
Do those rules even make sense, logically? In all cases you're testing whether the source and destination mac address are identical, this wouldn't be the case for normal routed ethernet frames. Did you mean to match source *or* destination mac address = XXX ? If so, you'll need separate rules.

brada4, 2024-03-11:
You need just one forward hook.

```
table bridge dqos {
        chain whatever {
                type filter hook forward priority 0; policy accept;
                ether saddr {a,b} counter accept
                ether saddr {a,b} counter 
                ether daddr {a,b} counter accept
                ether daddr {a,b} counter
                limit rate over 3125 kbytes/second counter  drop
                counter
        }
```

hayzamjs, 2024-03-11:
> You need just one forward hook.
>
> ```
> table bridge dqos {
>         chain whatever {
>                 type filter hook forward priority 0; policy accept;
>                 ether saddr {a,b} counter [...]

"whitelisting" works with this, but on the non-whitelisted devices upload is not rate limited just download.

@jow- is right I was for some reason trying to substitute for a logical OR operator with just specifying daddr and saddr in the same rule it's pretty dumb, with separate rules as in my original case:

```
table bridge dqos
flush table bridge dqos

table bridge dqos {
                chain download {
                        type filter hook postrouting priority 0; policy accept;
                        ether daddr 98:0D:AF:45:29:05 accept
                        ether daddr 3C:7C:3F:23:5C:50 accept
                        limit rate over 3125 kbytes/second drop
                }

                chain upload {
                        type filter hook prerouting priority 0; policy accept;
                        ether saddr 98:0D:AF:45:29:05 accept
                        ether saddr 3C:7C:3F:23:5C:50 accept
                        limit rate over 3125 kbytes/second drop
                }
}
```

Is valid but the blanket rate limit `limit rate over 3125 kbytes/second drop` applies to all packets which doesn't make much sense either..this is not a bug (as I originally thought) so I'm closing my issue here. For anyone coming across the issue the only way to solve it is either by specifying the interfaces for up/down in their respective chains or by specifying limits for **ALL** macs/ips manually.

closed 2024-03-11 by hayzamjs

brada4, 2024-03-11:
Yes, my rule was to confirm a and b intercommunicate unrestricted while rest of network shares slow cap.

More adequate would be to add two acceptable/subscribed rates tbf-s to the guest line and let rest run wild.