openwrt/openwrt #14846: Kernel panic (mt7621)

RydjGit, 2024-03-10:
### Describe the bug

Device : ZBT-WG-1602-32M 
I have a mini-PCIE to USB 3.0 adapter connected to the device ([Renesas uPD720201 Main chip](https://www.aliexpress.us/item/1005005654236296.html)) the vesrion with 4 pin power connector.

the device works as expected and USB devices connected to it all get recognised (given driver is present) but I decided to make a mini Network storage system using these ports thus I connected 1 x SSD and 2 x 2.5" HDD using sata-USB3 adapters and shared them using Samba. they all work but I experinance random and sudden reboots when accessing these discs mainly *** WHEN COPYING AND WRITING LARGE NUMBER OF SMALL FILES OR USING TORRENT PROGRAMS LIKE ARIA2 OR RTORRRENT***

possible causes : 
1- Power insuffecincy :  (excluded)
    The router comes with a 12V 2 Amp adapter which I replaced with a 3 amp one.
    The Mini-PCIE power connector gets it 5 volts from a buck converter module which by itself is connected to another 12 V 2 amp adapter.

2- CPU thermal reset :  (excluded)
     unfortunotaly I cannot get the temp readings fot MT7621 soc but it has a large heatsink mounted.
     to confirm I did a stress-ng mediated cpu stress test and the system is stable as a rock with 100% cpu load .

the dmesg log sometimes does not even finish the Kernel Panic messege, here it is:

```[ 2649.080916] CPU 3 Unable to handle kernel paging request at virtual address 000051a0, epc == 8016cc04, ra == 801ab57c
[ 2649.091647] Oops[#1]:
[ 2649.093957] CPU: 3 PID: 36 Comm: kcompactd0 Not tainted 5.15.137 #0
[ 2649.100219] $ 0   : 00000000 00000001 00005150 fffffffc
[ 2649.105459] $ 4   : 80df896c 00000000 00000008 0000000c
[ 2649.110716] $ 8   : 000118e1 00000000 00000000 00000000
[ 2649.115944] $12   : f0000080 00004d60 00000000 00000000
[ 2649.121196] $16   : 80df896c 0000000c 00000000 81555e30
[ 2649.126426] $20   : 00000000 00000004 00000000 00000000
[ 2649.131679] $24   : c7af2a80 800107a8
[ 2649.136908] $28   : 81554000 81555cd8 0001c400 801ab57c
[ 2649.142160] Hi    : 00000000
[ 2649.145032] Lo    : 00000000
[ 2649.147924] epc   : 8016cc04 0x8016cc04
[ 2649.151747] ra    : 801ab57c 0x801ab57c
[ 2649.155571] Status: 1100fc03 KERNEL EXL IE
[ 2649.159777] Cause : 40800008 (ExcCode 02)
[ 2649.163765] BadVA : 000051a0
[ 2649.166638] PrId  : 0001992f (MIPS 1004Kc)
[ 2649.170743] Modules linked in: sdhci_pltfm sdhci uas ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda usb_storage sd_mod scsi_mod scsi_common pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat ledtrig_usbport sha512_generic seqiv jitterentropy_rng drbg hmac cmac ext4 mbcache jbd2 mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[ 2649.257684] Process kcompactd0 (pid: 36, threadinfo=71d8f156, task=1b93a7a9, tls=00000000)
[ 2649.265979] Stack : 81555cd8 81555cd8 80b0e584 00000003 81435000 801ab57c 81022f64 00005c00
[ 2649.274393]         80811920 8071b760 00007400 00018110 80b0e588 80df896c 0001c003 8016df80
[ 2649.282801]         80880000 81022f60 00000001 00000001 808123d8 80df8900 80a00000 80df891c
[ 2649.291168]         20831080 38e38e39 00000000 00005c00 00007400 8014d034 80b0e584 00000000
[ 2649.299571]         20831080 20002000 81555e30 80df8900 0000000c 0001c800 0001c400 00000000
[ 2649.307969]         ...
[ 2649.310442] Call Trace:
[ 2649.310455] [<801ab57c>] 0x801ab57c
[ 2649.316375] [<8071b760>] 0x8071b760
[ 2649.319885] [<8016df80>] 0x8016df80
[ 2649.323362] [<8014d034>] 0x8014d034
[ 2649.326853] [<8016f99c>] 0x8016f99c
[ 2649.330360] [<8016e4ec>] 0x8016e4ec
[ 2649.333842] [<8016e4ec>] 0x8016e4ec
[ 2649.337346] [<8016cba8>] 0x8016cba8
[ 2649.340839] [<8016e4ec>] 0x8016e4ec
[ 2649.344331] [<8016fe18>] 0x8016fe18
[ 2649.347839] [<800a3c3c>] 0x800a3c3c
[ 2649.351323] [<801700b0>] 0x801700b0
[ 2649.354807] [<8016fec8>] 0x8016fec8
[ 2649.358317] [<800778f4>] 0x800778f4
[ 2649.361792] [<8016fec8>] 0x8016fec8
[ 2649.365281] [<80051a6c>] 0x80051a6c
[ 2649.368788] [<8005192c>] 0x8005192c
[ 2649.372262] [<8005192c>] 0x8005192c
[ 2649.375748] [<8005192c>] 0x8005192c
[ 2649.379249] [<80002f98>] 0x80002f98

[ 2649.384228] Code: 00000000  10400008  8fbf0014 <8c420050> 10400005  00000000  8c420038  0002102b  03e00008

[ 2649.395657] ---[ end trace 2625775a7ef9d6a5 ]---
[ 2649.400298] Kernel panic - not syncing: Fatal exception
```
my only conclusion is that it is software related.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Zbtlink ZBT-WG1602-V04 (32M)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-10:
What is connected to the USB port? What is this long chain of USB drivers installed?

RydjGit, 2024-03-10:
External 2.5" SSD using a SATA-to-USB3.0 adapter , regarding drivers : I belive the used ones are : ```kmod-usb-xhci-hcd , kmod-usb3, kmod-usb-storage, kmod-usb-storage-extras And 
kmod-usb-storage-uas
```

brada4, 2024-03-10:
Likely usb power draw problem. And you need powered usb hub.
Try without usb device first, check its power draw  on windows or linux pc, in usb2 and usb3 modes. 
There are drivers for 20 years eom mp3 players, they dont cause problems, but add no benefit either.

RydjGit, 2024-03-11:
OK here is my testing: (read write I/O)
1- USB 2.0 mode : no problems at all working fine
2- USB 3.0 mode : Large files -> rare Kernel panic
                              multiple small files - > crash every time and reproducible

Regarding power : the USB Devices dont get power from router PSU but from a totally seperate power adapter with 24 watt (overkill) cabapility . 

but I will change the BUCK converter to a more powerfull one and see if the crash goes away.
I will report back.

RydjGit, 2024-03-11:
changed the buck converter to a more powerfull one and the results are:
1- the crashes still happen on  largenumber of  small files read/write but it is much less frequent 

dmesg Log now reports the following: 
```
[  175.893946] sd 1:0:0:0: [sda] tag#15 uas_eh_abort_handler 0 uas-tag 2 inflight: OUT
[  175.901771] sd 1:0:0:0: [sda] tag#15 CDB: opcode=0x2a 2a 00 06 22 88 00 00 04 00 00
[  177.493904] sd 1:0:0:0: [sda] tag#21 uas_eh_abort_handler 0 uas-tag 3 inflight: OUT
[  177.501685] sd 1:0:0:0: [sda] tag#21 CDB: opcode=0x2a 2a 00 06 24 b8 48 00 04 00 00
[  182.933765] sd 1:0:0:0: [sda] tag#4 uas_eh_abort_handler 0 uas-tag 1 inflight: OUT
[  182.941447] sd 1:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 06 2b b0 00 00 02 e0 00
[  185.253769] sd 1:0:0:0: [sda] tag#9 uas_eh_abort_handler 0 uas-tag 7 inflight: OUT
[  185.261450] sd 1:0:0:0: [sda] tag#9 CDB: opcode=0x2a 2a 00 06 2f 21 48 00 03 a0 00
[  185.323811] scsi host1: uas_eh_device_reset_handler start
[  185.476063] usb 4-1: reset SuperSpeed USB device number 3 using xhci_hcd
[  185.516733] scsi host1: uas_eh_device_reset_handler success
```

brada4, 2024-03-11:
Just to try , not 101% solution:
Now check with htop (enable kernel threads and extended CPU display) if all load falls into one CPU, and some IRQs eventually get unhandled and land as ERR in /proc/interrupts. irqbalance may or may not help if so.
kmod-iosched-bfq may group small disk accesses and mask the problem.
also disabling UAS (blacklisting kmod) may or may not help

RydjGit, 2024-03-11:
ok thank you for the sugestions, here is the summery of test results:
1- irqbalanace install and enable -> no effect (random resets still happen ).
2- cat /proc/interrupts -> no ERR i.e zero
3- disabling UAS -> that improved system stabilty but:
      A- I lost access to my SSD.
      B- One external HDD drive uses "usb-storage" driver, Reading and writing to this HDD using the same USB 3 port was stable.
      C- Renabled UAS driver : Reading from SSD (mostly OK) , Writing to SSD (Crash).
4- test using another HDD using UAS driver : same result frequent resets (crashes).
5- all tests above are done with "kmod-iosched-bfq" installed.
I dont know what is causing this problem .

brada4, 2024-03-11:
- no USB power issues
- no fault outside SoC

Problem may be
- USB3 driver for SoC USB - what if you move other disk to USB2 port?
- UAS driver - i am confused why it does not work with usb-storage. If it works in USB2 mode it certainly should.

For more detail you may install htop and in there enable kernel thread and cpu detail display, it could give some insight, but I suspect 100% one CPU core busy with USB interrupt.

RydjGit, 2024-03-11:
[image]

This is taken while writingto a UAS dependant drive

brada4, 2024-03-13:
You can help with backtrace though....

RydjGit, 2024-03-15:
as I am in the process of deploying an Openwrt build enviroment here is some update:
I decided to flash an openwrt fork called Rooter that has my router in their support list.
They didnt include ` kmod-usb-storage-uas kmod-usb-storage-extra` in their destributed build for my device only `kmod-usb-storage` and to my surprise the SSD and HDD that didnt work with vanilla Openwrt unluess the above mentioned kmods were installed, are working nicly and further the system is stable as  rock. 
conclusion: definitly a sofware issue 

some info on installed Rooter:

```
root@ROOter:/etc# cat openwrt_release
DISTRIB_ID="OpenWrt"
DISTRIB_RELEASE="23.05.0-rc2"
DISTRIB_REVISION=" "
DISTRIB_CODENAME=""
DISTRIB_TARGET="ramips/mt7621"
DISTRIB_DESCRIPTION="GoldenOrb_2024-03-13 ( OpenWrt 23.05.0-rc2 )"
```

```
root@ROOter:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 2: Dev 2, If 0, Class=, Driver=usb-storage, 5000M  (My 1 terabyte SSD) (working with usb-storage driver)
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M  (2x  4g modems )
    |__ Port 1: Dev 4, If 0, Class=, Driver=cdc_ether, 480M 
    |__ Port 1: Dev 4, If 1, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 4, If 2, Class=, Driver=cdc_acm, 480M
    |__ Port 1: Dev 4, If 3, Class=, Driver=cdc_acm, 480M
    |__ Port 1: Dev 4, If 4, Class=, Driver=cdc_acm, 480M
    |__ Port 1: Dev 4, If 5, Class=, Driver=cdc_acm, 480M
    |__ Port 1: Dev 4, If 6, Class=, Driver=cdc_acm, 480M
    |__ Port 1: Dev 4, If 7, Class=, Driver=cdc_acm, 480M
    |__ Port 2: Dev 3, If 0, Class=, Driver=cdc_ether, 480M
    |__ Port 2: Dev 3, If 7, Class=, Driver=cdc_acm, 480M
    |__ Port 2: Dev 3, If 5, Class=, Driver=cdc_acm, 480M
    |__ Port 2: Dev 3, If 3, Class=, Driver=cdc_acm, 480M
    |__ Port 2: Dev 3, If 1, Class=, Driver=cdc_ether, 480M
    |__ Port 2: Dev 3, If 6, Class=, Driver=cdc_acm, 480M
    |__ Port 2: Dev 3, If 4, Class=, Driver=cdc_acm, 480M
    |__ Port 2: Dev 3, If 2, Class=, Driver=cdc_acm, 480M

```
*note : I had this bug even with 23.05.0-rc2 () vanilla openwrt. (this puzzels me)
I am happy to provide any further info required.

brada4, 2024-03-15:
you could as well claim 4g dongle fixed the issue.
you had 2 disks on same bus in original report. 
It is power configuration certainly.

RydjGit, 2024-03-15:
Ok I agree that the panics have decreased after I changed to more powerfull power supply , but to prove my point I made a new build using the [official build system](https://firmware-selector.openwrt.org/?version=23.05.2&target=ramips%2Fmt7621&id=zbtlink_zbt-wg1602-v04-32m) :
 Installed Packages:

```
-uboot-envtools base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kmod-gpio-button-hotplug kmod-leds-gpio kmod-mt7603 kmod-mt76x2 kmod-nft-offload kmod-sdhci-mt7620 kmod-usb-ledtrig-usbport kmod-usb3 libc libgcc libustream-mbedtls logd luci mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-mbedtls luci luci-app-aria2 kmod-usb-storage htop btop usbutils pciutils dmesg block-mount kmod-fs-ntfs3 kmod-fs-ext4 kmod-fs-exfat e2fsprogs aria2 ariang kmod-usbmon kmod-fs-ksmbd ksmbd-server
```

here is the lsusb , no change in connected devices didnt touch any hardware config:

```
root@OpenWrt:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 2: Dev 2, If 0, Class=, Driver=, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=, 480M
    |__ Port 1: Dev 2, If 1, Class=, Driver=, 480M
    |__ Port 1: Dev 2, If 2, Class=, Driver=, 480M
    |__ Port 1: Dev 2, If 3, Class=, Driver=, 480M
    |__ Port 1: Dev 2, If 4, Class=, Driver=, 480M
    |__ Port 1: Dev 2, If 5, Class=, Driver=, 480M
    |__ Port 1: Dev 2, If 6, Class=, Driver=, 480M
    |__ Port 1: Dev 2, If 7, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 0, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 7, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 5, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 3, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 1, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 6, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 4, Class=, Driver=, 480M
    |__ Port 2: Dev 3, If 2, Class=, Driver=, 480M

```
why is my SSD not recognised?

and yes I didn't include the drivers for my two 4g modems as I dont use them, they were included by default in rooter.

brada4, 2024-03-15:
you can install kmod packages on top of basic openwrt image.
If you dont see device connected unlikely any driver will handle it.

brada4, 2024-03-15:
You can try onboard and PCIE (m2?) USB3 ports, those should work identically, but you never know. First some device must appear in lsusb, then install storage driver.

RydjGit, 2024-03-22:
Update :
I have been playing around with this  the past couple of days. I installed the mPCIe to USB 3.0 adapter into a windows Laptop with the same power delivery contraption, The chip worked but there was sudden intrruption of I/O I believe same with `uas_eh_device_reset_handler start` and looking into this further it was a firmware issue. I updated the firmware to version 2.0.6.0 and on windows I/O became faster stable and smooth, without anymore read/write intrrupts.

regarding Rooter os : crashes still happen, just needed to stress the I/O more so I had a hastly observation, sorry.

This mpcie card works without problems on windows with the same power delivery solution, ( not a  power problem)

the chip is  "renesas upd720202" online searching for this chip reveals that it doesnt play well with kernel 5.x !!!! even after firmware update. but the update was worth it (reading speed using samaba4 from SSD to my PC is 1 Gbit/sec my full bandwidth :) ).
 here is the repo for update tool for linux, I compiled this using official openwrt SDK  and flashed it([link](https://github.com/markusj/upd72020x-load))

long story short this chip can work on kernel 4.2x and 6.x but not 5.x

I am now on openwrt SNAPSHOT 6.1 kerenl and my router is downloading to SSD more than 50GB of data without any crashes,
I also conntected my other HDD to same bus so now I have 2 external  high power external storage devices connected and it has been running for 12 hours no problems (on 5.x kernel it would crash minutes or even seconds after I/O starts) 

SO YEAH IT IS DEFINILTY A SOFTWARE PROBLEM (mabe linux in general not specific to openwrt.)

close this please.

brada4, 2024-03-22:
Below textbox  a button for you.

closed 2024-03-22 by RydjGit