openwrt/openwrt #14874: MT7622B: AX6S: Failed to get patch semaphore

erpalma, 2024-03-13:
### Describe the bug

I own two AX6S (AX3200), one is newer and one is older (1y). The older one works fine on 23.05.2. The newer one has 2.4 WiFi not starting on both 23.05.2 and 1d3d6ef8269aca7ee34fb6ddd1c11d3fe45e785b (using PR #14770), of course on factory settings. Stock firmware works fine. On dmesg I get:
```
[   26.710686] mt7622-wmac 18000000.wmac: Message 00000010 (seq 1) timeout
[   26.717336] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   26.942056] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.161036] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.380983] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.600963] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.821048] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.041012] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.260994] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.480964] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.700957] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.921010] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
```
Reboot or reset does not fix the problem as in #13797. Hardware revision is the same RB01.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Steps to reproduce

Install OpenWRT, boot, enable 2.4 WiFi.

### Actual behaviour

2.4 WiFi not enabling

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kibernetiq, 2024-12-08:
OpenWrt version 23.05.4, 23.05.5 same errors on the same device

lmo-proton, 2024-12-14:
Same problem, 2.4g works on stock firmware on every reboot, while only 50/50 chance on OpenWrt firmware. Feels like a race condition inside the bootup process.

My solution is keep rebooting the device until the kernel log get these 2 lines:
```
[    7.210156] mt7622-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20190801210006a
[    7.383044] mt7622-wmac 18000000.wmac: N9 Firmware Version: _reserved_, Build Time: 20220630094834
```

Similar problem can also be found in https://github.com/openwrt/mt76/issues/774 in April 2023 (before 23.05 released)