openwrt/openwrt #14894: wget-nossl / wget-ssl break opkg and other tools that rely on "busybox wget" like sysupgrade

humaita-github, 2024-03-14:
### Describe the bug

Installing wget-nossl (or wget-ssl) will break opkg (e.g. cannot update list in CLI or in LUCI) and other tools that rely on "busybox wget". This is because wget-nossl will replace the wget command from busybox, and both wgets have different command line options. Removing wget-nossl will make the system return to the normal state.

opkg (and other tools reyling directly on wget like sysupgrade) should use the command "bin/busybox wget" instead of "wget" to avoid this problem.

This behaviour was tested on a Asus RT-AX53U and on a D-Link DAP-X1860, both running OpenWrt SNAPSHOT, r25519-7ea82bc17d, and also with previous snapshots. But I assume this occurs in other branches and versions as well.

### OpenWrt version

OpenWrt SNAPSHOT, r25519-7ea82bc17d

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

1) Install wget-nossl
2) Try to update lists in opkg

### Actual behaviour

Error as opkg cannot download files

### Expected behaviour

opkg downloads files

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

humaita-github, 2024-03-15:
Yes, this is the official version that you get from "Attended Sysupgrade" when upgrading OpenWRT in the SNAPSHOT branch.

brada4, 2024-03-15:
Nothing to do with busybox.
`ln -sf /bin/uclient-fetch /usr/bin/wget`

brada4, 2024-03-15:
Could be /bin/client-fetch here to make opkg inert to packages changing.
https://github.com/openwrt/opkg-lede/blob/d038e5b6d155784575f62a66a8bb7e874173e92e/libopkg/opkg_download.c#L160

cbm755, 2026-03-28:
On a newly-installed 25.12.5 with no migrated config, I was able to break my system by doing `apk add wget`, which symlinked `/bin/wget` to `/usr/libexec/wget_nossl`.

After this, all `apk` commands, even `apk update` would fail like this:
```
# apk update
ERROR: wget: exited with error 1
WARNING: updating and opening https://downloads.openwrt.org/releases/25.12.2/targets/ipq40xx/generic/packages/packages.adb: unexpected end of file
ERROR: wget: exited with error 1
WARNING: updating and opening https://downloads.openwrt.org/releases/25.12.2/packages/arm_cortex-a7_neon-vfpv4/base/packages.adb: unexpected end of file
ERROR: wget: exited with error 1
WARNING: updating and opening https://downloads.openwrt.org/releases/25.12.2/targets/ipq40xx/generic/kmods/6.12.74-1-71bc38810804b78a5aeac418fdc644af/packages.adb: unexpected end of file
```

I was able to fix my system by manually fixing the symlink back to `uclient-fetch` (as above in https://github.com/openwrt/openwrt/issues/14894#issuecomment-1999774894)

I then removed the problematic wget package:
```
# apk del wget
(1/7) Purging wget-nossl (1.25.0-r2)
  Executing wget-nossl-1.25.0-r2.pre-deinstall
  * add alternative: /usr/bin/wget -> /bin/uclient-fetch
```

(I had "curl" and "wget" in a list of manually-installed opkgs from 22.03.5)