openwrt/openwrt #14898: grep hangs in little probability on sunxi board

srdgame, 2024-03-15:
### Describe the bug

This happens only once in my Allwinner A40i (sunxi) board, when I tried to reproduce an reboot bug (which kernel keep says cpu0 interrupt hangs).  We take over the watchdog after our binary boots up, and then call "reboot" after then minutes.

below is the result of 'ps'

``` bash
 1086 root         0 IW<  [scsi_tmf_0]
 1127 root         0 IW<  [kworker/2:1H]
 1218 root         0 SW<  [loop0]
 1221 root         0 SW   [f2fs_flush-7:0]
 1222 root         0 SW   [f2fs_discard-7:]
 1223 root         0 SW   [f2fs_gc-7:0]
 1432 ubus      1024 S    /sbin/ubusd
 1445 root       700 S    /sbin/askfirst /usr/libexec/login.sh
 1503 root       820 S    /sbin/urngd
 1795 root         0 IW<  [cfg80211]
 2226 logd      1072 S    /sbin/logd -S 64
 2291 root      1864 S    /sbin/rpcd -s /var/run/ubus/ubus.sock -t 30
 2504 dnsmasq   1184 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c
 2577 root         0 SW   [jbd2/mmcblk1p3-]
 2578 root         0 IW<  [ext4-rsv-conver]
 2589 root       864 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p
 2727 root      4056 S    /usr/sbin/hostapd -s -g /var/run/hostapd/global
 2728 root      4056 S    /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_suppl
 2760 root         0 IW<  [kworker/0:2H-kb]
 2791 root         0 IW<  [kworker/3:1H]
 2795 root      1528 S    /sbin/netifd
 2857 root      1248 S    /usr/sbin/odhcpd
 3020 root      1360 S    /usr/sbin/uhttpd -f -h /www -r ThingsLink -x /cgi-bi
 3096 root       804 S    /usr/sbin/ser2net -n -c /tmp/ser2net.conf
 3148 root      1252 S    /sbin/blockd
 3150 root      1236 S    /bin/sh /etc/rc.common /etc/rc.d/S80ucitrack boot
 3206 root      1100 R    grep -sqE USE_PROCD=. /etc/init.d/network
 3227 root      1296 S    /bin/sh ./qmi.sh qmi setup 4g_wan {"proto":"qmi","de
 3237 root      1100 R    grep "UIM uninitialized"
 3253 root      1144 S    /bin/sh /sbin/hotplug-call net
 3257 root      1152 S    /bin/sh /sbin/hotplug-call net
 3262 root      1100 R    sed -ne /^[[:space:]]*net\..*\.wwan0\./p /etc/sysctl
 3263 root      1100 S    sysctl -e -p -
 3264 root      1100 S    logger -t sysctl
 3372 root      1108 S    /bin/ash --login
 3424 root         0 IW   [kworker/1:0-eve]
 3453 root         0 IW   [kworker/2:0-rcu]
 3463 root         0 IW   [kworker/3:2-rcu]
 3466 root         0 IW   [kworker/2:2-ipv]
 3481 root         0 IW   [kworker/3:0-mm_]
 3484 root         0 IW   [kworker/0:0-eve]
 3506 root         0 IW   [kworker/u8:2-ev]
 3512 root         0 IW   [kworker/u8:0-ev]
 3515 root         0 IW   [kworker/u8:3-ev]
 3526 root         0 IW   [kworker/1:2]
 3527 root         0 IW   [kworker/u8:1-ev]
 3528 root      1100 R    ps
/etc/init.d # [33596.325344] dwmac-sun8i 1c50000.ethernet eth1: Link is Down
[33601.525573] dwmac-sun8i 1c50000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[33604.645340] dwmac-sun8i 1c50000.ethernet eth1: Link is Down
[33606.725567] dwmac-sun8i 1c50000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[33629.605343] dwmac-sun8i 1c50000.ethernet eth1: Link is Down
[33631.685571] dwmac-sun8i 1c50000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx

```

And run grep in console will hang without any output.  After kill all 'grep' process, run 'ps':

``` bash
 2226 logd      1072 S    /sbin/logd -S 64
 2291 root      1864 S    /sbin/rpcd -s /var/run/ubus/ubus.sock -t 30
 2504 dnsmasq   1184 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c
 2577 root         0 SW   [jbd2/mmcblk1p3-]
 2578 root         0 IW<  [ext4-rsv-conver]
 2589 root       864 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p
 2727 root      4056 S    /usr/sbin/hostapd -s -g /var/run/hostapd/global
 2728 root      4056 S    /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_suppl
 2760 root         0 IW<  [kworker/0:2H-kb]
 2791 root         0 IW<  [kworker/3:1H]
 2795 root      1528 S    /sbin/netifd
 2857 root      1248 S    /usr/sbin/odhcpd
 3020 root      1360 S    /usr/sbin/uhttpd -f -h /www -r ThingsLink -x /cgi-bi
 3096 root       804 S    /usr/sbin/ser2net -n -c /tmp/ser2net.conf
 3148 root      1252 S    /sbin/blockd
 3150 root      1248 S    /bin/sh /etc/rc.common /etc/rc.d/S80ucitrack boot
 3227 root      1296 S    /bin/sh ./qmi.sh qmi setup 4g_wan {"proto":"qmi","de
 3253 root      1144 S    /bin/sh /sbin/hotplug-call net
 3257 root      1152 S    /bin/sh /sbin/hotplug-call net
 3372 root      1108 S    /bin/ash --login
 3424 root         0 IW   [kworker/1:0-eve]
 3453 root         0 IW   [kworker/2:0-rcu]
 3463 root         0 IW   [kworker/3:2-rcu]
 3466 root         0 IW   [kworker/2:2-eve]
 3481 root         0 IW   [kworker/3:0-mm_]
 3484 root         0 IW   [kworker/0:0-rcu]
 3532 root         0 IW   [kworker/u8:3-ev]
 3538 root         0 IW   [kworker/u8:0-ev]
 3561 root      1100 R    grep -sqE USE_PROCD=. /etc/init.d/firewall
 3564 root      1100 R    grep "Not supported"\|"Invalid QMI command"
 3565 root      1100 R    sed -ne /^[[:space:]]*net\..*\.wwan0\./p /etc/sysctl
 3566 root      1100 S    sysctl -e -p -
 3567 root      1100 S    logger -t sysctl
 3571 root         0 IW   [kworker/0:2-eve]
 3574 root         0 IW   [kworker/u8:2-ev]
 3579 root         0 IW   [kworker/1:2]
 3580 root         0 IW   [kworker/u8:1-ev]
 3581 root      1100 R    ps
/etc/init.d # [36594.645332] dwmac-sun8i 1c50000.ethernet eth1: Link is Down
[36595.685573] dwmac-sun8i 1c50000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[36597.765320] dwmac-sun8i 1c50000.ethernet eth1: Link is Down
[36598.805580] dwmac-sun8i 1c50000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[36601.925331] dwmac-sun8i 1c50000.ethernet eth1: Link is Down
[36604.005558] dwmac-sun8i 1c50000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
```

The F2FS filesystem works fine, as I can run 'cat /etc/config/network' successfully.

There are commits in my own openwrt repo for apply my patches for A40i chip, so the openwrt version may not be useful for you.

### OpenWrt version

r17294-6145ee6b6e

### OpenWrt release

21.02-SNAPSHOT

### OpenWrt target/subtarget

sunxi/cortexa7

### Device

ThingsLink K2X

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-15:
The grep commands are around mobile modem. Can you try to re-create problem using same mobile modem with a supported openwrt device & release?

srdgame, 2024-03-15:
> The grep commands are around mobile modem. Can you try to re-create problem using same mobile modem with a supported openwrt device & release?

I will try to reproduce it with board that removed mobile modem. As this is hard to reproduce, so I have no idea when this will be happen again.

In my understanding that 'grep -seE USE_PROCD=. /etc/config/network' command has no relationship with qmi (mobile modem). 

I have tried 'cat /etc/openwrt_release | grep ID' which supposed to output DISTRIB_ID='OpenWrt', actually this hangs.  So I think this hangs happens in musl's pipe or kernel's pipe?

srdgame, 2024-03-15:
> Invalid Release reported. `21.02-SNAPSHOT` Is this from a clean repository?

No, this is not from a clean repo

closed 2024-03-15 by ynezz