openwrt/openwrt #14901: build: user defined shell variable CHECK leads do `make download` failure

boretom, 2024-03-15:
### Describe the bug

If the user defines an environment variable CHECK, e.g CHECK=1 and runs `make download` then it seems as if `make check` runs also before the download.

This leads to 
a) warnings like `WARNING: bison-3.8.2.tar.xz is missing, please run make download before re-running this check`
b) failure to download the kernel-headers:

```
$ CHECK=1 make download V=sc
<snip lots of lines>
make[2]: Entering directory '/home/jane/projects/openwrt/toolchain/kernel-headers'
WARNING: linux-.tar.xz is missing, please run make download before re-running this check
mkdir -p /home/jane/projects/openwrt/dl
SHELL= flock /home/jane/projects/openwrt/tmp/.linux-.tar.xz.flock -c '          /home/jane/projects/openwrt/scripts/download.pl "/home/jane/projects/openwrt/dl" "linux-.tar.xz" "x" "" "@KERNEL/linux/kernel/v.x"      '
make[2]: Leaving directory '/home/jane/projects/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/download#0.01#0.00#0.02
    ERROR: toolchain/kernel-headers failed to build.
<snip again lots of lines>
```

In the corresponding forum thread ["make download" fails](https://forum.openwrt.org/t/make-download-fails/190482/) @hnyman pointed to [build: implement make check and make package/X/check](https://github.com/openwrt/openwrt/commit/7a315b0b5d6aa91695853a8647383876e4b49a7a) to the relevant commit from back in December 2016.

Since the build system is quite complex I'm not sure if adding `CHECK:=` (and probably `DUMP:=`) to the TOPDIR/`Makefile` is enough.

If already on it maybe it's worth checking if in `rules.mk` the `ifeq ($(DUMP),)` (line 9 to 11 or maybe line 9 to 13) should be moved after `ifneq ($(filter check,$(MAKECMDGOALS)),)`.

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

armsr/armv8

### Device

buildsystem

### Image kind

Official downloaded image

### Steps to reproduce

1) checkout OpenWrt source & cd into it
2) `./scripts/feeds update -a && ./scripts/feeds install -a`
3) run

```
$ cat > .config << EOC
CONFIG_TARGET_armsr=y
CONFIG_TARGET_armsr_armv8=y
CONFIG_TARGET_armsr_armv8_DEVICE_generic=y
EOC
```

4) `make defconfig`
5) `CHECK=1 make download V=sc | tee make-download.log`
6) fgrep -B5 -A1 ERROR make-download.log

```
WARNING: linux-.tar.xz is missing, please run make download before re-running this check
mkdir -p /home/jane/projects/openwrt/dl
SHELL= flock /home/jane/projects/openwrt/tmp/.linux-.tar.xz.flock -c '          /home/jane/projects/openwrt/scripts/download.pl "/home/jane/projects/openwrt/dl" "linux-.tar.xz" "x" "" "@KERNEL/linux/kernel/v.x"      '
make[2]: Leaving directory '/home/jane/projects/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/download#0.01#0.00#0.02
    ERROR: toolchain/kernel-headers failed to build.
make[1]: Leaving directory '/home/jane/projects/openwrt'
```

### Diffconfig

```text
CONFIG_TARGET_armsr=y
CONFIG_TARGET_armsr_armv8=y
CONFIG_TARGET_armsr_armv8_DEVICE_generic=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2024-03-18:
@boretom yes it's a bug but can you better describe your usage?

boretom, 2024-03-18:
Hey @Ansuel, another user opened the thread on the forum because he followed the OpenWrt guide but didn't make it past the `make download`.

I did some debugging and came to the conclusion that if CHECK is set the error appears. And it turned out he had the CHECK variable set in the users shell for his kernel developing.

Many eons ago I was taught to use prefixes in methods or variable naming and this could also in this case help prevent such issues. And if it's an internal variable then lower case would be recommended according to the `make` documentation.