openwrt/openwrt #14906: hostapd: unable to start HT160 on 149 channel

testuser7, 2024-03-15:
### Describe the bug

Linksys MX8500 support new 5GHz channels (169, 173 and 177):
```
iwinfo phy0 freqlist
  5.180 GHz (Band: 5 GHz, Channel 36) [NO_HT40-]
  5.200 GHz (Band: 5 GHz, Channel 40) 
  5.220 GHz (Band: 5 GHz, Channel 44) 
  5.240 GHz (Band: 5 GHz, Channel 48) 
  5.260 GHz (Band: 5 GHz, Channel 52) 
  5.280 GHz (Band: 5 GHz, Channel 56) 
  5.300 GHz (Band: 5 GHz, Channel 60) 
  5.320 GHz (Band: 5 GHz, Channel 64) [NO_HT40+]
  5.500 GHz (Band: 5 GHz, Channel 100) [NO_HT40-]
  5.520 GHz (Band: 5 GHz, Channel 104) 
  5.540 GHz (Band: 5 GHz, Channel 108) 
  5.560 GHz (Band: 5 GHz, Channel 112) 
  5.580 GHz (Band: 5 GHz, Channel 116) 
  5.600 GHz (Band: 5 GHz, Channel 120) 
  5.620 GHz (Band: 5 GHz, Channel 124) 
  5.640 GHz (Band: 5 GHz, Channel 128) 
  5.660 GHz (Band: 5 GHz, Channel 132) 
  5.680 GHz (Band: 5 GHz, Channel 136) 
  5.700 GHz (Band: 5 GHz, Channel 140) 
  5.720 GHz (Band: 5 GHz, Channel 144) [NO_HT40+]
  5.745 GHz (Band: 5 GHz, Channel 149) [NO_HT40-]
  5.765 GHz (Band: 5 GHz, Channel 153) 
  5.785 GHz (Band: 5 GHz, Channel 157) 
  5.805 GHz (Band: 5 GHz, Channel 161) 
  5.825 GHz (Band: 5 GHz, Channel 165) 
  5.845 GHz (Band: 5 GHz, Channel 169) 
  5.865 GHz (Band: 5 GHz, Channel 173) 
  5.885 GHz (Band: 5 GHz, Channel 177) [NO_HT40+]
```

Currently, htmode `HE160` cannot be run on these channels.

### OpenWrt version

r0+25568-18c82ad434

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Linksys MX8500

### Image kind

Self-built image

### Steps to reproduce

radio configuration:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi'
        option channel '149'
        option band '5g'
        option htmode 'HE160'
        option disabled '0'
        option country 'US'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '6g'
        option encryption 'sae-mixed'
        option key '*'

```

### Actual behaviour

```
Thu Jan  1 00:02:57 1970 daemon.notice netifd: radio0 (2864): WARNING: Variable 'data' does not exist or is not an array/object
Thu Jan  1 00:02:57 1970 daemon.notice hostapd: Set new config for phy phy0:
Thu Jan  1 00:02:57 1970 daemon.notice wpa_supplicant[2095]: Set new config for phy phy0
Thu Jan  1 00:02:58 1970 daemon.notice wpa_supplicant[2095]: Set new config for phy phy0
Thu Jan  1 00:02:58 1970 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Thu Jan  1 00:02:58 1970 daemon.notice hostapd: Restart interface for phy phy0
Thu Jan  1 00:02:58 1970 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=149 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=149 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=149 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu
Thu Jan  1 00:02:58 1970 daemon.notice netifd: Wireless device 'radio0' is now up
Thu Jan  1 00:02:58 1970 kern.info kernel: [  178.239040] br-lan: port 5(phy0-ap0) entered blocking state
Thu Jan  1 00:02:58 1970 kern.info kernel: [  178.239083] br-lan: port 5(phy0-ap0) entered disabled state
Thu Jan  1 00:02:58 1970 kern.info kernel: [  178.243507] ath11k c000000.wifi phy0-ap0: entered allmulticast mode
Thu Jan  1 00:02:58 1970 kern.info kernel: [  178.249178] ath11k c000000.wifi phy0-ap0: entered promiscuous mode
Thu Jan  1 00:02:58 1970 daemon.warn hostapd: nl80211: Failed to add interface phy0-ap0 into bridge br-lan: Resource busy
Thu Jan  1 00:02:58 1970 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Jan  1 00:02:58 1970 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Jan  1 00:02:58 1970 daemon.err hostapd: Interface initialization failed
Thu Jan  1 00:02:58 1970 daemon.notice hostapd: phy0-ap0: interface state HT_SCAN->DISABLED
Thu Jan  1 00:02:58 1970 daemon.notice hostapd: phy0-ap0: AP-DISABLED
```

### Additional info

`HE80` htmode on channels `149` and `165` works.

### Diffconfig

```text
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq807x=y
CONFIG_TARGET_qualcommax_ipq807x_DEVICE_linksys_mx8500=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-15:
Not really (no-ir means no initial radiation, i.e somebody has to be transmitting in that channel before you)
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1796

brada4, 2024-03-15:
`iw list` shows radio caps and region interdection.

testuser7, 2024-03-16:
> `iw list` shows radio caps and region interSection.

```
root@OpenWrt:/# iw phy0 info
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 16
	max scan IEs length: 142 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Available Antennas: TX 0xf0 RX 0xf0
	Configured Antennas: TX 0xf0 RX 0xf0
	Supported interface modes:
		 * managed
		 * AP
		 * monitor
		 * mesh point
	Band 2:
		Capabilities: 0x19ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-31
		VHT Capabilities (0x339b79f6):
			Max MPDU length: 11454
			Supported Channel Width: 160 MHz
			RX LDPC
			short GI (80 MHz)
			short GI (160/80+80 MHz)
			TX STBC
			SU Beamformer
			SU Beamformee
			MU Beamformer
			MU Beamformee
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 0 Mbps
		VHT extended NSS: supported
		HE Iftypes: managed
[... 281 lines cut ...]
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Maximum associated stations in AP mode: 512
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ ACK_SIGNAL_SUPPORT ]: ack signal level support
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ ENABLE_FTM_RESPONDER ]: enable FTM (Fine Time Measurement) responder
		* [ STA_TX_PWR ]: TX power control per station
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BSS_COLOR ]: BSS coloring support
```

testuser7, 2024-03-16:
> Not really (no-ir means no initial radiation, i.e somebody has to be transmitting in that channel before you) https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1796

Doesn't `no-ir` apply only when we don't use `no-outdoor`?
The entire 6g band is `no-ir` for the US.

brada4, 2024-03-16:
It means no initial transmission. Say set regdomain 00 and try to start 5ghz radio at all.

testuser7, 2024-03-16:
> It means no initial transmission. Say set regdomain 00 and try to start 5ghz radio at all.

```
Thu Jan  1 00:03:11 1970 daemon.notice netifd: radio0 (2786): WARNING: Variable 'data' does not exist or is not an array/object
Thu Jan  1 00:03:11 1970 daemon.notice hostapd: Set new config for phy phy0:
Thu Jan  1 00:03:11 1970 daemon.notice wpa_supplicant[2016]: Set new config for phy phy0
Thu Jan  1 00:03:11 1970 daemon.notice wpa_supplicant[2016]: Set new config for phy phy0
Thu Jan  1 00:03:11 1970 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Thu Jan  1 00:03:11 1970 daemon.notice hostapd: Restart interface for phy phy0
Thu Jan  1 00:03:11 1970 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=00 ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=149 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=149 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=149 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu
Thu Jan  1 00:03:11 1970 daemon.err hostapd: Line 7: Invalid country_code '00'
Thu Jan  1 00:03:11 1970 daemon.err hostapd: Cannot enable IEEE 802.11d without setting the country_code
Thu Jan  1 00:03:11 1970 daemon.err hostapd: 2 errors found in configuration file '<inline>'
Thu Jan  1 00:03:11 1970 daemon.err hostapd: Failed to set up interface with data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=00 ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=149 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=149 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=149 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_req
Thu Jan  1 00:03:11 1970 daemon.notice hostapd: hostapd.add_iface failed for phy phy0 ifname=phy0-ap0
Thu Jan  1 00:03:11 1970 daemon.notice netifd: Wireless device 'radio0' is now up
```

brada4, 2024-03-16:
By means of nothing for country in config file... Wifi will come up in ch36 IF thete is any other ap before it.
You can trace ACS like missing passive scan in netcard by rising wifi debug level for hostap, by default there is not much logged.

testuser7, 2024-03-16:
> You can trace ACS like missing passive scan in netcard by rising wifi debug level for hostap, by default there is not much logged.

Do I need to build `hostapd` with some special configuration to have debug logs?
I have set `Minimum debug message priority` to `0`.
```
Thu Jan  1 00:09:30 1970 daemon.notice netifd: radio0 (2936): WARNING: Variable 'data' does not exist or is not an array/object
Thu Jan  1 00:09:30 1970 daemon.notice hostapd: Set new config for phy phy0:
Thu Jan  1 00:09:30 1970 daemon.notice wpa_supplicant[2011]: Set new config for phy phy0
Thu Jan  1 00:09:31 1970 daemon.notice wpa_supplicant[2011]: Set new config for phy phy0
Thu Jan  1 00:09:31 1970 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Thu Jan  1 00:09:31 1970 daemon.notice hostapd: Restart interface for phy phy0
Thu Jan  1 00:09:31 1970 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=0 logger_stdout=127 logger_stdout_level=0 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=149 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=149 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=149 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu
Thu Jan  1 00:09:31 1970 kern.info kernel: [  571.245475] br-lan: port 5(phy0-ap0) entered blocking state
Thu Jan  1 00:09:31 1970 kern.info kernel: [  571.245513] br-lan: port 5(phy0-ap0) entered disabled state
Thu Jan  1 00:09:31 1970 kern.info kernel: [  571.250061] device phy0-ap0 entered promiscuous mode
Thu Jan  1 00:09:31 1970 kern.info kernel: [  571.255474] br-lan: port 5(phy0-ap0) entered blocking state
Thu Jan  1 00:09:31 1970 kern.info kernel: [  571.260636] br-lan: port 5(phy0-ap0) entered forwarding state
Thu Jan  1 00:09:31 1970 daemon.warn hostapd: nl80211: Failed to add interface phy0-ap0 into bridge br-lan: Resource busy
Thu Jan  1 00:09:31 1970 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Jan  1 00:09:31 1970 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Jan  1 00:09:31 1970 kern.info kernel: [  571.267036] br-lan: port 5(phy0-ap0) entered disabled state
Thu Jan  1 00:09:31 1970 daemon.notice netifd: Wireless device 'radio0' is now up
Thu Jan  1 00:09:31 1970 daemon.err hostapd: Interface initialization failed
Thu Jan  1 00:09:31 1970 daemon.notice hostapd: phy0-ap0: interface state HT_SCAN->DISABLED
Thu Jan  1 00:09:31 1970 daemon.notice hostapd: phy0-ap0: AP-DISABLED
```

brada4, 2024-03-16:
Add  `option channels '36 100 149'` to allow fallback and get more detailed scan report.
There is no constructive warning when i set single channel outside fegulatory specs,  but i get full radio survey at debug level 1 (0 will have all raw frames, rotating log in millisecond)

testuser7, 2024-03-16:
There is an issue with changing log level: https://github.com/openwrt/openwrt/issues/8061
Debug logs:
```
Thu Jan  1 00:05:16 1970 daemon.notice netifd: radio0 (2954): WARNING: Variable 'data' does not exist or is not an array/object
Thu Jan  1 00:05:16 1970 daemon.notice hostapd: Set new config for phy phy0:
Thu Jan  1 00:05:16 1970 daemon.notice wpa_supplicant[2015]: Set new config for phy phy0
Thu Jan  1 00:05:16 1970 daemon.notice wpa_supplicant[2015]: Set new config for phy phy0
Thu Jan  1 00:05:16 1970 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Thu Jan  1 00:05:16 1970 daemon.notice hostapd: Restart interface for phy phy0
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: RTM_NEWLINK: ifi_index=10 ifname=phy0-ap0 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1002 ()
Thu Jan  1 00:05:16 1970 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=1 logger_stdout=127 logger_stdout_level=1 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=149 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=2 vht_oper_centr_freq_seg0_idx=149 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=2 he_oper_centr_freq_seg0_idx=149 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:2
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:4
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:6
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:13
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:11
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:12
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:8
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:9
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:10
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Using driver-based off-channel TX
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 04 00 00 00 00 00 00 40
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 04 00 00 00 00 00 00 40
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities for interface type unknown
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Extended capabilities - hexdump(len=8): 04 00 40 00 00 00 00 40
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Extended capabilities mask - hexdump(len=8): 04 00 40 00 00 00 00 40
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities for interface type STATION
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Extended capabilities - hexdump(len=10): 04 00 40 00 00 00 00 40 00 20
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Extended capabilities mask - hexdump(len=10): 04 00 40 00 00 00 00 40 00 20
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities for interface type AP
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Extended capabilities - hexdump(len=11): 04 00 40 00 00 00 00 40 00 40 08
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Extended capabilities mask - hexdump(len=11): 04 00 40 00 00 00 00 40 00 40 08
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: mbssid: max interfaces 17, max profile periodicity 8
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: key_mgmt=0x1ff0f enc=0xfec auth=0x7 flags=0x5800511d343d1e0 rrm_flags=0x30 probe_resp_offloads=0x0 max_stations=512 max_remain_on_chan=5000 max_scan_ssids=16
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: interface phy0-ap0 in phy phy0
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Set mode ifindex 10 iftype 3 (AP)
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Setup AP(phy0-ap0) - device_ap_sme=0 use_monitor=0
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Subscribe to mgmt frames with AP handle 0x7fa8bcc500
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x7fa8bcc500 match= multicast=0
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x7fa8bcc500 match= multicast=0
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x7fa8bcc500 match= multicast=0
Thu Jan  1 00:05:16 1970 daemon.debug hostapd: nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x7fa8bcc500 match= multicast=0
[... cut ...]
Thu Jan  1 00:05:17 1970 daemon.debug hostapd: Available channel: 165
Thu Jan  1 00:05:17 1970 daemon.debug hostapd: Available channel: 169
Thu Jan  1 00:05:17 1970 daemon.debug hostapd: Available channel: 173
Thu Jan  1 00:05:17 1970 daemon.debug hostapd: Available channel: 177
Thu Jan  1 00:05:17 1970 daemon.err hostapd: Interface initialization failed
Thu Jan  1 00:05:17 1970 daemon.notice hostapd: phy0-ap0: interface state HT_SCAN->DISABLED
Thu Jan  1 00:05:17 1970 daemon.notice hostapd: phy0-ap0: AP-DISABLED
```

brada4, 2024-03-16:
Maybe something in firmware chokes at those channels. Really it says that channel configured was surveyed as good...
Add fallback channels config, even it sounds nonsensical - at least you will get wifi up, and some day it will land in chosen channel.

testuser7, 2024-03-16:
Something is wrong:
```
Thu Jan  1 00:05:17 1970 daemon.debug hostapd: DFS chan_idx seems wrong; num-ch: 28 ch-no: 135 conf-ch-no: 149 11n: 1 sec-ch: 1 vht-oper-width: 2
```
Channels 149-177 are DFS-free:
```
root@OpenWrt:/# iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0 (self-managed)
country US: DFS-FCC
	(2402 - 2472 @ 40), (6, 30), (N/A)
	(5170 - 5250 @ 80), (6, 30), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (6, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (6, 24), (0 ms), DFS, AUTO-BW
	(5735 - 5895 @ 160), (6, 30), (N/A), AUTO-BW
```

brada4, 2024-03-16:
@robimarko may know better how to debug it. The return from (closed source) firmware looks either garbage or inconsistent. For now add spare channels, every radio (re-)start will attempt ch 149 but on failure go with others.

robimarko, 2024-03-17:
Sadly no, I dont have ideas about the root cause of this

testuser7, 2024-03-17:
Can debug logs from the ath11k driver may help? E.g. WMI?

brada4, 2024-03-17:
Actually better in forums. 
whats worth here is to roll through few recent firmwares just in case some version works right away.

https://git.codelinaro.org/clo/ath-firmware/ath11k-firmware

testuser7, 2024-03-17:
I'm using latest firmware `WLAN.HK.2.9.0.1-01977-QCAHKSWPL_SILICONZ-1`: https://github.com/quic/upstream-wifi-fw/
with updated regdb in BDF from here: https://git.codelinaro.org/clo/qsdk/oss/ath11k-bdf/-/tree/NHSS.QSDK.12.4.5

brada4, 2024-03-17:
I exactly meant to try to downgrade it a bit, like if at least error message changes....

Shine-, 2024-03-18:
These channels are marked as "NO-IR" in the RegDB, which means no AP is allowed on these channels **at all**.

The reason for this is explained in the [discussion](https://lists.infradead.org/pipermail/wireless-regdb/2021-July/001252.html) related to the [corresponding RegDB patch](https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/commit/?id=9839e1e389b5d252c61b6bfb11bc798c2125a309): Apparently in the RegDB, there's currently no possibility to set the "passive scan" flag (required for these channels in the US) without the "NO-IR" flag. Thus, these channels can only be used in STA mode, but not AP mode.

There was also a [forum discussion](https://forum.openwrt.org/t/new-indoor-5-ghz-channels/156942) about that a while ago.

Slightly related, the same (NO-IR) is set for the whole 5GHz band in the World regdomain - OpenWrt simply patches that out (see `package/firmware/wireless-regdb/patches/500-world-regd-5GHz.patch`), to make the 5GHz band usable in AP mode.

Shine-, 2024-03-18:
Sorry, not familiar with ath11k, my comment was regarding RegDB in general.

Shine-, 2024-03-18:
On second thought, the Linux RegDB is missing an entry like `(5730 - 5895 @ 160)`, `(5730 - 5895 @ 80)` or similar for regdom US, which could explain why not even ch165@80MHz will work (on non-ath11k chips) according the forum thread linked above.

Now as you state that with ath11k, ch165@80MHz works, perhaps ath11k's internal RegDB **does** have an 80MHz entry for that range, but not a 160MHz entry? Is that internal RegDB available anywhere in readable form? That would likely give you your answers.

Disclaimer: I'm not in regdom US, so this is all theoretical. I'm forbidden by local regulatory rules to use channel 177 myself.

testuser7, 2024-03-18:
Here is the output from `iw reg get`: https://github.com/openwrt/openwrt/issues/14906#issuecomment-2002142359
`phy0` regdb is `self-managed`.

brada4, 2024-03-18:
I also added a link to current regdb and your output matches. no-ir and radar are hw capabilities. Channels render unusable if required cap is missing. 
Another test method - walk all band in 20 or 40 mhz chunks, that way you learn good event sequence and make better guesses around ?error?

testuser7, 2024-03-18:
Debug logs - 80MHz on 165 channel:
```
Thu Jan  1 00:10:37 1970 daemon.notice netifd: radio0 (3200): WARNING: Variable 'data' does not exist or is not an array/object
Thu Jan  1 00:10:37 1970 daemon.notice hostapd: Set new config for phy phy0:
Thu Jan  1 00:10:37 1970 daemon.notice wpa_supplicant[2007]: Set new config for phy phy0
Thu Jan  1 00:10:37 1970 daemon.notice wpa_supplicant[2007]: Set new config for phy phy0
Thu Jan  1 00:10:37 1970 daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Thu Jan  1 00:10:37 1970 daemon.notice hostapd: Restart interface for phy phy0
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: RTM_NEWLINK: ifi_index=10 ifname=phy0-ap0 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1002 ()
Thu Jan  1 00:10:37 1970 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=165 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=171 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=1 he_oper_centr_freq_seg0_idx=171 he_su_beamformer=1 he_su_beamformee=1 he_mu_beamformer=1 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_re
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:2
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:4
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:6
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:13
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:11
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:12
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:8
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:9
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Supported cipher 00-0f-ac:10
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Using driver-based off-channel TX
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 04 00 00 00 00 00 00 40
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 04 00 00 00 00 00 00 40
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities for interface type unknown
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Extended capabilities - hexdump(len=8): 04 00 40 00 00 00 00 40
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Extended capabilities mask - hexdump(len=8): 04 00 40 00 00 00 00 40
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities for interface type STATION
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Extended capabilities - hexdump(len=10): 04 00 40 00 00 00 00 40 00 20
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Extended capabilities mask - hexdump(len=10): 04 00 40 00 00 00 00 40 00 20
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Driver-advertised extended capabilities for interface type AP
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Extended capabilities - hexdump(len=11): 04 00 40 00 00 00 00 40 00 40 08
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Extended capabilities mask - hexdump(len=11): 04 00 40 00 00 00 00 40 00 40 08
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: mbssid: max interfaces 17, max profile periodicity 8
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: key_mgmt=0x1ff0f enc=0xfec auth=0x7 flags=0x5800511d343d1e0 rrm_flags=0x30 probe_resp_offloads=0x0 max_stations=512 max_remain_on_chan=5000 max_scan_ssids=16
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: interface phy0-ap0 in phy phy0
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Set mode ifindex 10 iftype 3 (AP)
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Setup AP(phy0-ap0) - device_ap_sme=0 use_monitor=0
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Subscribe to mgmt frames with AP handle 0x7fb6904500
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x7fb6904500 match= multicast=0
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x7fb6904500 match= multicast=0
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x7fb6904500 match= multicast=0
Thu Jan  1 00:10:37 1970 daemon.debug hostapd: nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x7fb6904500 match= multicast=0
[... cut ...]
Thu Jan  1 00:10:51 1970 daemon.debug hostapd: HE: MU EDCA Parameter Set element - hexdump(len=13): 00 08 a9 ff 2f a9 ff 45 75 ff 65 75 ff
Thu Jan  1 00:10:51 1970 daemon.debug hostapd: nl80211: send_mlme - da=10:f6:0a:19:ca:9e noack=1 freq=0 no_cck=0 offchanok=0 wait_time=0 no_encrypt=0 fc=0x50 (WLAN_FC_STYPE_PROBE_RESP) nlmode=3
Thu Jan  1 00:10:51 1970 daemon.debug hostapd: nl80211: send_mlme - Use bss->freq=5825
Thu Jan  1 00:10:51 1970 daemon.debug hostapd: nl80211: send_mlme -> send_frame_cmd
```

brada4, 2024-03-18:
For 160 you need to specify caps tdirectly to use less antennas (STS=numantennas-1)
```
Beamformee STS <= 80Mhz: 7
				Beamformee STS > 80Mhz: 3
```
If you tried fallback 160 channels it would cut in right away.

testuser7, 2024-03-18:
I think that configuration file is incorrect for 160MHz and channel 149.
The value of parameters `vht_oper_centr_freq_seg0_idx` and `he_oper_centr_freq_seg0_idx` should be `163`, not `149`.

testuser7, 2024-03-18:
Fix: https://github.com/openwrt/openwrt/pull/14926

brada4, 2024-03-18:
It is one off in narower case with odd numbers to be picky
actually everything works omitting that parameter altogether.

wlshdjj, 2024-03-19:
> Fix: #14926
Fixed:
				case "$channel" in
					36|40|44|48|52|56|60|64) idx=50;;
					100|104|108|112|116|120|124|128) idx=114;;
					149|153|157|161|165|169|173|177) idx=163;;
				esac

But it still unenable to start.:
Tue Mar 19 18:17:21 2024 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Mar 19 18:17:21 2024 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Tue Mar 19 18:17:21 2024 daemon.notice netifd: Wireless device 'radio1' is now up
Tue Mar 19 18:17:22 2024 daemon.err hostapd: nl80211: kernel reports: (extension) channel is disabled
Tue Mar 19 18:17:22 2024 daemon.err hostapd: Could not set channel for kernel driver
Tue Mar 19 18:17:22 2024 daemon.err hostapd: Interface initialization failed
Tue Mar 19 18:17:22 2024 daemon.notice hostapd: wlan1: interface state HT_SCAN->DISABLED
Tue Mar 19 18:17:22 2024 daemon.notice hostapd: wlan1: AP-DISABLED

testuser7, 2024-03-19:
> But it still unenable to start.: Tue Mar 19 18:17:21 2024 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE Tue Mar 19 18:17:21 2024 daemon.notice hostapd: wlan1: interface [...]

Can you show this command output (change `X` with your interface number):
`iwinfo phyX freqlist`
?

brada4, 2024-03-19:
> channel is disabled

those channels are fresh, only allowed in US and very few other places (not yet EU/CN/IN)

commit a91b79fd (2024-03-23): mac80211: add missing config for third 160MHz width for 5GHz radio

Without this configuration it is not possible to run the radio using HE160 on channels 149-177.

Fixes: #14906
commit e82759fd (2024-07-08): mac80211: add missing config for third 160MHz width for 5GHz radio

Without this configuration it is not possible to run the radio using HE160 on channels 149-177.

Fixes: #14906

(cherry picked from commit a91b79fd04d58e711273d08e0b1246942b2eec98)
Link: https://github.com/openwrt/openwrt/pull/15898
commit 3122bb60 (2024-07-08): mac80211: add missing config for third 160MHz width for 5GHz radio

Without this configuration it is not possible to run the radio using HE160 on channels 149-177.

Fixes: #14906

(cherry picked from commit a91b79fd04d58e711273d08e0b1246942b2eec98)
Link: https://github.com/openwrt/openwrt/pull/15899
closed 2024-03-23 by openwrt-bot