openwrt/openwrt #14921: TP-Link C2600: Sysupgrade broken?

flipreverse, 2024-03-17:
### Describe the bug

I'm currently unable to flash a new OpenWRT image via sysupgrade. Flashing via TFTP method works.

### OpenWrt version

r25530-51752fa14b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Self-built image

### Steps to reproduce

- Upload an image to, for example, `/tmp/`
- Run `sysupgrade /tmp/openwrt-ipq806x-generic-tplink_c2600-squashfs-sysupgrade.bin`
- Wait for device to be up and running

### Actual behaviour

The old firmware version previously installed is running.

### Expected behaviour

My new, self-built version should be booted.

### Diffconfig

```text
CONFIG_TARGET_ipq806x=y
CONFIG_TARGET_ipq806x_generic=y
CONFIG_TARGET_ipq806x_generic_DEVICE_tplink_c2600=y
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH10K_LEDS=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BUILD_LOG=y
CONFIG_BUILD_PATENTED=y
# CONFIG_BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT is not set
CONFIG_BUSYBOX_CONFIG_BUNZIP2=y
CONFIG_BUSYBOX_CONFIG_BZCAT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_BZIP2_DECOMPRESS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_COMPRESS_USAGE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CROND_CALL_SENDMAIL=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CROND_D=y
CONFIG_BUSYBOX_CONFIG_FEATURE_DEFAULT_PASSWD_ALGO="md5"
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVEHISTORY=y
# CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_MMIN is not set
# CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_NEWER is not set
CONFIG_BUSYBOX_CONFIG_FEATURE_MIME_CHARSET=""
CONFIG_BUSYBOX_CONFIG_FEATURE_MOUNT_FAKE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_MOUNT_LABEL=y
CONFIG_BUSYBOX_CONFIG_FEATURE_MOUNT_VERBOSE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_PREFER_IPV4_ADDRESS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_REVERSE_SEARCH=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SHOW_THREADS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_STAT_FILESYSTEM=y
CONFIG_BUSYBOX_CONFIG_FEATURE_STAT_FORMAT=y
[... 400 lines cut ...]
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_xtables-legacy=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCAP_HAS_NETFILTER=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PCRE_JIT_ENABLED=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_WOLFSSL_HAS_ECC25519 is not set
# CONFIG_WOLFSSL_HAS_OPENVPN is not set
CONFIG_WPA_MBO_SUPPORT=y
# CONFIG_LIBCURL_UNIX_SOCKETS is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-hwmon-core=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-18:
> CONFIG_BIND_ENABLE_DOH

check the size of image, you have 27mb of space after system partitions for kernel rootfs overlay.

flipreverse, 2024-03-18:
> > CONFIG_BIND_ENABLE_DOH
> 
> check the size of image, you have 27mb of space after system partitions for kernel rootfs overlay.

I'll do. Does this, however, explain why I can still flash via TFTP method?

brada4, 2024-03-18:
Can you try default image, like `auc -f`?

flipreverse, 2024-03-18:
> Can you try default image, like `auc -f`?

Attended sysupgrade isn't installed. I'll try the default image from the website

brada4, 2024-03-18:
Yes, try default image, then move default packages to squashfs, then install any custom thing you may have.
Could be 16MB limit in sysupgrade, but I cant locate that in sources easily.

hardfalcon, 2024-04-09:
Possible duplicate of https://github.com/openwrt/openwrt/issues/14931.

flipreverse, 2024-04-09:
> Yes, try default image, then move default packages to squashfs, then install any custom thing you may have. Could be 16MB limit in sysupgrade, but I cant locate that in sources easily.

I've successfully flashed the [OpenWRT factory image](https://downloads.openwrt.org/releases/23.05.2/targets/ipq806x/generic/openwrt-23.05.2-ipq806x-generic-tplink_c2600-squashfs-factory.bin) linked at the device's [page](https://openwrt.org/toh/tp-link/archer_c2600_v1) via tftp.  Afterwards I also flashed my custom image via the webinterface. I'll keep you posted whether further flash attempts via sysupgrade/webinterface succeed.

flipreverse, 2025-01-18:
I don't what has changed over the months. Sysupgrade is working again. I can flash my C2600 without any issue.

closed 2025-01-18 by flipreverse