openwrt/openwrt #14923: 树莓派3B无法刷入sysupgrade.img

wangxue415, 2024-03-18:
### Describe the bug

我使用lede仓库自行编译了树莓派3B的固件，并刷入tfcard，成功启动，但是当尝试刷入sysupgrade.img固件时，出现错误，如下
root@OpenWrt:/tmp# sysupgrade openwrt-bcm27xx-bcm2710-rpi-3-squashfs-sysupgrade.img
Sun Mar 17 18:23:03 CST 2024 upgrade: Image metadata not present
Sun Mar 17 18:23:03 CST 2024 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware
Sun Mar 17 18:23:03 CST 2024 upgrade: Reading partition table from bootdisk...
Sun Mar 17 18:23:03 CST 2024 upgrade: Reading partition table from image...
Image check failed.
随后我使用-F选项，成功刷入并启动。
我看到其它Issues提到使用-F选项可以刷入，但原因是跨版本升级，我这个不是升级版本，想知道原因。

### OpenWrt version

R24.2.2

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2710

### Device

Raspberry Pi 3 Model B Rev 1.2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

fnoopv, 2024-03-18:
24.2.2 ?

catsimple, 2024-03-18:
神经，lede仓库编译的固件，跑来openwrt仓库发issue

wangxue415, 2024-03-18:
抱歉，搞错了

closed 2024-03-18 by wangxue415