openwrt/openwrt #14931: sysupgrade -b produces invalid tar.gz file

brackenhill-mob, 2024-03-19:
### Describe the bug

sysupgrade -b produces a non-zero byte tar.gz file, but trying to extract it gives "tar: invalid tar magic" as an error message.

There are a number of revisions to sysupgrade between r25567 and r25277 (the previous version I used). r25277 version works as expected

### OpenWrt version

r25567-0dfc0495fc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

Dell Inc. OptiPlex 3020

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brackenhill-mob, 2024-03-19:
I know that. If you gunzip the .gz file to get the tar and then try and untar that file it fails.

The bug is in sysupgrade not creating a tar file correctly, making the .gz file invalid. As I said in my original post, the sysupgrade from revision r25277 works as expected.

This is a major bug as all backup configuration .gz files created in LUCI are also invalid/corrupt.

jow-, 2024-03-19:
Please provide such a broken file so that we can take a look at it. A quick test with https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-x86-64-generic-squashfs-combined.img.gz  (r25589-f84ed09d2c) does not show the problem:

```
jow@j7:~$ ssh root@192.168.1.1 'sysupgrade -b -' | tar -tz
Warning: Permanently added '192.168.1.1' (ED25519) to the list of known hosts.
etc/uci-defaults/10_disable_services
etc/config/dhcp
etc/config/dropbear
etc/config/firewall
etc/config/network
etc/config/system
etc/dropbear/dropbear_ed25519_host_key
etc/dropbear/dropbear_rsa_host_key
etc/group
etc/hosts
etc/inittab
etc/nftables.d/10-custom-filter-chains.nft
etc/nftables.d/README
etc/opkg/keys/b5043e70f9a75cde
etc/passwd
etc/profile
etc/rc.local
etc/shadow
etc/shells
etc/shinit
etc/sysctl.conf
jow@j7:~$ 
```

You mentioned self-built image, please attach the diffconfig output as requested by the issue template.

brackenhill-mob, 2024-03-20:
The output of diffconfig is below (couldn't do it originally as our build scripts don't keep the build folders once the images are created). Hope it looks ok as preview removes line breaks even though I've put it in a code block.

`
can't find file Config.in
can't open file 'tmp/.diffconfig.stage1' at ./scripts/kconfig.pl line 32.
can't find file Config.in
can't open file 'tmp/.diffconfig.stage2' at ./scripts/kconfig.pl line 32.
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_x86_64_DEVICE_generic=y
CONFIG_TARGET_DEVICE_PACKAGES_x86_64_DEVICE_generic=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
`

Just tried again with r25589 and get the same bad backup file. The config backup was created with the command `sysupgrade -b backup.tar.gz`.

However ...

On a whim, I replaced my working /etc with the clean /etc in the new system tar.gz and ran sysupgrade again. This time it created a valid tar.gz file so I need to investigate my working /etc folder to see if I can identify what is making sysupgrade barf.

What is odd is that if I run `tar -cf backup.tar \`sysupgrade -l\` && gzip backup.tar` I get a valid tar.gz with my working /etc folder (this is what I've been doing once I found this issue) The \ in the command should be ` but despite escaping the back tick it doesn't display properly in preview

brackenhill-mob, 2024-03-20:
I'll report back when I figure out what is culprit in my /etc folder so please don't close this for the moment

brackenhill-mob, 2024-03-20:
OK solved it. And it's really weird

I have a /etc/init.d script for the MiroTalk service that we run on the router. As a test, I removed this script, and sysupgrade -b created a valid tar.gz.

This script has a case...esac block in it with an echo statement in the *) catchall bit. When I remove this echo line, I can produce a valid .tar.gz file. I can put an echo statement in anywhere else in the case block, just not in this *) bit which is where it needed. The line is ` echo "Usage: start|stop|restart|status"`, but any echo line (ie `echo test`) causes the invalid file issue. As I said weird!

At some point since r25277,  sysupgrade appears to have started to parse /etc/init.d scripts. This MiroTalk script has not changed in >12 months and I've always been able to create a backup successfully until now (my installation scripts rely on it being a valid file)

I'd love to know the thinking behind this change. @jow- ?

brackenhill-mob, 2024-03-21:
Thanks @jow- that's very helpful to me. I wrote my first init.d script over 20 years ago, and have clearly not kept up with the times with respect to procd! Time to brush up on my skills ;)

I've attached my poorly written script so that you and @rmilecki have something that produces a known error to test against. HTH.

If I may, I'll let you close this when it's appropriate to do so.

And many thanks for your help.
[mirotalk.tar.gz](https://github.com/openwrt/openwrt/files/14693054/mirotalk.tar.gz)

zhouzhiwen2000, 2024-07-22:
Any progress?

leelx78, 2024-11-09:
Guys any workaround for this? This is a MAJOR bug for me and I don't know how to backup and restore from ssh.

Ideas? Can I create the tar file manually? I did a lot of trial-and-error but nothing worked so far

t0saki, 2024-12-03:
This issue also troubles me. Both the tar file generated by web UI and `sysupgrade -b` are considered invalid. For now, I use the command `sysupgrade -l | tar -czf backup.tar.gz -T -` to back up my configuration (which should be similar to the previously mentioned `tar -cf backup.tar \`sysupgrade -l\` && gzip backup.tar`). However, I haven’t tested whether this backup can be effectively restored. Hopefully this issue will be resolved someday.

idarek, 2025-07-03:
Facing the same issue
```
OpenWrt 24.10.2 r28739-d9340319c6 / LuCI openwrt-24.10 branch 25.179.72978~e977ed6
```

Thanks @t0saki for the command `sysupgrade -l | tar -czf backup.tar.gz -T -` that will allow me to get a backup that I can use until this is permanently sorted.

My discussion that lead me back to this issue:
https://forum.openwrt.org/t/extracting-unpack-backup-file-in-openwrt-24-10-series/235434/15