openwrt/openwrt #14932: Fw4 not working with VLAN (802.1q)  after installing dockerd

dc-me, 2024-03-19:
### Describe the bug

I'v created a few vlans with DSA,  firewall is working fine as expected, after installed dockerd, it's not working, with the vlan the package actually come from the actual device, not the vlan bridge, thus the firewall is not working, with the defualt global forward not set to accept, the lan client can't ping to each other eventhough the lan zone has set forward to accept. the cause is that it's not treat the package from the vlan rather from an actual device.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

With  everthing default, create few vlans, set all vlans forward to accept, but global as default reject, vlans internal client can ping right now, install dockerd, then test it again, firewall no working as expect for the vlans.

### Actual behaviour

vlans firewall work as before not installing dockerd

### Expected behaviour

vlans firewall are as before without dockerd.

### Additional info

firewall not treat vlans as vlans, the package actually come from a device. makes it not working without allow global forward, not even with layer 2 communications.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-19:
Attach:
`nft list ruleset`
`fw4 print`
`/etc/config/firewall`
`/etc/config/network`
`iptables-nft-save`
`iptables-legacy-save`  (ideally empty or better not installed)

You should clear out mid parts of MAC addresses and public IPs.

brada4, 2024-03-19:
The zone forward is something else, bridge forward filtering is possible but out of context of fw4.

dc-me, 2024-03-19:
I don't have any special firewall setup, just the default, and the added some vlans, all the vlans forward are working, after installing docker, it's not working, I think it has something with iptables as docker depends on this.

brada4, 2024-03-19:
Can you help with info:
- the "some vlans" as list (i can dig that from config files too)
- the ruleset and the stuff added by iptables-nft from 1st commands
- assurance iptables-legacy is NOT mixed in.

dc-me, 2024-03-19:
[image]
[image]
[image]

dc-me, 2024-03-19:
I don't have other rules that's reject or drop, all are accept, everything is default, The client within vlan can't ping to each other, without installing docker, everything is ok.

dc-me, 2024-03-19:
It's very easy to replicate the issue, with everything deafult, create some vlans test vlan clients ping, it's ok, now install dockerd, it broke, it's not working, what's working is that if the interface is still bridge, then it's fine, firewall will hornor the config, but if it's vlan with DSA, it borke.

brada4, 2024-03-19:
You should not use vlan1 with DSA.
nor legacy iptables. At present default-forward-drop just drops any nat or masq packet mangled by other tables.
nor ever add wifi interface to a bridge port. hostapd does it by itself.
Once all misconfigurations above are addressed feel free to install docker again.
It is imperative to attach text files, compressing if bigger than a screen.

dc-me, 2024-03-19:
legacy iptables is a dockerd dependency, I don't think is a misconfiguration, as if I don't install dockerd everything is ok.

brada4, 2024-03-20:
You should pre-install iptables-nft and ip6tables-nft o avoid -zz-legacy packages. It was reported at least once in ../packages. 
https://github.com/openwrt/packages/blob/d8bd171f838a4b1d2a3861ec6138ec1b9e2b3fca/utils/dockerd/Makefile#L39

Please re-configure network by:
a. adding network to wifi ap definitions and removing wifi ap-s from bridge membership. 
b. eliminating vlan ID 1
c. pre-installing ip?tables-nft

backup config for yourself  ;-)

Now install docker and if it is broken extract 6 files. fw4 and config/network will contain IPs, just type anything over those.

In principle this should have been reported in packages tracker, but you got wild misconfigurations, to the point it is fairly expected packets are lost in unpredictable places.

dc-me, 2024-03-20:
After did everything you said, it's still the same, after installing docker the local lan client can't ping each other, I've attached a custom uci-default that can replicate the issues, the vlan 1 I haven't changed it, but I've tested it, still not working, By the way I did should let the hostapd join the bridge, but after fix still not working, Also why vlan1 can't be used, I changed it still does nothing.
[99-custom.txt](https://github.com/openwrt/openwrt/files/14671129/99-custom.txt)

dc-me, 2024-03-20:
[image]
This is the wireshark data packet from the wifi device wl1-ap1 which belongs to 192.168.1.0/24 subnet, the local network should work at the layer 2 right? what can't they ping each other ? the can ping the gateway, but if I set the global forward to accept is working, eventhough the lan zone has forword to accept, but it shouldn't go forward, it's the local network should act like a switch right?

brada4, 2024-03-20:
Install `ip-bridge` and check `bridge vlan` , really "should work" depends on dual racing configurations.

dc-me, 2024-03-20:
[image]
looks normal

brada4, 2024-03-20:
vlan30 is both tagged and untagged at br-lan. They are isolated, so depending on luck packets arrive in one or other place at br-lan. It takes editing one out from /etc/config/network and courage to restart/reboot after. 

Run 
`fw4 print > /tmp/fw4.out`
and
`nft list ruleset > /tmp/nftables.out`
then draw diff from them:
`diff -w /tmp/fw4.out /tmp/nftables.out`

post that as text section
\`\`\`diff
```diff
+2
-3
```
\`\`\`

It should show docker additions to nftables rules - more or less NAT from docker0 bridge to wan and incoming "exposed" ports from all ifaces. Question is if "all interfaces" got poisoned with dsa parent and/or something else misfired.

dc-me, 2024-03-22:
Find the culprit see the pic below:
[image]
[image]
```
# Do not edit, changes to this file will be lost on upgrades
# /etc/sysctl.conf can be used to customize sysctl settings

# disable bridge firewalling by default
net.bridge.bridge-nf-call-arptables=0
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0
# Do not edit, changes to this file will be lost on upgrades
# /etc/sysctl.conf can be used to customize sysctl settings

# enable bridge firewalling for docker
net.bridge.bridge-nf-call-ip6tables=1
net.bridge.bridge-nf-call-iptables=1
```

brada4, 2024-03-22:
You have not yet provided anything resembling firewall rule. dual vlan30 your way guarantees network glitch even plugging cable.

brada4, 2024-03-22:
also you need to pre-install arptables-nft and ebtables-nft, for same effect as dual hooks when legacy versions are used.

dc-me, 2024-03-22:
Actually openwrt's default 
net.bridge.bridge-nf-call-arptables=0
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0
is just not make bridge traffic go through firewall, which defeats the firewall setting of forward(the bridge forward input interface and output interface being the same)

after installing docker it enables the bridge traffic goes through firewall, and thus should make the zone forwrad work but here's the thing, I add a log to forward chain with: sudo nft add rule inet fw4 forward log, The log shows:
[image]

**It should come from br-lan.valn** But it's comming form br-lan !!!!!

brada4, 2024-03-22:
Replace picture with text, I am afraid without context packets legitimately sink in your misconfigured bridge.

dc-me, 2024-03-22:
@brada4 What do you mean by dual vlan 30, I don't have dual vlan 30
[image]
[image]

brada4, 2024-03-22:
[image]
Frankly no idea why your config is nibbling under the keyboard. Attach config files and lets start from the beginning.

dc-me, 2024-03-22:
Did you test on a fresh install openwrt and create some vlans using DSA and enable net.bridge.bridge-nf-call-ip6tables=1
net.bridge.bridge-nf-call-iptables=1 (just as dockerd did) does it work with any vlan clients that can ping each other? I don't think this is related to my config, it yours config as I said, and works, then could be something else, otherwise is a bug that firewall loged input interface is **_br-lan_** and not **_br-lan.vlanid_** , thus makes it not working right for the luci firewall zone configs.

brada4, 2024-03-22:
I dont get net.bridge structure and dockers' creativity runs short.
i install 5x ????tables-nft
I config 2 vlans
I install docker
install hello-world
nothing in ebtables level.

Install all tables-zz-legacy
Install docker
System console works, config looks sane but nothing works across br-lan....
After reboot it works quite sporadically, like some lost pings between VM-s at times.

We are not trading about your configs. Your picture shows vlan30 being tagged and untagged at br-lan. One works at any point, so other does not and small state change flips the switch. You show obvious misconfiguration that  can isolate bridge hosts on its own, it points fingers off docks.

brada4, 2024-03-22:
(and very bad UX if default dependencies is so messed up)

dc-me, 2024-04-03:
Some other references: https://netdevconf.info/1.1/proceedings/papers/Bridge-filter-with-nftables.pdf
https://openwrt.org/docs/guide-user/firewall/fw3_configurations/bridge

brada4, 2024-04-03:
??? 
any chance to see your nftables ruleset? 
In all levels of iptables and nftables absence of rules means accept.

dc-me, 2024-04-03:
The clients within a vlan can not ping each other, setting the zone forward to accept does not do anything, because of 
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0
And even after docker turns on it's still not gonna work because the iifname is br-lan and it's not br-lan.vlanid, so makes me thinking that the zone forwrding has no use case, FORWARD rules for a zone describe what happens to traffic passing between different interfaces belonging in the same zone.
Not the forwarding which like forward to another zone like wan.

brada4, 2024-04-03:
@jow-  i am puzzled why xt ct slips into docker table while normal ct is fine in fw4

brada4, 2024-04-04:
addrtype needs kmod-nft-fib
masquerade needs kmod-nft-nat (it is working in srcnat chain but not in dock???)
same about xt match conntrack - ct state match in other rule looks ok.

The very default docker ie bridge called docker0 with nat and exposed ports certainly converts to nftables by iptables-nft.