openwrt/openwrt #14936: Ramips Targets Are Not Building After Kernel Version 6.1

khanjui, 2024-03-20:
### Describe the bug

See discussion: https://github.com/openwrt/openwrt/issues/14546#issuecomment-2002074867

### OpenWrt version

r25519-7ea82bc17d

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Steps to reproduce

Latest SNAPSHOT for mt7621 is from 2024-03-12 11:22:23

### Expected behaviour

Should build after getting fixed.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-03-20:
Only mt7620 compiles correctly, the remaining ones, i.e. mt7621, mt76x8, rt288x, rt305x, rt3883show compilation errors.

predators46, 2024-03-20:
it's normal if the firmware size is too large

namiltd, 2024-03-20:
Trying to find the cause, I locally compiled routers that show errors (e.g. tplink_re350-v1) and the local build had no errors.

namiltd, 2024-03-20:
The issue should not be closed yet because only one subtarget, my7621, has been fixed. mt76x8, rt288x, rt305x and rt3883 still not working.

namiltd, 2024-03-22:
I just generated a PR https://github.com/openwrt/openwrt/pull/14958 to enable building with kernel 6.1.

hauke, 2024-03-23:
Fixed in bce7b4f8e5bc01fa29ca45db88bb55bb0d301265

closed 2024-03-23 by hauke