openwrt/openwrt #14941: Ubiquiti Rocket M2 XM Settings lost after reboot

mucke1985, 2024-03-20:
### Describe the bug

The flash driver is unable to erase or write the flash chip.
The Rocket was sold with two different flash memories and mine had a slow memory installed.

With the software version OpenWrt 19.07.7 r11306-c4a6851c72 everything still works perfectly, but with the current version the problem arises.

Others in the forum have already reported the problem. -> https://forum.openwrt.org/t/settings-lost-on-reboot-ubiquiti-rocketm2/179955/5

### OpenWrt version

qwertz

### OpenWrt release

openwrt-23.05.0-ath79-generic-ubnt_rocket-m-squashfs-sysupgrade

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti Rocket M

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-20:
Can you attach `dmesg` output from broken version in safe mode (and fix version tags in post at same effort)

mucke1985, 2024-03-21:
> Can you attach `dmesg` output from broken version in safe mode (and fix version tags in post at same effort)

root@OpenWrt:/# dmesg
[    0.000000] Linux version 5.15.134 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0      r23497-6637af95aa) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Mon Oct 9 21:45:35 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti Rocket M (XM)
[    0.000000] SoC: Atheros AR7241 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55976K/65536K available (6077K kernel code, 591K rwdata, 780K rodata, 1184K init, 2     17K bss, 9560K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9801335621 ns
[    0.000002] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012737021ns
[    0.008017] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084180] pid_max: default: 32768 minimum: 301
[    0.090446] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097744] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.116264] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627500     00 ns
[    0.126137] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.133223] pinctrl core: initialized pinctrl subsystem
[    0.141660] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.148610] thermal_sys: Registered thermal governor 'step_wise'
[    0.169288] clocksource: Switched to clocksource MIPS
[    0.182848] NET: Registered PF_INET protocol family
[    0.188121] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.197032] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.205458] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.213213] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.220883] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.227932] TCP: Hash tables configured (established 1024 bind 1024)
[    0.234540] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.241165] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.248777] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.254560] PCI: CLS 0 bytes, default 32
[    0.263975] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.280532] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.286355] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat,      Inc.
[    0.298509] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.309836] pinctrl-single 18040028.pinmux: 64 pins, size 8
[    0.317154] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.328019] printk: console [ttyS0] disabled
[    0.332481] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 12187500) is a 16550A
[    0.341209] printk: console [ttyS0] enabled
[    0.349612] printk: bootconsole [early0] disabled
[    0.388189] spi-nor spi0.0: mx25l6405d (8192 Kbytes)
[    0.393376] 6 fixed-partitions partitions found on MTD device spi0.0
[    0.399861] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.406556] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.413762] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.420535] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.427995] Creating 6 MTD partitions on "spi0.0":
[... 110 lines cut ...]
[  103.679392] jffs2: Newly-erased block contained word 0x19852003 at offset 0x00120000
[  103.714322] jffs2: Newly-erased block contained word 0x19852003 at offset 0x00110000
[  103.754670] jffs2: Newly-erased block contained word 0x19852003 at offset 0x00100000
[  103.779365] jffs2: Newly-erased block contained word 0x19852003 at offset 0x000f0000
[  103.804753] jffs2: Newly-erased block contained word 0x19852003 at offset 0x000e0000
[  103.834700] jffs2: Newly-erased block contained word 0x19852003 at offset 0x000d0000
[... cut ...]
[  104.154682] jffs2: Newly-erased block contained word 0x19852003 at offset 0x00010000
[  104.184708] jffs2: Newly-erased block contained word 0xdeadc0de at offset 0x00000000
[  104.199848] done.
[  104.201827] jffs2: notice: (2032) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of      xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[  106.256699] br-lan: port 1(eth0) entered blocking state
[  106.262094] br-lan: port 1(eth0) entered disabled state
[  106.267818] device eth0 entered promiscuous mode
root@OpenWrt:/#

mucke1985, 2024-03-21:
> Can you attach `dmesg` output from broken version in safe mode (and fix version tags in post at same effort)

Unfortunately the tags were not included in the overview.

brada4, 2024-03-21:
Try 23.05.2, it has some mediatek flash driver fixes.

mucke1985, 2024-03-21:
> Try 23.05.2, it has some mediatek flash driver fixes (not atheros but may be in right place)

This version also has the same effect

root@OpenWrt:/etc/config# dmesg
[    0.000000] Linux version 5.15.137 (builder@buildhost) (mips-openwrt-linux-mu             sl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40             .0) #0 Tue Nov 14 13:38:11 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is Ubiquiti Rocket M (XM)
[    0.000000] SoC: Atheros AR7241 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32              bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jff             s2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, lin             ear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, line             ar)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55912K/65536K available (6079K kernel code, 597K rwdata,              780K rodata, 1240K init, 217K bss, 9624K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 390.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_i             dle_ns: 9801335621 ns
[    0.000002] sched_clock: 32 bits at 195MHz, resolution 5ns, wraps every 11012             737021ns
[    0.008021] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.084177] pid_max: default: 32768 minimum: 301
[    0.090432] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linea             r)
[    0.097757] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes,              linear)
[    0.116230] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, ma             x_idle_ns: 19112604462750000 ns
[    0.126110] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.133198] pinctrl core: initialized pinctrl subsystem
[    0.141621] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.148568] thermal_sys: Registered thermal governor 'step_wise'
[    0.169183] clocksource: Switched to clocksource MIPS
[    0.182741] NET: Registered PF_INET protocol family
[    0.188012] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear             )
[    0.196914] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096              bytes, linear)
[    0.205343] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes,              linear)
[    0.213097] TCP established hash table entries: 1024 (order: 0, 4096 bytes, l             inear)
[    0.220758] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.227801] TCP: Hash tables configured (established 1024 bind 1024)
[    0.234414] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.241038] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.248644] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.254417] PCI: CLS 0 bytes, default 32
[    0.263824] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.280225] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.286053] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORIT             Y) (c) 2001-2006 Red Hat, Inc.
[    0.298203] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 2             51)
[    0.309527] pinctrl-single 18040028.pinmux: 64 pins, size 8
[    0.316830] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.327700] printk: console [ttyS0] disabled
[    0.332169] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 121             87500) is a 16550A
[    0.340888] printk: console [ttyS0] enabled
[    0.349293] printk: bootconsole [early0] disabled
[    0.387892] spi-nor spi0.0: mx25l6405d (8192 Kbytes)
[    0.393073] 6 fixed-partitions partitions found on MTD device spi0.0
[    0.399555] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.406256] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[    0.413472] OF: Bad cell count for /ahb/spi@1f000000/flash@0/partitions
[... 100 lines cut ...]
[   18.753700] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   18.757620] ieee80211 phy0: Atheros AR9280 Rev:2 mem=0xb0000000, irq=13
[   18.857480] kmodloader: done loading kernel modules from /etc/modules.d/*
[   23.591667] urngd: v1.0.2 started.
[   28.479289] random: crng init done
[   28.482738] random: 31 urandom warning(s) missed due to ratelimiting
[... cut ...]
[   96.409030] jffs2: notice: (2038) jffs2_build_xattr_subsystem: complete build             ing xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead,              0 orphan) found.
[   96.501103] eth0: link up (100Mbps/Full duplex)
[   96.505738] br-lan: port 1(eth0) entered blocking state
[   96.511042] br-lan: port 1(eth0) entered forwarding state
[   96.569388] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   97.582027] overlayfs: upper fs does not support tmpfile.
root@OpenWrt:/etc/config# ^C

root@OpenWrt:/etc/config# [  338.819552] eth0: link down
[  338.822833] br-lan: port 1(eth0) entered disabled state

brada4, 2024-03-21:
It got a bit ahead,try to reset again. Seems some frequency or delay is wrong that kernel reads block too early while flash controller is still erasing it.

mucke1985, 2024-03-22:
> It got a bit ahead,try to reset again. Seems some frequency or delay is wrong that kernel reads block too early while flash controller is still erasing it.

I made this entry so that someone would take care of correcting the Official downloaded image.

Unfortunately I can't do this myself.

In any case, it's very bad when an official image doesn't fit a device, and someone who isn't an "expert" wants to flash a router for the first time and "destroys" their device in the process and After he "destroyed" his device he says OpenWRT sucks.
I believe that this is not in the spirit of the OpenWRT community.

brada4, 2024-03-22:
Yours is far from destroyed. At two boots - does flash erase error appear at same address twice  in a row indicating write protect or it nibblezs indicating clock issue

mucke1985, 2024-03-22:
> Yours is far from destroyed. At two boots - does flash erase error appear at same address twice in a row indicating write protect or it nibblezs indicating clock issue

 don't know what to do next and I don't think I can solve the problem either. Something probably changed before compiling?

Unfortunately, I'm not a programmer, but I think for someone who has a better idea of ​​where the speed needs to be changed, it can be done immediately

brada4, 2024-03-22:
https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset

perform firstboot reset in failsafe mode.
tell if message
```
jffs2: Newly-erased block contained word 0xdeadc0de at offset 0x0 0000000
```
changes after.
Y: speed
N: ro

Try v21.02.7 and 22.03.6 to try to pinpoint when the problem was introduced.

mucke1985, 2024-03-23:
> https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset
>
> perform firstboot reset in failsafe mode. tell if message
>
> ```
> jffs2: Newly-erased block contained word 0xdeadc0de at [...]

I don't understand this instruction -> changes after. Y: speed N: ro

This flash memory is installed in my device ->mx25l6408emi-12g
Datasheet -> https://www.macronix.com/Lists/Datasheet/Attachments/8618/MX25L6408E,%203V,%2064Mb,%20v1.4.pdf

Is it possible that I have the same problem that is described here? ->https://github.com/openwrt/openwrt/pull/3536

brada4, 2024-03-23:
If the errormessage changes after next reset it is a clocking problem.

mucke1985, 2024-03-24:
> https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset
>
> perform firstboot reset in failsafe mode. tell if message
>
> ```
> jffs2: Newly-erased block contained word 0xdeadc0de at [...]

OpenWrt 21.02.7, r16847-f8282da11e  -> Works well
OpenWrt 22.03.6, r20265-f85a79bcb4  -> Works well

stargieg, 2024-03-26:
I have the same problem with a "ubnt_litebeam-ac-gen2".
I have installed version 23.05.2 according to the instructions from 'https://openwrt.org/toh/ubiquiti/common#wa_v874xc_v874'.
Configuration changes do not survive the reboot.
I also cannot install new firmware with sysupgrade. Sysupgrade runs without error message.
At the moment I do not have a serial console connected.
dmesg mount df
https://gist.github.com/stargieg/f64a73e61599b5d08c0c9a566af23fec

mucke1985, 2024-03-28:
> If the errormessage changes after next reset it is a clocking problem.

Thanks for your support

Greetings

brada4, 2024-03-28:
You should not close issue which persists...

mucke1985, 2024-03-31:
> You should not close issue which persists...

I didn't close it

mucke1985, 2024-03-31:
> You should not close issue which persists...

sorry was unintentional

stargieg, 2024-04-17:
Now i have litebeam-ac-gen2 with the same problem on my desk with serial and tftp.
The initramfs 23.05.0 - 23.05.3 dose not work. 'Uncompressing Kernel Image ... ERROR: LzmaDecode.c, 543'
The initramfs 22.03.6 works.
sysupgrade 22.03.6 works.
sysupgrade 23.05.3 works.

load openwrt-23.05.3-ath79-generic-ubnt_litebeam-ac-gen2-initramfs-kernel.bin via tftp
```
ar7240> bootm 0x81000000
## Booting image at 81000000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.150
   Created:      2024-03-22  22:09:42 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    5624367 Bytes =  5.4 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum at 0x81000040 ...OK
   Uncompressing Kernel Image ... ERROR: LzmaDecode.c, 543

Decoding error = 1
LZMA ERROR 1 - must RESET board to recover

Resetting...

U-Boot 1.1.4-s1102 (Mar  5 2019 - 17:13:19)
```

LipkeGu, 2024-07-31:
`[ 0.387892] spi-nor spi0.0: mx25l6405d (8192 Kbytes)`

fixed by: #14237