openwrt/openwrt #14942: WAN does not connect

castillofrancodamian, 2024-03-20:
### Describe the bug

After updating the image with kernel 6.1 it is no longer possible to connect to the WAN.

### OpenWrt version

r25591-362f9c41a4

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

MERCUSYS MR70X v1

### Image kind

Self-built image

### Actual behaviour

This is what the log shows

```
Wed Mar 20 08:56:01 2024 daemon.notice netifd: Interface 'wan' is disabled
Wed Mar 20 08:56:01 2024 daemon.notice netifd: Interface 'wan' is enabled
Wed Mar 20 08:56:01 2024 kern.warn kernel: [   96.054419] mtk_soc_eth 1e100000.ethernet wan: validation of  with support 00000000,00000000,00006000 and advertisement 00000000,00000000,00000000 failed: -EINVAL
Wed Mar 20 08:56:01 2024 kern.err kernel: [   96.069218] mtk_soc_eth 1e100000.ethernet wan: mtk_open: could not attach PHY: -22
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-20:
Duplicate almost #14673

brada4, 2024-03-20:
Temporary workaround if you dont target full gigabit may be repurposing vacant lan port as wan- remove it from lan bridge, then transfer wan configuration to that port(or use release - whichever is viable)

castillofrancodamian, 2024-03-20:
@brada4 Unfortunately I have all 3 LAN ports used. I had to go back to 23.05.2.

castillofrancodamian, 2024-04-27:
Apparently fixed with https://github.com/openwrt/openwrt/commit/a8dde7e5bd6d289db6485cf57d3512ea62eaa827.

dangowrt, 2024-04-27:
> Self-built image

@castillofrancodamian Can you please confirm that the problem also exists in images built by the buildbots, ie. https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-mercusys_mr70x-v1-squashfs-sysupgrade.bin in your case? Because the issue could be that the PHY driver itself is missing in your (maybe modified?) kernel configuration, for example. 
Please also share **a complete bootlog** and **instructions on how to reproduce your image (ie. your `.config`)**.

rany2, 2024-04-27:
It looks like https://github.com/openwrt/openwrt/commit/fe9d2ccbc3d9dc0852df474f0e3cc07f93beae8b should resolve this issue but please correct me if I'm mistaken.

castillofrancodamian, 2024-04-27:
Image created with ImageBuilder

<details><summary>logread</summary>
<p>

```
root@mr70x:~# logread
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Early memory node ranges
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s18240 r8192 d22720 u49152
Fri Apr 26 23:19:12 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s18240 r8192 d22720 u49152 alloc=12*4096
Fri Apr 26 23:19:12 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Fri Apr 26 23:19:12 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00011310
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00011310
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] Memory: 118060K/131072K available (7718K kernel code, 632K rwdata, 1660K rodata, 1240K init, 232K bss, 13012K reserved, 0K cma-reserved)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000]       Tracing variant of Tasks RCU enabled.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] NR_IRQS: 256
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.008027] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.066220] pid_max: default: 32768 minimum: 301
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.071670] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.078880] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.089754] cblist_init_generic: Setting adjustable number of callback queues.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.096971] cblist_init_generic: Setting shift to 2 and lim to 1.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.103466] rcu: Hierarchical SRCU implementation.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.108195] rcu:  Max phase no-delay instances is 1000.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.114100] smp: Bringing up secondary CPUs ...
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.119428] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.119453] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Apr 26 23:19:12 2024 kern.warn kernel: [    0.119468] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.119515] CPU1 revision is: 0001992f (MIPS 1004Kc)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.178803] Synchronize counters for CPU 1: done.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.210760] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.210779] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Apr 26 23:19:12 2024 kern.warn kernel: [    0.210791] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.210822] CPU2 revision is: 0001992f (MIPS 1004Kc)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.270005] Synchronize counters for CPU 2: done.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.300626] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.300645] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Apr 26 23:19:12 2024 kern.warn kernel: [    0.300656] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.300690] CPU3 revision is: 0001992f (MIPS 1004Kc)
Fri Apr 26 23:19:12 2024 kern.info kernel: [    0.355190] Synchronize counters for CPU 3: done.
[... cut ...]
ucode-mod-nl80211 - 2024.04.07~5507654a-r1
ucode-mod-rtnl - 2024.04.07~5507654a-r1
ucode-mod-ubus - 2024.04.07~5507654a-r1
ucode-mod-uci - 2024.04.07~5507654a-r1
ucode-mod-uloop - 2024.04.07~5507654a-r1
uhttpd - 2023.06.25~34a8a74d-r3
uhttpd-mod-ubus - 2023.06.25~34a8a74d-r3
urandom-seed - 3
urngd - 2023.11.01~44365eb1-r1
usign - 2020.05.23~f1f65026-r1
wifi-scripts - 1.0-r1
wireguard-tools - 1.0.20210914-r3
wireless-regdb - 2024.01.23-r1
wpad-mbedtls - 2024.03.09~695277a5-r1
xtables-nft - 1.8.8-r2
zlib - 1.3.1-r1
```

</p>
</details>

dangowrt, 2024-04-27:
Thank you for all the details. Please try to access to PHYs using the `mdio-tools` package and share the output.
Ie. build or generate an image which includes the `mdio-tools` package, then to
```
mdio mdio-bus
```
or
```
mdio mt7530-0
```
should both show you the PHYs from address 0~5 with PHY-ID `0x03a29412`. If that's *not* the case for the PHY at address 0, then we got a smoking gun and shall investigate from there. If the PHYs are all present, something else must be weird, but then we already got the tools at hand to dig further.

castillofrancodamian, 2024-04-28:
@dangowrt New self-build.

<details><summary>openwrt-ramips-mt7621-mercusys_mr70x-v1.manifest</summary>
<p>

```
6in4 - 28
6to4 - 13
avahi-dbus-daemon - 0.8-r9
base-files - 1590~6e5edec159
busybox - 1.36.1-r1
ca-bundle - 20240203-r1
cgi-io - 2022.08.10~901b0f04-r21
collectd - 5.12.0-r50
collectd-mod-cpu - 5.12.0-r50
collectd-mod-cpufreq - 5.12.0-r50
collectd-mod-exec - 5.12.0-r50
collectd-mod-interface - 5.12.0-r50
collectd-mod-iwinfo - 5.12.0-r50
collectd-mod-load - 5.12.0-r50
collectd-mod-memory - 5.12.0-r50
collectd-mod-network - 5.12.0-r50
collectd-mod-rrdtool - 5.12.0-r50
collectd-mod-sensors - 5.12.0-r50
collectd-mod-sqm - 5.12.0-r50
collectd-mod-wireless - 5.12.0-r50
curl - 8.7.1-r2
dbus - 1.14.10-r1
ddns-scripts - 2.8.2-r43
ddns-scripts-services - 2.8.2-r43
dnsmasq - 2.90-r2
dropbear - 2022.83-r1
etherwake - 1.09-r5
firewall4 - 2023.11.03~698a5335-r1
fstools - 2024.01.22~08cd7083-r1
fwtool - 2019.11.12~8f7fe925-r1
getrandom - 2024.04.26~85f10530-r1
hostapd-common - 2024.03.09~695277a5-r1
hostapd-utils - 2024.03.09~695277a5-r1
ip-tiny - 6.7.0-r1
ip6tables-nft - 1.8.8-r2
iperf3 - 3.16-r1
ipset - 7.21-r1
iptables-mod-conntrack-extra - 1.8.8-r2
iptables-mod-ipopt - 1.8.8-r2
iptables-nft - 1.8.8-r2
irqbalance - 1.9.4-r1
iw - 5.19-r1
iwinfo - 2024.03.23~79a96150-r1
jansson4 - 2.14-r3
jshn - 2024.03.29~eb9bcb64-r1
jsonfilter - 2024.01.23~594cfa86-r1
kernel - 6.6.28~2ced0c8e89b37b67d968ad2fe049863e-r1
kmod-cfg80211 - 6.6.28.6.6.15-r1
kmod-crypto-aead - 6.6.28-r1
kmod-crypto-authenc - 6.6.28-r1
kmod-crypto-ccm - 6.6.28-r1
kmod-crypto-cmac - 6.6.28-r1
kmod-crypto-crc32c - 6.6.28-r1
kmod-crypto-ctr - 6.6.28-r1
kmod-crypto-des - 6.6.28-r1
kmod-crypto-gcm - 6.6.28-r1
kmod-crypto-geniv - 6.6.28-r1
kmod-crypto-gf128 - 6.6.28-r1
kmod-crypto-ghash - 6.6.28-r1
kmod-crypto-hash - 6.6.28-r1
kmod-crypto-hmac - 6.6.28-r1
kmod-crypto-hw-eip93 - 6.6.28-r1
kmod-crypto-kpp - 6.6.28-r1
kmod-crypto-lib-chacha20 - 6.6.28-r1
[... 890 lines cut ...]
0x08  0x00000000  down
0x09  0x00000000  down
0x0a  0x00000000  down
0x0b  0x00000000  down
0x0c  0x00000000  down
0x0d  0x00000000  down
0x0e  0x00000000  down
0x0f  0x00000000  down
0x10  0x00000000  down
0x11  0x00000000  down
0x12  0x00000000  down
0x13  0x00000000  down
0x14  0x00000000  down
0x15  0x00000000  down
0x16  0x00000000  down
0x17  0x00000000  down
0x18  0x00000000  down
0x19  0x00000000  down
0x1a  0x00000000  down
0x1b  0x00000000  down
0x1c  0x00000000  down
0x1d  0x00000000  down
0x1e  0x00000000  down
0x1f  0x9001040c  down
```

DragonBluep, 2024-04-28:
BTW, it seems that all affected devices are tplink routers.
https://github.com/openwrt/openwrt/issues/14596
https://github.com/openwrt/openwrt/issues/14673
I guess device tree has some issue during the boot.

arinc9, 2024-04-28:
@castillofrancodamian please test the image for your board at this link:

https://files.arinc9.com/OpenWrt/

Diff:

```
diff --git a/target/linux/generic/hack-6.1/940-net-dsa-mt7530-use-switch-MDIO-bus-for-PHY-muxing.patch b/target/linux/generic/hack-6.1/940-net-dsa-mt7530-use-switch-MDIO-bus-for-PHY-muxing.patch
new file mode 100644
index 0000000000..3acb21ab25
--- /dev/null
+++ b/target/linux/generic/hack-6.1/940-net-dsa-mt7530-use-switch-MDIO-bus-for-PHY-muxing.patch
@@ -0,0 +1,29 @@
+From 5f94d7358cceda721983b29245010a9d4ab0c808 Mon Sep 17 00:00:00 2001
+From: =?UTF-8?q?Ar=C4=B1n=C3=A7=20=C3=9CNAL?= <arinc.unal@arinc9.com>
+Date: Sun, 28 Apr 2024 13:38:46 +0300
+Subject: [PATCH] net: dsa: mt7530: use switch MDIO bus for PHY muxing
+MIME-Version: 1.0
+Content-Type: text/plain; charset=UTF-8
+Content-Transfer-Encoding: 8bit
+
+Signed-off-by: Arınç ÜNAL <arinc.unal@arinc9.com>
+---
+ drivers/net/dsa/mt7530.c | 2 +-
+ 1 file changed, 1 insertion(+), 1 deletion(-)
+
+diff --git a/drivers/net/dsa/mt7530.c b/drivers/net/dsa/mt7530.c
+index 7dc2f0ac8962..fa7a1ab292d2 100644
+--- a/drivers/net/dsa/mt7530.c
++++ b/drivers/net/dsa/mt7530.c
+@@ -2585,7 +2585,7 @@ mt7530_setup(struct dsa_switch *ds)
+ 			if (!phy_node)
+ 				continue;
+ 
+-			if (phy_node->parent == priv->dev->of_node->parent) {
++			if (phy_node->parent->parent == priv->dev->of_node) {
+ 				ret = of_get_phy_mode(mac_np, &interface);
+ 				if (ret && ret != -ENODEV) {
+ 					of_node_put(mac_np);
+-- 
+2.40.1
+
diff --git a/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts b/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts
index fb14bd7829..2e573fff27 100644
--- a/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts
+++ b/target/linux/ramips/dts/mt7621_mercusys_mr70x-v1.dts
@@ -145,27 +145,48 @@
 	nvmem-cell-names = "mac-address";
 };
 
-&mdio {
-	ethphy0: ethernet-phy@0 {
-		reg = <0>;
+&switch0 {
+	mdio {
+		#address-cells = <1>;
+		#size-cells = <0>;
+
+		ethphy0: ethernet-phy@0 {
+			reg = <0>;
+		};
+
+		ethphy1: ethernet-phy@1 {
+			reg = <1>;
+			interrupts = <1>;
+		};
+
+		ethphy2: ethernet-phy@2 {
+			reg = <2>;
+			interrupts = <2>;
+		};
+
+		ethphy3: ethernet-phy@3 {
+			reg = <3>;
+			interrupts = <3>;
+		};
 	};
-};
 
-&switch0 {
 	ports {
 		port@1 {
 			status = "okay";
 			label = "lan1";
+			phy-handle = <&ethphy1>;
 		};
 
 		port@2 {
 			status = "okay";
 			label = "lan2";
+			phy-handle = <&ethphy2>;
 		};
 
 		port@3 {
 			status = "okay";
 			label = "lan3";
+			phy-handle = <&ethphy3>;
 		};
 	};
 };
diff --git a/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts b/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts
index d814cba261..057f0b11a7 100644
--- a/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts
+++ b/target/linux/ramips/dts/mt7621_tplink_eap235-wall-v1.dts
@@ -189,27 +189,48 @@
 	nvmem-cell-names = "mac-address";
 };
 
-&mdio {
-	ethphy0: ethernet-phy@0 {
-		reg = <0>;
+&switch0 {
+	mdio {
+		#address-cells = <1>;
+		#size-cells = <0>;
+
+		ethphy0: ethernet-phy@0 {
+			reg = <0>;
+		};
+
+		ethphy1: ethernet-phy@1 {
+			reg = <1>;
+			interrupts = <1>;
+		};
+
+		ethphy2: ethernet-phy@2 {
+			reg = <2>;
+			interrupts = <2>;
+		};
+
+		ethphy3: ethernet-phy@3 {
+			reg = <3>;
+			interrupts = <3>;
+		};
 	};
-};
 
-&switch0 {
 	ports {
 		port@1 {
 			status = "okay";
 			label = "lan3";
+			phy-handle = <&ethphy1>;
 		};
 
 		port@2 {
 			status = "okay";
 			label = "lan2";
+			phy-handle = <&ethphy2>;
 		};
 
 		port@3 {
 			status = "okay";
 			label = "lan1";
+			phy-handle = <&ethphy3>;
 		};
 	};
 };
```

castillofrancodamian, 2024-04-28:
> @castillofrancodamian please test the image for your board at this link:
>
> https://files.arinc9.com/OpenWrt/
>
> Diff:
>
> ```
> diff --git [...]

What is that for and where do I have to try it?

arinc9, 2024-04-28:
It is for the issue you've reported here. You have to try it on the device where you experience the issue on.

KosmiK2001, 2024-04-30:
> It is for the issue you've reported here. You have to try it on the device where you experience the issue on. If you're specifically asking what to do with the diff, that's just for transparency. Just test the image.

check this version of your patch plz.
original dont applying correctrly.
https://pastebin.com/DbCfk3v9

arinc9, 2024-05-01:
Daniel commited my patches that is supposed to address this issue. Please try the latest snapshot.

KosmiK2001, 2024-05-01:
> Daniel commited my patches that is supposed to address this issue. Please try the latest snapshot.

6.1. Its ALIVE!!

castillofrancodamian, 2024-05-02:
@arinc9 The WAN is now working!

```
root@mr70x:~# mdio mdio-bus
 DEV      PHY-ID  LINK
0x00  0x03a29412  up
0x01  0x03a29412  up
0x02  0x03a29412  up
0x03  0x03a29412  up
0x04  0x03a29412  down
0x05  0x00000000  down
0x06  0x00000000  down
0x07  0x00000000  down
0x08  0x00000000  down
0x09  0x00000000  down
0x0a  0x00000000  down
0x0b  0x00000000  down
0x0c  0x00000000  down
0x0d  0x00000000  down
0x0e  0x00000000  down
0x0f  0x00000000  down
0x10  0x00000000  down
0x11  0x00000000  down
0x12  0x00000000  down
0x13  0x00000000  down
0x14  0x00000000  down
0x15  0x00000000  down
0x16  0x00000000  down
0x17  0x00000000  down
0x18  0x00000000  down
0x19  0x00000000  down
0x1a  0x00000000  down
0x1b  0x00000000  down
0x1c  0x00000000  down
0x1d  0x00000000  down
0x1e  0x00000000  down
0x1f  0x9000040c  up
root@mr70x:~# mdio mt7530-0
 DEV      PHY-ID  LINK
0x00  0x03a29412  up
0x01  0x03a29412  up
0x02  0x03a29412  up
0x03  0x03a29412  up
0x04  0x03a29412  down
0x05  0x00000000  down
0x06  0x00000000  down
0x07  0x00000000  down
0x08  0x00000000  down
0x09  0x00000000  down
0x0a  0x00000000  down
0x0b  0x00000000  down
0x0c  0x00000000  down
0x0d  0x00000000  down
0x0e  0x00000000  down
0x0f  0x00000000  down
0x10  0x00000000  down
0x11  0x00000000  down
0x12  0x00000000  down
0x13  0x00000000  down
0x14  0x00000000  down
0x15  0x00000000  down
0x16  0x00000000  down
0x17  0x00000000  down
0x18  0x00000000  down
0x19  0x00000000  down
0x1a  0x00000000  down
0x1b  0x00000000  down
0x1c  0x00000000  down
0x1d  0x00000000  down
0x1e  0x00000000  down
0x1f  0x9000040c  up
```

arinc9, 2024-05-02:
Time to close the issue then.

closed 2024-05-02 by castillofrancodamian