openwrt/openwrt #14945: crash during boot MT7621 owrt-master linux 6.1.82

wwortel, 2024-03-21:
### Describe the bug

On Ubiquiti EdgerouterX-SFP. Image for MT7621 compiles without issues. But after a sysupgrade the kernel will crash. Also loading ..kernel.bin from same compile process into RAM and booting into that has same problem.
[crash_log.txt](https://github.com/openwrt/openwrt/files/14692800/crash_log.txt)

### OpenWrt version

2e98d0af420d66d3bc9882ff7b026861f870c610

### OpenWrt release

2e98d0af420d66d3bc9882ff7b026861f870c610

### OpenWrt target/subtarget

MediaTek Ralink MIPS/MT7621 based boards

### Device

Ubiquiti EdgeRouter X SFP

### Image kind

Self-built image

### Steps to reproduce

compile image after git pull and sysupgrade from working earlier image the newly created ...sysupgrade.bin.
The kernel gets loaded and decompressed but crashes while starting. With serial interface connected loading the ...kernel.bin image using tftp/bootp and starting into that has same problem. See attached log snippet preceding the kernel panic. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

wwortel, 2024-03-21:
noticed that images for the Ubnt Edgerouter X SFP have been removed from snapshots because kernel 6.1 image is to big.
But ordinary compile from up to date github master branch still allows compilation for this device.
The commit number I got from the 'git show' command. 'git pull' gives 'up to date'. Am stuck now since cannot compile OpenWrt for the Edgerouter anymore .-(

ivanich, 2024-03-23:
Can confirm that I can successfully compile even with edgerouterx disable commit and have all the needed images generated, on the other hand I didn't flash this build with 6.1.82 kernel but still have 6.1.81 running on the device.

wwortel, 2024-03-23:
The Edgerouter-X-sfp has a PHY chip AR8033-AL1A. The driver for this, with function at8031_probe() is in driver/net/phy/qcom: at803x.c
Comparing OpenWrt kernel 6.1.82 with kernel.org, there are changes from later kernels backported, and the at803x.c has seen quite a bit of work between consecutive 6.x kernel versions. Perhaps there is some flaw in this backporting. Hope someone here is familiar with this part of driver code.

john-tho, 2024-03-25:
> at8031_probe() is in drivers/net/phy/qcom/at803x.c

I am pretty rusty, but from a very quick read, looks like `*priv` might be null dereferenced here,  initialised in `at8031_probe` before being allocated in `at803x_probe`?

Try this (move `priv` initialisation after  `at803x_probe` call)?
```patch
--- build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-6.1.82/drivers/net/phy/qcom/at803x.c.orig	2024-03-26 02:51:12.686689734 +1000
+++ build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-6.1.82/drivers/net/phy/qcom/at803x.c	2024-03-26 02:51:37.530961979 +1000
@@ -797,7 +797,7 @@
 
 static int at8031_probe(struct phy_device *phydev)
 {
-	struct at803x_priv *priv = phydev->priv;
+	struct at803x_priv *priv;
 	int mode_cfg;
 	int ccr;
 	int ret;
@@ -806,6 +806,8 @@
 	if (ret)
 		return ret;
 
+	priv = phydev->priv;
+
 	/* Only supported on AR8031/AR8033, the AR8030/AR8035 use strapping
 	 * options.
 	 */
```

ping @Ansuel from https://github.com/torvalds/linux/commit/25d2ba94005fac18fe68878cddff59a67e115554
25d2ba94005f (" net: phy: at803x: move specific at8031 probe mode check to dedicated probe")

---

https://elixir.bootlin.com/linux/v6.9-rc1/source/drivers/net/phy/qcom/at803x.c
```C
static int at803x_probe(struct phy_device *phydev)
{
	struct device *dev = &phydev->mdio.dev;
	struct at803x_priv *priv;
	int ret;

	priv = devm_kzalloc(dev, sizeof(*priv), GFP_KERNEL);
	if (!priv)
		return -ENOMEM;

	phydev->priv = priv;

	ret = at803x_parse_dt(phydev);
	if (ret)
		return ret;

	return 0;
}
```

```C
static int at8031_probe(struct phy_device *phydev)
{
	struct at803x_priv *priv = phydev->priv;
	int mode_cfg;
	int ccr;
	int ret;

	ret = at803x_probe(phydev);
	if (ret)
		return ret;

	/* Only supported on AR8031/AR8033, the AR8030/AR8035 use strapping
	 * options.
	 */
```

wwortel, 2024-03-25:
@john-tho 
Brilliant as inox steel, i.s.o. rust :-)
Just recompiled with the patch you suggested added to target/generic/hack-6.1 and now the kernel.bin image starts fine on the Ubnt Edgerouter X sfp. 
[730-net-phy-at803x-fix.patch.txt](https://github.com/openwrt/openwrt/files/14747869/730-net-phy-at803x-fix.patch.txt)
Thank you!. What is the proper way to introduce this fix into OpenWrt ?

Ansuel, 2024-03-25:
@wwortel I will handle it np.

wwortel, 2024-03-26:
two questions inspired by having looked into this driver:
1. the driver checks for a number of device tree blob entries, none of which are in the Edgerouter X sfp' dts.
Some entries have been added in recent kernel revisions. 
of_property_read_bool(node, "qca,keep-pll-enabled")
of_property_read_bool(node, "qca,disable-smarteee")
of_property_read_bool(node, "qca,disable-hibernation-mode")
of_property_read_u32(node, "qca,smarteee-tw-us-1g", &tw)
of_property_read_u32(node, "qca,smarteee-tw-us-100m", &tw)
of_property_read_u32(node, "qca,clk-out-frequency", &freq)
of_property_read_u32(node, "qca,clk-out-strength", &strength)
Should any of these be in the Edgerouter X SFP' dtb?

2. Noticed that default in the kernel config 'SFP cage support' is not enabled. Shouldn't this be enabled when the Edgerouter X SFP is selected as target?

Ansuel, 2024-04-02:
Just to update on this, patch got merged upstream and will be backported on stable kernel... 
If I have time I will take care of backporting a patch manually on openwrt.

john-tho, 2024-04-03:
Thanks for getting this fix upstream quickly Ansuel!

> backporting a patch manually on openwrt. (or it will be included in a while with some kernel bump)

We backport this seperate at8031 probe from 6.8 in [707-v6.8-06](https://github.com/openwrt/openwrt/blob/main/target/linux/generic/backport-6.6/707-v6.8-06-net-phy-at803x-move-specific-at8031-probe-mode-check.patch), so we will need to manually add. Either adjust there, or add the [fix patch](https://git.kernel.org/netdev/net/c/6a4aee277740) afterwards.

Ansuel, 2024-04-07:
Hi, found time to backport the fix for this.

Ansuel, 2024-04-07:
Fixed in 449b5701d726a0b7ef7371bb94d0f36645a7a2bc

closed 2024-04-07 by Ansuel