openwrt/openwrt #14947: WAX206: kernel errors - HW Problem, mt7915e timeouts, failed to set key

NilsRo, 2024-03-21:
### Describe the bug

Several errors are shown in the kernel.log. I do not know if they are related, as they happens with different components of the device:

Timeout of the mt7915e 5Ghz Wifi device 
```
Mar 19 13:37:32 OpenWrt5 kernel: [1956055.386206] mt7915e 0000:01:00.0: 
Message 00002ced (seq 11) timeout
Mar 19 13:37:53 OpenWrt5 kernel: [1956075.866179] mt7915e 0000:01:00.0: Message 00005aed (seq 12) timeout
Mar 19 13:38:13 OpenWrt5 kernel: [1956096.346164] mt7915e 0000:01:00.0: Message 0000aded (seq 13) timeout
Mar 19 13:38:34 OpenWrt5 kernel: [1956116.836165] mt7915e 0000:01:00.0: Message 00002ced (seq 14) timeout
Mar 19 13:38:54 OpenWrt5 kernel: [1956137.306144] mt7915e 0000:01:00.0: Message 00005aed (seq 15) timeout
Mar 19 13:38:54 OpenWrt5 kernel: [1956137.328350] wl1-ap0: HW problem - can not stop rx aggregation for 8c:7a:aa:51:5c:18 tid 0
Mar 19 13:38:54 OpenWrt5 kernel: [1956137.336827] wl1-ap0: HW problem - can not stop rx aggregation for 8c:7a:aa:51:5c:18 tid 1
Mar 19 13:38:54 OpenWrt5 kernel: [1956137.345188] wl1-ap0: HW problem - can not stop rx aggregation for 8c:7a:aa:51:5c:18 tid 5
Mar 19 13:38:54 OpenWrt5 kernel: [1956137.353579] wl1-ap0: HW problem - can not stop rx aggregation for 8c:7a:aa:51:5c:18 tid 6
```

Looks strange as OpenWrt sems to set a uncommon MAC address to the Wifi interfaces.
```
Mar 19 13:38:58 OpenWrt5 kernel: [1956141.066172] wl1-ap0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-12)
Mar 19 13:38:58 OpenWrt5 kernel: [1956141.372189] wl1-ap0: HW problem - can not stop rx aggregation for b0:7d:64:74:cf:4d tid 0
Mar 19 13:54:52 OpenWrt5 kernel: [1957095.395343] wl1-ap0: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-12)
Mar 19 13:56:14 OpenWrt5 kernel: [1957177.314330] wl1-ap1: failed to set key (2, ff:ff:ff:ff:ff:ff) to hardware (-12)
```

[kernel.log](https://github.com/openwrt/openwrt/files/14694137/kernel.log)

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Netgear WAX206

### Image kind

Official downloaded image

### Steps to reproduce

Do not know, happens from time to time. but not very often.

### Actual behaviour

Error happens from time to time with loosing Wifi

### Expected behaviour

No errors, stable Wifi

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-21:
Is it co-related with band roaming? `logread -e 02:00:00:00:00` related to one dual-band client may help

sed s/aa:bb/XX:YY/g may help to mask client octets.

NilsRo, 2024-03-21:
I looks like this. After the error occurs I see many IEEE 802.11: did not acknowledge authentication Response and band steering is broken that was not before. My iPhone losses its connection. 

```
Mar 19 11:45:14 OpenWrt5 hostapd: wl1-ap0: STA-OPMODE-SMPS-MODE-CHANGED aa:bb:cc:74:cf:4d dynamic
Mar 19 11:45:14 OpenWrt5 hostapd: wl1-ap0: STA-OPMODE-SMPS-MODE-CHANGED aa:bb:cc:74:cf:4d off
Mar 19 11:45:15 OpenWrt5 hostapd: wl1-ap0: STA-OPMODE-SMPS-MODE-CHANGED aa:bb:cc:74:cf:4d dynamic
Mar 19 12:02:38 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 12:02:38 OpenWrt5 hostapd: wl1-ap0: STA aa:bb:cc:6e:76:6b IEEE 802.11: associated (aid 1)
Mar 19 12:02:38 OpenWrt5 hostapd: wl1-ap0: AP-STA-CONNECTED aa:bb:cc:6e:76:6b auth_alg=ft
Mar 19 12:02:47 OpenWrt5 hostapd: wl1-ap0: AP-STA-DISCONNECTED aa:bb:cc:6e:76:6b
Mar 19 12:05:40 OpenWrt5 hostapd: wl1-ap2: AP-STA-DISCONNECTED aa:bb:cc:f2:66:51
Mar 19 12:05:43 OpenWrt5 hostapd: wl0-ap2: STA aa:bb:cc:f2:66:51 IEEE 802.11: authenticated
Mar 19 12:05:43 OpenWrt5 hostapd: wl0-ap2: STA aa:bb:cc:f2:66:51 IEEE 802.11: associated (aid 1)
Mar 19 12:05:43 OpenWrt5 hostapd: wl0-ap2: AP-STA-CONNECTED aa:bb:cc:f2:66:51 auth_alg=open
Mar 19 12:05:43 OpenWrt5 hostapd: wl0-ap2: STA aa:bb:cc:f2:66:51 RADIUS: starting accounting session 202FCA8E5571E7D3
Mar 19 12:05:43 OpenWrt5 hostapd: wl0-ap2: STA aa:bb:cc:f2:66:51 WPA: pairwise key handshake completed (RSN)
Mar 19 12:05:43 OpenWrt5 hostapd: wl0-ap2: EAPOL-4WAY-HS-COMPLETED aa:bb:cc:f2:66:51
Mar 19 12:13:55 OpenWrt5 hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Mar 19 12:13:55 OpenWrt5 hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Mar 19 12:15:17 OpenWrt5 hostapd: nl80211: wpa_driver_nl80211_event_receive->nl_recvmsgs failed: -5
Mar 19 12:15:17 OpenWrt5 hostapd: nl80211: wpa_driver_nl80211_event_receive->nl_recvmsgs failed: -5
Mar 19 12:15:17 OpenWrt5 hostapd: nl80211: wpa_driver_nl80211_event_receive->nl_recvmsgs failed: -5
Mar 19 12:29:37 OpenWrt5 hostapd: nl80211: wpa_driver_nl80211_event_receive->nl_recvmsgs failed: -5
Mar 19 12:29:37 OpenWrt5 hostapd: wl1-ap0: STA-OPMODE-SMPS-MODE-CHANGED aa:bb:cc:74:cf:4d off
Mar 19 12:40:53 OpenWrt5 hostapd: nl80211: wpa_driver_nl80211_event_receive->nl_recvmsgs failed: -5
Mar 19 12:40:53 OpenWrt5 hostapd: nl80211: wpa_driver_nl80211_event_receive->nl_recvmsgs failed: -5
Mar 19 13:38:54 OpenWrt5 hostapd: wl1-ap0: AP-STA-DISCONNECTED aa:bb:cc:51:5c:18
Mar 19 13:38:56 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:56 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:56 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:57 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:57 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:57 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:58 OpenWrt5 hostapd: wl1-ap0: STA aa:bb:cc:64:74:cf:4d WPA: group key handshake failed (RSN) after 4 tries
Mar 19 13:38:58 OpenWrt5 hostapd: wl1-ap0: AP-STA-DISCONNECTED b0:7d:64:74:cf:4d
Mar 19 13:38:58 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:58 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:59 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:59 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:38:59 OpenWrt5 hostapd: wl1-ap0: STA aa:bb:cc:51:5c:18 IEEE 802.11: did not acknowledge authentication response
Mar 19 13:39:00 OpenWrt5 hostapd: nl80211: kernel reports: key addition failed
Mar 19 13:39:00 OpenWrt5 hostapd: wl1-ap0: STA aa:bb:cc:51:5c:18 IEEE 802.11: did not acknowledge authentication response

```

brada4, 2024-03-21:
For workaround you can disable -r (FT) , at least Android or NetworkManager still steers smoothly, probably longer than normative 0.3s
I get those messages but no connectivity problems btw...

ntzb, 2024-04-08:
I see that msg too on my ax3000t wed-enabled AP:
```
Mon Apr  8 05:48:28 2024 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Mon Apr  8 05:48:28 2024 kern.err kernel: [129015.396466] platform 15010000.wed: Message 00000006 (seq 38421) timeout
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.403478] ------------[ cut here ]------------
Mon Apr  8 05:48:28 2024 kern.warn kernel: [129015.408432] WARNING: CPU: 1 PID: 11475 at 0xffffffc0009ff1e0 [mac80211@00000000ffe7390d+0x83000]
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.417567] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ip_tables chacha_neon cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables slhc poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha compat cls_flower act_vlan crypto_safexcel cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ip6_udp_tunnel
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.417754]  udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio gpio_button_hotplug aquantia crc_ccitt
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.516840] CPU: 1 PID: 11475 Comm: kworker/u4:4 Not tainted 5.15.150 #0
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.523876] Hardware name: Xiaomi Mi Router AX3000T (OpenWrt U-Boot layout) (DT)
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.531605] Workqueue: phy0 0xffffffc0009fd890 [mac80211@00000000ffe7390d+0x83000]
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.539520] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.546816] pc : 0xffffffc0009ff1e0 [mac80211@00000000ffe7390d+0x83000]
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.553767] lr : 0xffffffc0009ff0a8 [mac80211@00000000ffe7390d+0x83000]
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.560716] sp : ffffffc00bc1bc80
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.564366] x29: ffffffc00bc1bc80 x28: ffffffc008b12000 x27: 0000000000000001
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.571839] x26: ffffff80002a8b80 x25: ffffff80020a0880 x24: ffffffc008b12000
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.579311] x23: ffffffc0009fdd78 x22: ffffff80039120e8 x21: 0000000000000001
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.586785] x20: ffffff800628ca00 x19: ffffff8003910000 x18: ffffffc008a4a2a8
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.594257] x17: 0000000000000000 x16: 0000000000000000 x15: 00000000000004b3
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.601731] x14: 0000000000000191 x13: ffffffc00bc1b768 x12: ffffffc008aa22a8
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.609202] x11: 000002b18632a958 x10: ffffff80022017c0 x9 : 0000000000000001
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.616675] x8 : 0000000000000000 x7 : 0000000000a8a1e6 x6 : 0000000000000001
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.624147] x5 : ffffffc00895ff40 x4 : ffffffc008960010 x3 : 0000000000000000
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.631619] x2 : 0000000000000001 x1 : 0000000000000002 x0 : 00000000ffffff92
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.639090] Call trace:
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.641874]  0xffffffc0009ff1e0 [mac80211@00000000ffe7390d+0x83000]
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.648481]  0xffffffc0009fdd78 [mac80211@00000000ffe7390d+0x83000]
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.655086]  0xffffffc008050610
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.658565]  0xffffffc008051400
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.662042]  0xffffffc008058b4c
Mon Apr  8 05:48:28 2024 kern.debug kernel: [129015.665518]  0xffffffc00801581c
Mon Apr  8 05:48:28 2024 kern.warn kernel: [129015.668996] ---[ end trace 4f93986d2bb66b7d ]---
```

prokudin07, 2024-05-11:
I have the same problem 
`hostapd: nl80211: kernel reports: key addition failed`
lost WiFi connection with turn on FT and 
xiaomi ax3000t - AP only

Hurricos, 2024-07-09:
> hostapd: nl80211: kernel reports: key addition failed

I get these errors frequently on ath9k in a DAWN, v/r/k-enabled  environment, so it's not a mt76 issue.