openwrt/openwrt #14951: mt7628 drops ethernet on every network reload due to switch reconfiguration

wryun, 2024-03-22:
### Describe the bug

When you reload the network on an mt7628 (VoCore2), the ethernet link is dropped for a few seconds as the switch reconfigures.

This appears to be caused by:
```
reload_service() {
	local rv=0

	init_switch
	ubus call network reload || rv=1
	/sbin/wifi reload_legacy
	return $rv
}
```
which ends up doing:
```
swconfig dev switch0 load network
```
leading to:
```
[ 1681.443310] rt3050-esw 10110000.esw: port 0 link down
[ 1681.449286] br0: port 1(eth0.1) entered disabled state
```
This appears to be because of `esw_set_port_disable(esw, disable)` in `esw_apply_config(dev)` in `esw_rt3050.c`.

IMO, ideally either init_switch or swconfig would recognise that the switch was already configured in the desired way.

This is more annoying than one might expect, because it cause issues with LuCI detecting that the device has come back after a reconfiguration, only for it to potentially disappear again (as something else causes a network reload from the same config change).

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt76x8

### Device

VoCore2

### Image kind

Official downloaded image

### Steps to reproduce

Install image and run `service network reload` while watching kernel logs.

### Actual behaviour

```
[ 1681.443310] rt3050-esw 10110000.esw: port 0 link down
[ 1681.449286] br0: port 1(eth0.1) entered disabled state
```

### Expected behaviour

Nothing - there were no network changes.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2024-04-04:
I belive all network reloads would introduce temporairly drops, this is not the only device behaving like that

46cv8, 2024-04-14:
This is exactly what I've been investigating for the last few hours too. Any call to wifi reload, also causes a network reload, which causes all clients downloads to die. It would be really great if there was a way to fix this. Great detective work! This also happens on mt7621 (that's what my router is based on and it has the exact same issue.)

shripadrayewar, 2024-08-28:
anyone figured ot way to fix this? its really annoying when all connected users get diconnected upon any config change

wryun, 2024-08-28:
I've done this quick hack (won't work in all situations):

```
diff --git a/package/network/config/netifd/files/etc/init.d/network b/package/network/config/netifd/files/etc/init.d/network
index dc208c4ce0..14347cdd06 100755
--- a/package/network/config/netifd/files/etc/init.d/network
+++ b/package/network/config/netifd/files/etc/init.d/network
@@ -10,6 +10,7 @@ init_switch() {
 
        include /lib/network
        setup_switch
+       uci show network | grep '^network.@switch' > /var/swconfig.current
 }
 
 start_service() {
@@ -28,7 +29,9 @@ start_service() {
 reload_service() {
        local rv=0
 
+       if ! uci show network | grep '^network.@switch' | cmp -s /var/swconfig.current; then
                init_switch
+       fi
        ubus call network reload || rv=1
        /sbin/wifi reload_legacy
        return $rv
```

46cv8, 2024-08-29:
> include /lib/network
>         setup_switch

Hey @wryun ,

I'm curious what in setup_switch is causing the network to drop all connections, I'm not even sure where it's declared as I can't find any file in /lib/network that defines setup switch? 
I just ran "uci show network | grep '^network.@switch'" and it is just empty for me. I'm guessing that means it's not going to ever run init_switch after the first time. My system has lots of entries for "network.@bridge-vlan" I'm guessing I probably need to look for changes to the lines related to "network.@bridge-vlan" on my system. 
Still curious though what about "include /lib/network" and "setup_switch" is causing it to drop all connections....

wryun, 2024-08-30:
It's in switch.sh.

It calls the swconfig command as mentioned in the original post, which in my case ends up resetting the switch hardware as described (see package/network/config/swconfig/src/cli.c).