openwrt/openwrt #14954: Interface option network.wan6.sourcefilter='0' is ignored

jorti, 2024-03-22:
### Describe the bug

I'm configuring IPv6 masquerading as [described in the wiki](https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6), however setting the option `network.wan6.sourcefilter='0'` does not appear to have any effect as the default IPv6 route still have a "from" restriction:

```
# uci show network.wan6
network.wan6=interface
network.wan6.proto='dhcpv6'
network.wan6.reqaddress='try'
network.wan6.reqprefix='auto'
network.wan6.device='wan.20'
network.wan6.norelease='1'
network.wan6.sourcefilter='0'  <-----

# uci show network.wan
network.wan=interface
network.wan.proto='pppoe'
network.wan.device='wan.20'
network.wan.username='username'
network.wan.password='password'
network.wan.ipv6='auto'

# uci show firewall.@zone[6]
firewall.cfg46dc81=zone
firewall.cfg46dc81.name='wan'
firewall.cfg46dc81.output='ACCEPT'
firewall.cfg46dc81.masq='1'
firewall.cfg46dc81.mtu_fix='1'
firewall.cfg46dc81.input='DROP'
firewall.cfg46dc81.forward='DROP'
firewall.cfg46dc81.network='wan' 'wan6'
firewall.cfg46dc81.masq6='1' <----

# ip -6 route show dev pppoe-wan
default from 2001:0db8:8403:1e00::/56 via fe80::1 dev pppoe-wan  metric 512  <---
default from 2001:0db8:84ff:ffff::4f75:2050 via fe80::1 dev pppoe-wan  metric 512 <---
fe80::1 dev pppoe-wan  metric 256 
fe80::4f75:2050 dev pppoe-wan  metric 256 
multicast ff00::/8 dev pppoe-wan  metric 256 
```

The effects are that the hosts in the dmz network (which only have a ULA prefix) cannot access external IPv6 addresses.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

1. Set masq6 in WAN zone:
```
uci set firewall.@zone[1].masq6="1"
uci commit firewall
service firewall restart
```

2. Disable source filter in WAN6 interface:
```
uci set network.wan6.sourcefilter="0"
uci commit network
service network restart
```

3. Assign only ULA prefixes in internal dmz network:
```
uci set network.dmz.ip6class="local"
uci commit network
service network restart
```

4. Announce default route in internal dmz network:
```
uci set dhcp.dmz.ra_default="1"
uci commit dhcp
service odhcpd restart
```

### Actual behaviour

Default IPv6 routes have a "from" rescriction:

```
# ip -6 route show dev pppoe-wan | grep default
default from 2001:0db8:8403:1e00::/56 via fe80::1 dev pppoe-wan  metric 512
default from 2001:0db8:84ff:ffff::4f75:2050 via fe80::1 dev pppoe-wan  metric 512
```

### Expected behaviour

No source filter in the IPv6 routes. They should be something like this:

```
# ip -6 route show dev pppoe-wan | grep default
default via fe80::1 dev pppoe-wan  metric 512
default via fe80::1 dev pppoe-wan  metric 512
```

### Additional info

The underlying WAN device uses PPPoE.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-22:
src is source address preferred, absolutely required if there is no address on interface (like often case that pppoe tunnel inner and outer has same IP)

ptpt52, 2024-03-24:
PR https://github.com/openwrt/openwrt/pull/14466 should fix this issue