openwrt/openwrt #14956: (Network device 'eth1' link is down) and all devices dissociated due to "inactivity"

mnewiraq, 2024-03-22:
### Describe the bug

(Network device 'eth1' link is down) and all devices dissociated due to "inactivity".

the devices will not get connected back to the wireless network unless I reboot the router OR I toggle off and on the wireless button on the router rear panel.

### OpenWrt version

18.06.9

### OpenWrt release

r8077-7cbbab7246

### OpenWrt target/subtarget

NA

### Device

TL-WR841N

### Image kind

Official downloaded image

### Additional info

https://forum.openwrt.org/t/openwrt-wireless-issue/191917/7?u=mnewiraq

in the above reply to my post, you will notice that the issue will start with (Network device 'eth1' link is down).

all cables were checked and found healthy.

how to solve the issue and how to make openwrt not to dissociate any device?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mnewiraq, 2024-03-22:
`Tue Mar 19 20:14:18 2024 daemon.notice netifd: Network device 'eth1' link is down
Tue Mar 19 20:14:18 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Mar 19 20:14:18 2024 kern.info kernel: [93721.016024] eth1: link down
Tue Mar 19 20:14:18 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Tue Mar 19 20:14:18 2024 daemon.warn dnsmasq[1529]: no servers found in /tmp/resolv.conf.auto, will retry
Tue Mar 19 20:14:18 2024 daemon.notice netifd: wan (7324): udhcpc: received SIGTERM
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a2:0f:86:2a:5e:9a
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:28:f8:ff:40:a6
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 22:4c:82:9e:7a:f9
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 84:c8:a0:b6:96:e2
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 12:2e:eb:1b:d2:f6
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: AP-DISABLED
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue Mar 19 20:14:18 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Mar 19 20:14:18 2024 kern.info kernel: [93721.727231] device wlan0 left promiscuous mode
Tue Mar 19 20:14:18 2024 kern.info kernel: [93721.732030] br-lan: port 2(wlan0) entered disabled state
Tue Mar 19 20:14:18 2024 daemon.notice netifd: Network device 'wlan0' link is down
Tue Mar 19 20:14:19 2024 daemon.notice netifd: bridge 'br-lan' link is down
Tue Mar 19 20:14:19 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Mar 19 20:14:19 2024 kern.info kernel: [93722.057369] eth1: link up (100Mbps/Full duplex)
Tue Mar 19 20:14:19 2024 daemon.notice netifd: Network device 'eth1' link is up
Tue Mar 19 20:14:19 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Mar 19 20:14:19 2024 daemon.notice netifd: Interface 'wan' is setting up now
Tue Mar 19 20:14:19 2024 daemon.notice netifd: Interface 'wan6' has link connectivity
Tue Mar 19 20:14:19 2024 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Mar 19 20:14:19 2024 daemon.notice netifd: wan (7825): udhcpc: started, v1.28.4
Tue Mar 19 20:14:19 2024 daemon.notice netifd: wan (7825): udhcpc: sending discover
Tue Mar 19 20:14:21 2024 daemon.notice netifd: wan (7825): udhcpc: sending select for 192.168.100.23
Tue Mar 19 20:14:21 2024 daemon.notice netifd: wan (7825): udhcpc: lease of 192.168.100.23 obtained, lease time 259200
Tue Mar 19 20:14:22 2024 daemon.notice netifd: Interface 'wan' is now up
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: reading /tmp/resolv.conf.auto
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using local addresses only for domain test
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using local addresses only for domain onion
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using local addresses only for domain localhost
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using local addresses only for domain local
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using local addresses only for domain invalid
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using local addresses only for domain bind
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using local addresses only for domain lan
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using nameserver 8.8.8.8#53
Tue Mar 19 20:14:22 2024 daemon.info dnsmasq[1529]: using nameserver 8.8.4.4#53
Tue Mar 19 20:14:22 2024 user.notice firewall: Reloading firewall due to ifup of wan (eth1)
Tue Mar 19 20:14:30 2024 daemon.notice netifd: Network device 'eth1' link is down
Tue Mar 19 20:14:30 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Mar 19 20:14:30 2024 kern.info kernel: [93733.495834] eth1: link down
Tue Mar 19 20:14:30 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Tue Mar 19 20:14:30 2024 daemon.notice netifd: wan (7825): udhcpc: received SIGTERM
Tue Mar 19 20:14:32 2024 daemon.notice netifd: Network device 'eth1' link is up
Tue Mar 19 20:14:32 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Mar 19 20:14:32 2024 daemon.notice netifd: Interface 'wan' is setting up now
Tue Mar 19 20:14:32 2024 kern.info kernel: [93735.577353] eth1: link up (100Mbps/Full duplex)
Tue Mar 19 20:14:32 2024 daemon.notice netifd: Interface 'wan6' has link connectivity
Tue Mar 19 20:14:32 2024 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Mar 19 20:14:32 2024 daemon.notice netifd: wan (7981): udhcpc: started, v1.28.4
Tue Mar 19 20:14:33 2024 daemon.notice netifd: wan (7981): udhcpc: sending discover
Tue Mar 19 20:14:33 2024 daemon.notice netifd: wan (7981): udhcpc: sending select for 192.168.100.23
Tue Mar 19 20:14:33 2024 daemon.notice netifd: wan (7981): udhcpc: lease of 192.168.100.23 obtained, lease time 259200
Tue Mar 19 20:14:33 2024 daemon.warn dnsmasq[1529]: no servers found in /tmp/resolv.conf.auto, will retry
Tue Mar 19 20:14:33 2024 daemon.notice netifd: Interface 'wan' is now up
Tue Mar 19 20:14:33 2024 daemon.info dnsmasq[1529]: reading /tmp/resolv.conf.auto
Tue Mar 19 20:14:33 2024 daemon.info dnsmasq[1529]: using local addresses only for domain test
[... cut ...]
Tue Mar 19 20:15:22 2024 daemon.info dnsmasq-dhcp[1529]: DHCPACK(br-lan) 192.168.2.143 22:4c:82:9e:7a:f9 Pixel-6
Tue Mar 19 20:15:22 2024 daemon.warn dnsmasq-dhcp[1529]: not giving name OpenWrt.lan to the DHCP lease of 192.168.2.212 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.2.1
Tue Mar 19 20:15:22 2024 daemon.warn dnsmasq-dhcp[1529]: not giving name OpenWrt to the DHCP lease of 192.168.2.212 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.2.1
Tue Mar 19 20:15:29 2024 daemon.err uhttpd[903]: luci: accepted login on / for root from 192.168.2.163`

brada4, 2024-03-22:
You need to report against supported releases. 841N is marketing series of dozen different devices. If OpenWRT 23.05.2 does not support your hardware probably dd-wrt can still stretch devices' service life. All the attributes to fill in are on top of luci's first page.