openwrt/openwrt #14962: Archer A6 & TL-WR940N device keeps disconnecting and reconnecting

Andrei-Iosifescu123, 2024-03-23:
### Describe the bug

My Huawei P20 Pro keeps disconnecting and reconnecting to my Wi-Fi network. I have 2 routers, an Archer A6 v2 and TL-WR940N v6 (used as a wireless repeater). This issue happens on both routers, but I'm pretty sure it didn't happen before. Something like this also happens on my Kali Linux laptop, where one second the signal is full, the next second it is weak or zero and full again.

I've tried switching to FT over DS and changing the reassociation deadline to no avail. I hope there is a solution to this that doesn't involve throwing away the WR940N.

### OpenWrt version

`r23497-6637af95aa` & `r11405-2a3558b0de`

### OpenWrt release

`23.05.0` & `19.07.9`

### OpenWrt target/subtarget

`ath79/generic` &  `ar71xx/tiny`

### Device

`Archer A6v2` & `TL-WR940Nv6`

### Image kind

Self-built image

### Steps to reproduce

Connect to wifi, then move farther away towards the 2nd AP. Quickly after the device roams to the second AP, it will start disconnecting and reconnecting.

### Actual behaviour

When the device is roaming between APs, it does not stay on one AP, rather it disconnects and reconnects continuously, even though I am stationary.

### Expected behaviour

When the device is roaming between APs, it stays connected to the AP unless I move closer to the other AP.

### Additional info

### Logs

- Phone

```
03-23 13:20:01.860 15568 15568 D wpa_supplicant: nl80211: Roam event
03-23 13:20:01.860 15568 15568 D wpa_supplicant: nl80211: current bssid e4:**:**:**:ba:03
03-23 13:20:01.860 15568 15568 D wpa_supplicant: nl80211: Set drv->ssid based on req_ie to 'Andrei'
03-23 13:20:01.864 15568 15568 D wpa_supplicant: nl80211: Associated with 52:**:**:**:0d:0a
03-23 13:20:01.865 15568 15568 D wpa_supplicant: nl80211: Operating frequency for the associated BSS from scan results: 2412 MHz
03-23 13:20:01.865 15568 15568 D wpa_supplicant: wlan0: Event ASSOC (0) received
03-23 13:20:01.865 15568 15568 D wpa_supplicant: wlan0: Association info event
03-23 13:20:01.865 15568 15568 D wpa_supplicant: wlan0: freq=2412 MHz
03-23 13:20:01.865 15568 15568 D wpa_supplicant: RSN: PMKSA cache search - network_ctx=(nil) try_opportunistic=0 akmp=0x0
03-23 13:20:01.865 15568 15568 D wpa_supplicant: RSN: No PMKSA cache entry found
03-23 13:20:01.865 15568 15568 D wpa_supplicant: wlan0: RSN: PMKID from assoc IE not found from PMKSA cache
03-23 13:20:01.865 15568 15568 D wpa_supplicant: FT: Capability and Policy: 0x00
03-23 13:20:01.865 15568 15568 D wpa_supplicant: wlan0: State: COMPLETED -> ASSOCIATED
03-23 13:20:01.865 15568 15568 D wpa_supplicant: mdpp-wpa: MDPP switch is off
03-23 13:20:01.869 15568 15568 D wpa_supplicant: set wlan mtu to 1500
03-23 13:20:01.877 15568 15568 D wpa_supplicant: Notifying state change event to hidl control: 6
03-23 13:20:01.877 15568 15568 D wpa_supplicant: wlan0: freq[0]: 2412, flags=0x1
03-23 13:20:01.877 15568 15568 D wpa_supplicant: P2P: channels: 81:1,2,3,4,5,6,7,8,9,10,11,12,13 115:36,40,44,48 124:149,153,157,161 125:149,153,157,161,165 130:36,40,44,48
03-23 13:20:01.877 15568 15568 D wpa_supplicant: wlan0: Associated to a new BSS: BSSID=52:**:**:**:0d:0a
03-23 13:20:01.877 15568 15568 D wpa_supplicant: Notifying bssid changed to hidl control
03-23 13:20:01.887 15568 15568 D wpa_supplicant:    addr=52:**:**:**:0d:0a
03-23 13:20:01.890 15568 15568 D wpa_supplicant: wlan0: WPA: clearing AP WPA IE
03-23 13:20:01.890 15568 15568 I wpa_supplicant: wlan0: Associated with 52:**:**:**:0d:0a
03-23 13:20:01.890 15568 15568 D wpa_supplicant: wlan0: WPA: Association event - clear replay counter
03-23 13:20:01.890 15568 15568 D wpa_supplicant: wlan0: WPA: Clear old PTK
[... 299 lines cut ...]
### Diffconfig

```text
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Andrei-Iosifescu123, 2024-03-23:
There are 2 devices affected by this problem, so 2 targets, releases and versions. 

> Invalid Target/Subtarget reported. `` ath79/generic` & `ar71xx/tiny `` Is this from a supported device?

brada4, 2024-03-23:
What about 23.05.2 and 22.03.6 ? 
i.e you set up one of ^^ supported versions as WDS relay ( your WDS setup looks sane) then connect 

Why do you have STP? It does not help if you loop soc switch anyway.

Andrei-Iosifescu123, 2024-03-23:
> What about 23.05.2 and 22.03.6 ? i.e you set up one of ^^ supported versions as WDS relay ( your WDS setup looks sane) then connect
> 
> Why do you have STP? It does not help if you loop soc switch anyway.

I haven't tried other versions so I don't know. I have STP because I heard it can help prevent network loops.

brada4, 2024-03-23:
It keeps br-lan new interfaces blocked for 5 or 20 secs when coming up. If you loop like wifi with ethernet not sure whats worse - bringing down eth0.2 or the wifi adaptor....

brada4, 2024-03-23:
Also check if old version shows same encryptions in beacons like new one:
https://github.com/VREMSoftwareDevelopment/WiFiAnalyzer
FT wont happen if they differ.

brada4, 2024-03-23:
Also the country code is missing.

Andrei-Iosifescu123, 2024-03-24:
> Also check if old version shows same encryptions in beacons like new one: https://github.com/VREMSoftwareDevelopment/WiFiAnalyzer FT wont happen if they differ.

How can I do that?

brada4, 2024-03-24:
Shown is what overlaps with STA capabilities.
Just disable FT, it will roam maintaining connections as long as ap name and dhcp server remains same

Andrei-Iosifescu123, 2024-03-24:
> Shown is what overlaps with STA capabilities. Just disable FT, it will roam maintaining connections as long as ap name and dhcp server remains same

It seems my phone won't roam without disconnecting if FT is disabled. It will stay connected to the AP until it goes out of range completely, only then will it connect to the other AP.

brada4, 2024-03-24:
Then you are pretty much out of luck. You could have like usteer or batman on 22/23 to kick weaker clients across no matter their preference or protocol support, but at your present router hardware/software combo this is a no-go.
Some of these are <50 bucks just in case.
https://openwrt.org/toh/views/toh_available_16128_ax-wifi

Andrei-Iosifescu123, 2024-03-24:
> Then you are pretty much out of luck. You could have like usteer or batman on 22/23 to kick weaker clients across no matter their preference or protocol support, but at your present router [...]

I would buy a new router but I want to be sure it will fix the problem.

brada4, 2024-03-24:
You can try usteer between 2.4 and (now unused) 5ghz radios for all your compatibility measurements. Just add another AP, with same authentication/name in both bands and try if you can make phone happy.

Andrei-Iosifescu123, 2024-03-24:
> You can try usteer between 2.4 and (now unused) 5ghz radios for all your compatibility measurements. Just add another AP, with same authentication/name in both bands and try if you can make phone happy.

It only works on OpenWrt 21.02+, so I can't install it on the wr940n. I ran wireshark and found the deauth reason is `Previous authentication no longer valid (0x0002)`. Do you have any idea what this could mean?

brada4, 2024-03-24:
You can set log level to 1 as in https://openwrt.org/docs/guide-developer/debugging#logging_hostapd_behaviour to find more what leads to the problem. Increase log size at least on new router. It will take few minutes for it to "kick" device from 2.4 to 5Ghz and youll get yout log.
Note that there is no chance to fix any problem with old release, you need to do all testing against 2 radios in C6 and hope found settings work back on other router....

brada4, 2024-03-24:
Hey is it A6 or C6 ????

Andrei-Iosifescu123, 2024-03-24:
> Hey is it A6 or C6 ????

A6, but I think it is the same as C6, just a different branding.

Andrei-Iosifescu123, 2024-03-24:
> You can set log level to 1 as in https://openwrt.org/docs/guide-developer/debugging#logging_hostapd_behaviour to find more what leads to the problem. Increase log size at least on new router. It [...]

5Ghz is not enabled, only 2.4

brada4, 2024-03-24:
No way, v3 are identical, v1 v2 are different. You need to flash exactly what is on the label. The one wire difference (lucky you did not brick it) can be like LAN5 LED controlling 5GHz wifi power instead....
Test plan:
Reset router while flash exactly 23.05.2 for what is written on the label.
Set up two identical AP-s for 2.4 and 5GHz
Set up more log level and size
Restart router
Connect with your phone, it will take 5min for kick, or just tap to disconnect it and let it roam.
Next - replace wpad-minimal-mbedtls with wpad-mbedtls (I think it enables V on top of K or vice versa) -> reboot and try again
Next - if really needed - enable R - you need to enable FT OTA and set same mobility domain for AP-s in mobility domain.
no usteer in v18, but at least you can try to replicate "working" settings from ABC6

Andrei-Iosifescu123, 2024-03-24:
> No way, v3 are identical, v1 v2 are different. You need to flash exactly what is on the label. The one wire difference (lucky you did not brick it) can be like LAN5 LED controlling 5GHz wifi power instead.... Test plan: Reset router while flash exactly 23.05.2 for what is written on the label. Set up two identical AP-s for 2.4 and 5GHz Set up more log level and size Restart router Connect with your phone, it will take 5min for kick, or just tap in luci to disconnect it and let it roam. Next - replace wpad-minimal-mbedtls with wpad-mbedtls (I think it enables V on top of K or vice versa) -> reboot and try again Next - if really needed - enable R - you need to enable FT OTA and set same mobility domain for AP-s in mobility domain. no usteer in v18, but at least you can try to replicate "working" settings from ABC6

It is Archer A6, and flashed with 23.05.  I don't think I can fit the full wpad package.

brada4, 2024-03-24:
Can you check interop against supported release? If old release can be adapted fine, but if not it cannot be fixed.
Cmon 16MB flash is still plenty, you need to juggle with auc and firmware-selector on 8MB flash.

Andrei-Iosifescu123, 2024-03-24:
> You can set log level to 1 as in https://openwrt.org/docs/guide-developer/debugging#logging_hostapd_behaviour to find more what leads to the problem. Increase log size at least on new router. It [...]

Logs from TL-WR940N:
```
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: binding station to interface 'wlan0-1'
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: authentication OK (FT)
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-AUTHENTICATE.indication(88:11:96:cb:3a:d3, FT)
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: association OK (aid 4)
Sun Mar 24 15:30:05 2024 daemon.info hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: associated (aid 4)
Sun Mar 24 15:30:05 2024 daemon.notice hostapd: wlan0-1: AP-STA-CONNECTED 88:11:96:cb:3a:d3
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-REASSOCIATE.indication(88:11:96:cb:3a:d3)
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: binding station to interface 'wlan0-1'
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 WPA: event 6 notification
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 WPA: FT authentication already completed - do not start 4-way handshake
Sun Mar 24 15:30:15 2024 daemon.notice hostapd: wlan0-1: AP-STA-DISCONNECTED 88:11:96:cb:3a:d3
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 WPA: event 2 notification
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.1X: unauthorizing port
Sun Mar 24 15:30:15 2024 daemon.info hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: disassociated
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DISASSOCIATE.indication(88:11:96:cb:3a:d3, 8)
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DELETEKEYS.request(88:11:96:cb:3a:d3)
Sun Mar 24 15:30:16 2024 daemon.info hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Mar 24 15:30:16 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DEAUTHENTICATE.indication(88:11:96:cb:3a:d3, 2)
Sun Mar 24 15:30:16 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DELETEKEYS.request(88:11:96:cb:3a:d3)
```
Archer A6:
```
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: binding station to interface 'wlan0-1'
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: authentication OK (FT)
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-AUTHENTICATE.indication(88:11:96:cb:3a:d3, FT)
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: association OK (aid 4)
Sun Mar 24 15:30:05 2024 daemon.info hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: associated (aid 4)
Sun Mar 24 15:30:05 2024 daemon.notice hostapd: wlan0-1: AP-STA-CONNECTED 88:11:96:cb:3a:d3
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-REASSOCIATE.indication(88:11:96:cb:3a:d3)
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: binding station to interface 'wlan0-1'
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 WPA: event 6 notification
Sun Mar 24 15:30:05 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 WPA: FT authentication already completed - do not start 4-way handshake
Sun Mar 24 15:30:15 2024 daemon.notice hostapd: wlan0-1: AP-STA-DISCONNECTED 88:11:96:cb:3a:d3
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 WPA: event 2 notification
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.1X: unauthorizing port
Sun Mar 24 15:30:15 2024 daemon.info hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: disassociated
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DISASSOCIATE.indication(88:11:96:cb:3a:d3, 8)
Sun Mar 24 15:30:15 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DELETEKEYS.request(88:11:96:cb:3a:d3)
Sun Mar 24 15:30:16 2024 daemon.info hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Mar 24 15:30:16 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DEAUTHENTICATE.indication(88:11:96:cb:3a:d3, 2)
Sun Mar 24 15:30:16 2024 daemon.debug hostapd: wlan0-1: STA 88:11:96:cb:3a:d3 MLME: MLME-DELETEKEYS.request(88:11:96:cb:3a:d3)
```

Andrei-Iosifescu123, 2024-03-24:
> Can you check interop against supported release? If old release can be adapted fine, but if not it cannot be fixed. Cmon 16MB flash is still plenty, you need to juggle with auc and firmware-selector on 8MB flash.

I still don't think I can fit wpad full since I have openvpn and a couple other packages. If I can't solve this issue, i'll just buy a new router, maybe that works.

brada4, 2024-03-24:
The utmost priority is to flash right release for the hardware on your router.
you can handle wpad replacement via firmware selector or run auc (a minuscule package composing image with updates for installed packages using cloud) right after replacement to recover overlay space taken by duplicate files. If no better idea install package and check if you still have flash under half full, auc if not.

Andrei-Iosifescu123, 2024-03-24:
> The utmost priority is to flash right release for the hardware on your router. you can handle wpad replacement via firmware selector or run auc (a minuscule package composing image with updates for [...]

I don't think it's an issue with the Archer A6 firmware, everything else works as it should and disabling 802.11r solves the problem but my phone won't connect to the other AP until the signal drops.

brada4, 2024-03-24:
Standard behaviour is like that, just that modern router has 2 radios to tune settings so that your phone is not disconnecting repeatedly in place of just connecting to better place transparently.

brada4, 2024-03-25:
Android 9 needs to reassociate to network if FT gets enabled and cannot have FT and no-FT APs in the mash. Try various "conservative mode" options in android developer menu too.

Andrei-Iosifescu123, 2024-03-26:
> Android 9 needs to reassociate to network if FT gets enabled and cannot have FT and no-FT APs in the mash. Try various "conservative mode" options in android developer menu too.

I'm on Android 10

brada4, 2024-03-26:
What I am saying - get stable roaming between 2 bands on the more modern router FLASHED WITH CORRECT OPENWRT, then try to adapt same settings on the ancient one. It might be somewhat fearful to sysupgrade work vpn, but you have to do it one day anyway. Check sysupgrade -l if all ovpn config is kept in backup that is same as files kept over sysupgrade or auc.

Andrei-Iosifescu123, 2024-03-26:
> What I am saying - get stable roaming between 2 bands on the more modern router FLASHED WITH CORRECT OPENWRT, then try to adapt same settings on the ancient one. It might be somewhat fearful to [...]

I don't know if 802.11r also applies to roaming between the 2.4 and 5Ghz bands on the same router. Anyways, I've bought a new router to replace the old one and compiled an image with wpad for the other one. Hope that will help.

brada4, 2024-03-26:
You can repackage using firmware-selector, like online version of imagebuilder. FT certainly applies to local steering too.

Andrei-Iosifescu123, 2024-03-28:
> What I am saying - get stable roaming between 2 bands on the more modern router FLASHED WITH CORRECT OPENWRT, then try to adapt same settings on the ancient one. It might be somewhat fearful to [...]

The same issue on roaming between 2.4Ghz and 5Ghz on the Archer.

Andrei-Iosifescu123, 2024-03-31:
> You can run usteer without FT, FT is about shortcut authentication, likely prime suspect for key errors. (it is in the log at least) Also some Linux with networkmanager or aome windows 10+ or recent OSX could give extra asssurance it works, and you are adapting network to your phone essentially.

I know there are alternatives, but this is an issue that needs to be fixed.

brada4, 2024-03-31:
If your phone has broken crypto with FT, openwrt cannot make it send correct authentication. You need to reduce overlapping crypto modes to avoid buggy parts.
Best you can do is to adjust steering thresholds that weak signal triggers roaming without FT or radio reports from clients.

Andrei-Iosifescu123, 2024-03-31:
> If your phone has broken crypto with FT, openwrt cannot make it send correct authentication. You need to reduce overlapping crypto modes to avoid buggy parts. Best you can do is to adjust steering thresholds that weak signal triggers roaming without FT or radio reports from clients.

I don't think I can adjust the steering threshold. Also, this happens on my Linux laptop too, so the issue goes beyond my phone.

brada4, 2024-03-31:
In your logs there is FT failure against old AP only. Then cross-flashed kinda supported Archer device....

What are errors (around failed transtion) on archer at log level 1? 

usteer by default just moves clients out of overloaded band, the weak signal thresholds are not used by default. You can hint weak client to try to roam, like much older standard than recent K-V-R standards.

For archer - i suppose (ethtool -i) the network driver is ath10k-ct? You can try replacing it with "upstream" driver and firmware (keeping board) without -ct, it helps on some devices, on some not. Like make both ways in firmware-selector, it either gets much better or no-op, just download first "revert" firmware, then "forward".

Andrei-Iosifescu123, 2024-06-28:
> In your logs there is FT failure against old AP only. Then cross-flashed kinda supported Archer device....
>
> What are errors (around failed transtion) on archer at log level 1?
>
> usteer by default [...]

I have `kmod-ath10k`, no `ct`. This problem still persists, if I enable FT on both routers on both bands it frequently disconnects and reconnects. It might be a phone issue, I'm not sure. If that's the case, Huawei is to blame, every other phone doesen't have a problem.

redcat42, 2024-10-24:
how are you supplying power to both the routers?

are you using a single 12volt power supply and using a Y splitter cable to split the power to both the routers?

or

do both have their own individual power supply?

they both need to be powered separately/individually