openwrt/openwrt #14964: restarts network and reports an error

lqh280, 2024-03-24:
### Describe the bug

I build openwrt myself, but found DIR _package/base-files/files/sbin_ lost file _wifi_ . Causing inability to restart the _network_.  Err shows: (After creating the WiFi file, even if it is empty, the network can be restarted.)
[image]

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r25309-c378927ef8'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r25309-c378927ef8'
DISTRIB_TAINTS='no-all'

### OpenWrt version

r25309-c378927ef8

### OpenWrt release

r25309-c378927ef8

### OpenWrt target/subtarget

x86/64

### Device

X86-64

### Image kind

Self-built image

### Steps to reproduce

service network restart
or
/etc/init.d/network restart
can't work

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-24:
Error is harmless.

Try this with /etc/init.d/network 
```diff
stop_service() {
---        /sbin/wifi down
+++        [-x /sbin/wifi ] && /sbin/wifi down
        ifdown -a
```
Ill make due PR if you confirm it works

lqh280, 2024-03-26:
> Invalid Release reported. `r25309-c378927ef8` Is this from a clean repository?

yes, git clone from https://github.com/openwrt/openwrt

lqh280, 2024-03-26:
**main branches no file wifi**
[image]
**23.05 branches have file wifi**
[image]

brada4, 2024-03-26:
Space after square bracket sry

brada4, 2024-03-26:
it is no longer a dependency of netifd, thus needs conditional guard.
https://github.com/openwrt/openwrt/tree/main/package/network/config/wifi-scripts/files/sbin

brada4, 2024-03-26:
```
stop_service() {
---        /sbin/wifi down
+++        [ -x /sbin/wifi ] && /sbin/wifi down
        ifdown -a
```

brada4, 2024-03-26:
It will be fixed soon, if you want message go away you can download raw and replace init.d script by proposed one (maintaining +x attribute)

https://github.com/brada4/openwrt/blob/netif-wifi/package/network/config/netifd/files/etc/init.d/network

lqh280, 2024-03-27:
> ```diff
> stop_service() {
> ---        /sbin/wifi down
> +++        [ -x /sbin/wifi ] && /sbin/wifi down
>         ifdown -a
> ```

thanks, this code is working.

commit 207bfee8 (2024-11-30): netifd: check if /sbin/wifi exists before calling it

Avoid harmless error from network script by checking presence of
now-optional wifi support script, most notably confusing users of x86
snapshots.

Fixes: #14964

Link: https://github.com/openwrt/openwrt/pull/14986
commit 28de3ffe (2024-12-10): netifd: check if /sbin/wifi exists before calling it

Avoid harmless error from network script by checking presence of
now-optional wifi support script, most notably confusing users of x86
snapshots.

Fixes: #14964

Link: https://github.com/openwrt/openwrt/pull/14986

(cherry picked from commit 207bfee855320dc938f39e179b1d2e3b008140cb)
closed 2024-03-27 by lqh280