openwrt/openwrt #14965: Xiaomi AX3600 - ath11k c000000.wifi: failed to flush transmit queue

dr3amr, 2024-03-24:
### Describe the bug

Same behavior happens also with the Stable release, in an attempt to fix this, I switched to the Snapshot where same behavior is seen

Kernel log shows:
[   20.242269] br-lan: port 7(phy2-ap2) entered forwarding state
[   33.113494] l11: disabling
[65613.179701] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 22
[65618.299655] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 19
[65623.419694] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 28
[65628.539640] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 23
[65644.139631] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 7
[153265.425395] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 4
[153289.585351] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 18
[153294.705408] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 16
[153299.825320] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 18
[153304.945300] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 20
[153310.065291] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 7
[172834.224732] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
[173328.048288] device phy1-ap0 left promiscuous mode
[173328.048430] br-lan: port 4(phy1-ap0) entered disabled state

### OpenWrt version

r25591-362f9c41a4

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Start router
After some random time starts showing the error and devices disconnect from the Wifi, mainly though Intel AX201 ones, but phones also.

### Actual behaviour

devices disconnect from the Wifi, mainly though Intel AX201 ones, but phones also.

### Expected behaviour

Wifi to work properly

### Additional info

Around the same time I see in system logs also "Sun Mar 24 09:30:08 2024 daemon.notice hostapd: phy1-ap0: STA ... IEEE 802.11: did not acknowledge authentication response".

Same behavior happens also with the Stable release, in an attempt to fix this, I switched to the Snapshot where same behavior is seen

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dr3amr, 2024-04-16:
@Ansuel @robimarko ? I know there were discussions in the past on the old forum.

robimarko, 2024-04-16:
Well, sadly I dont know how to even begin debugging it

oded106e, 2024-04-20:
This problem happens to me too:
`Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.340120] ath11k c000000.wifi: dropping probe response as pending queue is almost full
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.340171] ath11k c000000.wifi: failed to queue management frame -28
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.348457] ath11k c000000.wifi: dropping probe response as pending queue is almost full
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.353660] ath11k c000000.wifi: failed to queue management frame -28
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.363079] ath11k c000000.wifi: dropping probe response as pending queue is almost full
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.368140] ath11k c000000.wifi: failed to queue management frame -28
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.377903] ath11k c000000.wifi: dropping probe response as pending queue is almost full
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.382653] ath11k c000000.wifi: failed to queue management frame -28
Sat Apr 20 23:08:07 2024 kern.warn kernel: [  915.392397] ath11k c000000.wifi: dropping probe response as pending queue is almost full
Sat Apr 20 23:19:16 2024 daemon.info hostapd: XXX-WIFI: STA 68:9a:XX:8b:XX:85 IEEE 802.11: disconnected due to excessive missing ACKs
`

oded106e, 2024-04-22:
Hey Guys, @Ansuel @robimarko @dr3amr 
What can be done about this situation? I tried switching versions - it still happens and disables the network.

oded106e, 2024-04-22:
I understand, with your permission I will ask a few more questions
I don't necessarily want to fix the problem right now on the latest version. 
I want to get a workaround (for normal internet connection)..
maybe you know why when I downgrade to a previous version of OpenWrt (for example 23.05.2 (r23630-842932a63d) the issue is still happend? 
Do I need to reset the router differently? (I tried to do it via luci > flash firmware > flash image). 
it possible to get snapshot version of january? I think that was the last upgrade I did that worked...

Thank you!

dr3amr, 2024-04-22:
@oded106e would seem though this is a ATH11 firmware issue, it was reported in the past also. It is reproduced both on latest stable and snapshot. There seems to be no fix at the moment (Qualcomm needs to fix it), last Snapshot, contain last fw version ustream, but issue still seen:

WLAN.HK.2.9.0.1-01977-QCAHKSWPL_SILICONZ-1
https://github.com/quic/upstream-wifi-fw/tree/main/ath11k-firmware/IPQ8074/hw2.0/2.9.0.1/WLAN.HK.2.9.0.1-01977-QCAHKSWPL_SILICONZ-1

For me this happens more often with Intel AX201 clients but also some Tuya ones.

Perhaps @Ansuel knows more about this.

AgustinLorenzo, 2024-08-06:
Hi boys,

About this failure see my comment: https://github.com/openwrt/openwrt/issues/15929#issuecomment-2227391307

Regards, Agustín

robimarko, 2024-08-06:
I have been testing that patch for a couple of days and it did not fix the TX queue flush falling for me.

I have now reflashed again and will it couple of more days

maurerle, 2025-10-03:
Can you please retest and check if #20293 fixes this issue :)