openwrt/openwrt #14966: openwrt.org Email - DKIM and DMARC not configured.

GoetzGoerisch, 2024-03-24:
### Describe the bug

As per discussion on IRC.

Mails from openwrt.org servers are currently declared as SPAM for many mail providers, as DMARC and DKIM is mandatory.

@jow- @paulfertser 

### OpenWrt version

2023.05.3

### OpenWrt release

2023.05.3

### OpenWrt target/subtarget

any

### Device

email

### Image kind

Official downloaded image

### Steps to reproduce

Get an email from wiki.openwrt.org

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

JuliusBairaktaris, 2024-04-10:
Hey @jow-, I would be happy to help you solve this issue. You can contact me at julius@bairaktaris.de

GoetzGoerisch, 2025-01-25:
DMARC is there.

closed 2025-01-25 by GoetzGoerisch