openwrt/openwrt #15008: WDR3600: ucode crashes on access to luci

es86de, 2024-03-29:
### Describe the bug

Webinterface is unstable. Browser shows:
``
Bad Gateway
The process did not produce any response
```

System Log entries for each crash:
```
Fri Mar 22 22:11:17 2024 kern.info kernel: [  106.424618] do_page_fault(): sending SIGSEGV to ucode for invalid read access from 00766564
Fri Mar 22 22:11:17 2024 kern.info kernel: [  106.433486] epc = 77da2dab in libucode.so.20230711[77d8a000+2f000]
Fri Mar 22 22:11:17 2024 kern.info kernel: [  106.439925] ra  = 77d969ef in libucode.so.20230711[77d8a000+2f000]
```

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WDR3600 v1

### Image kind

Official downloaded image

### Steps to reproduce

1. Install fresh 23.05.03 images for WDR3600, don't keep config
2. Browse through luci or refresh page several times
3. Approximately 1 of 5 page requests triggers ucode crash

### Additional info

Previous version without problems: 22.03.5

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-03-29:
Does not repeat on ArcherC7V2 right away.

Try this:
```
while true ; do ucode /www/cgi-bin/luci | wc ; done
```
If it does not crash for a minute try clicking in UI while still running ^^

If the console variant ever crashes we can catch the crash in debugger, will be a bit of challenge to get it up in /tmp.
The idea is to run loop command as gdb --batch, click on interface to provoke crash and then look at the process insides.

es86de, 2024-03-29:
Ucode crashes also from the console:
```
root@OpenWrt:~# while true ; do ucode /www/cgi-bin/luci | wc ; done
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
       78       221      3091
       78       221      3091
       78       221      3091
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
       78       221      3091
        0         0         0
Segmentation fault
        0         0         0
Segmentation fault
       78       221      3091
```

brada4, 2024-03-29:
Nice :-)
```
opkg install -d ram gdb
export LD_LIBRARY_PATH=/lib:/usr/lib:/tmp/lib:/tmp/usr/lib
gdb -version 
```
now replace command with
```
while true ; do /tmp/usr/bin/gdb --batch --args /usr/bin/ucode /www/cgi-bi/luci| wc ; done
```
When debugger breaks run
```
bt
info proc mappings
(exit with Ctrl-D)
```

ramdisk install will be gone on reboot
.

es86de, 2024-03-29:
Thank you for your support!

I installed gdb:
```
opkg install -d ram gdb
export PATH=$PATH:/tmp/bin:/tmp/usr/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/tmp/lib:/tmp/usr/lib
```

gdb runs:
```
root@OpenWrt:~# gdb --version
GNU gdb (GDB) 12.1
```

Then startet the loop:
```
root@OpenWrt:~# while true ; do echo RUN ; gdb --batch --args /usr/bin/ucode /www/cgi-bin/luci ; done
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
RUN
...
```
The debugger never broke the run. 

Then I tried to trigger it manually:
```
root@OpenWrt:~# gdb --args /usr/bin/ucode /www/cgi-bin/luci
(gdb) run
Starting program: /usr/bin/ucode /www/cgi-bin/luci
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.
Status: 403 Forbidden
x-luci-login-required: yes
...
<!-- Lua compatibility mode active: no -->
[Inferior 1 (process 3159) exited normally]
```

After several ``(gdb) run``, ucode crashed:
```
(gdb) run
Starting program: /usr/bin/ucode /www/cgi-bin/luci
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Program received signal SIGSEGV, Segmentation fault.
0x77f1edab in ?? ()
```
[... 68 lines cut ...]
        0x77e80000 0x77e9f000    0x1f000        0x0  r-xp   /usr/lib/ucode/fs.so
        0x77e9f000 0x77ea0000     0x1000     0xf000  r-xp   /usr/lib/ucode/fs.so
        0x77ea0000 0x77ea1000     0x1000    0x10000  rwxp   /usr/lib/ucode/fs.so
        0x77ea1000 0x77ea8000     0x7000        0x0  rwxp
        0x77ea9000 0x77eb2000     0x9000        0x0  rwxp
        0x77eb2000 0x77ee1000    0x2f000        0x0  r-xp   /lib/libgcc_s.so.1
        0x77ee1000 0x77ee2000     0x1000    0x1f000  r-xp   /lib/libgcc_s.so.1
        0x77ee2000 0x77ee3000     0x1000    0x20000  rwxp   /lib/libgcc_s.so.1
        0x77ee3000 0x77ee4000     0x1000        0x0  rwxp
        0x77ee4000 0x77f03000    0x1f000        0x0  r-xp   /usr/lib/libjson-c.so.5.2.0
        0x77f03000 0x77f04000     0x1000     0xf000  r-xp   /usr/lib/libjson-c.so.5.2.0
        0x77f04000 0x77f05000     0x1000    0x10000  rwxp   /usr/lib/libjson-c.so.5.2.0
        0x77f05000 0x77f06000     0x1000        0x0  rwxp
        0x77f06000 0x77f35000    0x2f000        0x0  r-xp   /usr/lib/libucode.so.20230711
        0x77f35000 0x77f36000     0x1000    0x2f000  r-xp   /usr/lib/libucode.so.20230711
        0x77f36000 0x77f37000     0x1000    0x30000  rwxp   /usr/lib/libucode.so.20230711
        0x77f37000 0x77f38000     0x1000        0x0  rwxp
        0x77f38000 0x77fe3000    0xab000        0x0  r-xp   /lib/libc.so
        0x77fe3000 0x77ff7000    0x14000        0x0  rwxp
        0x77ff7000 0x77ffa000     0x3000    0xaf000  rwxp   /lib/libc.so
        0x77ffa000 0x77ffc000     0x2000        0x0  rwxp
        0x7ffdb000 0x7fffc000    0x21000        0x0  rw-p   [stack]
        0x7fffc000 0x7fffd000     0x1000        0x0  r--p   [vvar]
        0x7fffd000 0x7fffe000     0x1000        0x0  r-xp   [vdso]
```

brada4, 2024-03-29:
@jow- hello

jow-, 2024-03-29:
Try to grab a coredump:
```
ulimit -c unlimited
while ucode /www/cgi-bin/luci; do :; done
```

Once a dump is written, attach it here.

es86de, 2024-03-30:
File is attached.

es86de, 2024-07-28:
Same problem with 23.05.4.

derjohn, 2024-08-06:
+1 , I am affected by that bug, too

derjohn, 2024-08-07:
WDL3600  OpenWrt 23.05.2, r23630-842932a63d ==> is affected

```
# while true ; do ucode /www/cgi-bin/luci | wc ; done
       78       221      3094
        0         0         0
Segmentation fault
       78       221      3094
       78       221      3094
```

But the "sister device"
WDL4300  OpenWrt 23.05.4, r24012-d8dd03c46f ==> not affected

```
root@OpenWrt-DaBo:~# while true ; do ucode /www/cgi-bin/luci | wc ; done
      129       381      4829
      129       381      4829
      129       381      4829
      129       381      4829
      129       381      4829
```

The devices don't have much differences except 2 vs 3 antennas.

P.S. The dot 2 release is also affected, I dont have 23.05.1 in production anymore

jow-, 2024-08-07:
@derjohn - could you arrange SSH access to one of the affected devices? Feel free to pm me at jo@mein.io or jow- on OFTC.

derjohn, 2024-08-07:
@jow- Check mail.

stargieg, 2024-08-07:
Can't reproduce with tl-wdr3600-v1 and OpenWrt 23.05.4 r24012-d8dd03c46f

derjohn, 2024-08-07:
> Can't reproduce with tl-wdr3600-v1 and OpenWrt 23.05.4 r24012-d8dd03c46f

I my case it's a least one v1.4 device .... I think I cannot check the version from the shell ? So further lookup would take some time.

es86de, 2024-08-07:
> Can't reproduce with tl-wdr3600-v1 and OpenWrt 23.05.4 r24012-d8dd03c46f

My device is HW version 1.5.

derjohn, 2024-08-08:
> ... cache, but it appears to go deeper than ucode.

So, ucodes uses a (external) compiler to make a just in time compile ... and that may be the broken component for the newer versions of the WDL3600 ? Will you open a follow up issue for that?

brada4, 2024-08-08:
It is a HW or kernel issue with memory stability, ucode compiles itself the bytecodes, just that result does not come back same as once stored in memory.

derjohn, 2024-08-08:
OMG, that's shit.
My v1.4 shows on the info page: Architecture Atheros AR9344 rev 2
Is the V1 .0 also a "rev 2" ?
There must be a differnce in HW spec, otherwise the V1.0 would also be affected.

brada4, 2024-08-08:
Worst of a kind:
https://en.m.wikipedia.org/wiki/Intel_Quark

Hope its just thermal isdue needing underclock.

es86de, 2024-08-08:
Thank you all for your support! We will have to stay on 22.03 with the WDR3600 then. 😢

derjohn, 2024-08-08:
> Worst of a kind: https://en.m.wikipedia.org/wiki/Intel_Quark
> 
> Hope its just thermal isdue needing underclock.

I really doubt that. Why the hell is only luci affected? Nothing else is crashing (Neither ssh, nor hostap, filesystem, etc.). Everythings works, except luci / ucode

derjohn, 2024-08-08:
> Atheros AR9344 rev

> Thank you all for your support! We will have to stay on 22.03 with the WDR3600 then. 😢

Why did you close the issue ? If noone looks into it, the bug will probably stay in future versions. Or did anyone open a followup issue somewhere?

es86de, 2024-08-09:
> Why did you close the issue ? If noone looks into it, the bug will probably stay in future versions. Or did anyone open a followup issue somewhere?

I have closed this issue, as the cause of the bug does not seem to be in OpenWRT/ucode, and therefore cannot be solved by this team.

@jow-, @brada4 should we reopen it?

brada4, 2024-08-09:
Keep it open, with stable repeater somebody may look into it, bug will not just go by itself.

jow-, 2024-08-13:
Unfortunately I am still unable to offer a solution but I was able to confirm that the issue is not ucode specific. Using this standalone reproducer, compiled with `.../mips-openwrt-linux-gcc -fpie -fpic -o /tmp/bug.mips /tmp/bug.c`

```c
#include <stdio.h>
#include <string.h>
#include <stdint.h>
#include <stdlib.h>

const uint8_t input[] = {
	0x47, 0x00, 0x00, 0x00, 0x09, 0x00, 0x00, 0x00, 0x71, 0x00, 0x00, 0x00, 0x72,
	0x00, 0x00, 0x00, 0x73, 0x00, 0x00, 0x00, 0x74, 0x00, 0x00, 0x00, 0x75, 0x00,
	0x00, 0x00, 0x76, 0x00, 0x00, 0x00, 0x77, 0x00, 0x00, 0x00, 0x78, 0x00, 0x00,
	0x00, 0x79, 0x00, 0x00, 0x00, 0x7a, 0x47, 0x00, 0x09, 0x00, 0x04, 0x00, 0x00,
	0x00, 0x01, 0x00, 0x00, 0x00, 0x7b, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00,
	0x03, 0x00, 0x00, 0x00, 0x7c, 0x03, 0x04, 0x00, 0x02, 0x64, 0x2f, 0x0f, 0x00,
	0x00, 0x00, 0x17, 0x80, 0x00, 0x00, 0x02, 0x80, 0x00, 0x00, 0x03, 0x80, 0x00,
	0x00, 0x0b, 0x7f, 0xff, 0xff, 0xfd, 0x0f, 0x00, 0x00, 0x00, 0x19, 0x80, 0x00,
	0x00, 0x01, 0x80, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0x07, 0x7f, 0xff, 0xff,
	0xfd, 0x0f, 0x00, 0x00, 0x00, 0x1b, 0x80, 0x00, 0x00, 0x03, 0x7f, 0xff, 0xff,
	0xfc, 0x7f, 0xff, 0xff, 0xfb, 0x7f, 0xff, 0xff, 0xfd, 0x0f, 0x00, 0x00, 0x00,
	0x1e, 0x7f, 0xff, 0xff, 0xff, 0x80, 0x00, 0x00, 0x04, 0x80, 0x00, 0x00, 0x05,
	0x0f, 0x00, 0x00, 0x00, 0x1f, 0x7f, 0xff, 0xff, 0xff, 0x80, 0x00, 0x00, 0x04,
	0x80, 0x00, 0x00, 0x05, 0x0f, 0x00, 0x00, 0x00, 0x20, 0x80, 0x00, 0x00, 0x00,
	0x80, 0x00, 0x00, 0x06, 0x80, 0x00, 0x00, 0x07, 0x0f, 0x00, 0x00, 0x00, 0x21,
	0x80, 0x00, 0x00, 0x01, 0x80, 0x00, 0x00, 0x00, 0x0f, 0x00, 0x00, 0x00, 0x23,
	0x7f, 0xff, 0xff, 0xff, 0x0c, 0x00, 0x00, 0x00, 0xa8, 0x01, 0x00, 0x00, 0x00,
	0x7d, 0x3d, 0x00, 0x00, 0x00, 0x01, 0x35, 0x10, 0x00, 0x00, 0x00, 0x24, 0x7f,
	0xff, 0xff, 0xf4, 0x80, 0x00, 0x00, 0x09, 0x80, 0x00, 0x00, 0x08, 0x10, 0x00,
	0x00, 0x00, 0x25, 0x80, 0x00, 0x00, 0x0a, 0x1c, 0x00, 0x00, 0x00, 0x00, 0x01,
	0x00, 0x00, 0x00, 0xae, 0x0f, 0x00, 0x00, 0x00, 0x26, 0x01, 0x00, 0x00, 0x00,
	0xb1, 0x0f, 0x00, 0x00, 0x00, 0x27, 0x01,
};

typedef struct {
	uint8_t *mem;
	size_t len;
} buf_t;

#define CHUNK_SIZE 8

static void add_byte(buf_t *buf, uint8_t byte)
{
	if (buf->len % CHUNK_SIZE == 0) {
		uint8_t *old = NULL;

		if (buf->len) {
			old = calloc(1, buf->len);

			if (!old) {
				fprintf(stderr, "calloc failure\n");
				exit(1);
			}

			memcpy(old, buf->mem, buf->len);
		}

		uint8_t *tmp = realloc(buf->mem, buf->len + CHUNK_SIZE);

		if (!tmp) {
			fprintf(stderr, "realloc failure\n");
			exit(1);
		}

		if (tmp != buf->mem) {
			fprintf(stderr, "base address change [%p->%p]\n",
				buf->mem, tmp);

			buf->mem = tmp;
		}

[... 73 lines cut ...]
00 01 00 00 00 7b 00 00 00 02 00 00 00 03 00 00 00 7c 03 04 00 02 64 2f 0f 00 
00 00 17 80 00 00 02 80 00 00 03 80 00 00 0b 7f ff ff fd 0f 00 00 00 19 80 00 
00 01 80 00 00 00 80 00 00 07 7f ff ff fd 0f 00 00 00 1b 80 00 00 03 7f ff ff 
fc 7f ff ff fb 7f ff ff fd 0f 00 00 00 1e 7f ff ff ff 80 00 00 04 80 00 00 05 
0f 00 00 00 1f 7f ff ff ff 00 00 00 04 80 00 00 05 0f 00 00 00 20 80 00 00 00 
80 00 00 06 80 00 00 07 0f 00 00 00 21 80 00 00 01 80 00 00 00 0f 00 00 00 23 
7f ff ff ff 0c 00 00 00 a8 01 00 00 00 7d 3d 00 00 00 01 35 10 00 00 00 

FINAL MEMORY:
47 00 00 00 09 00 00 00 71 00 00 00 72 00 00 00 73 00 00 00 74 00 00 00 75 00 
00 00 76 00 00 00 77 00 00 00 78 00 00 00 79 00 00 00 7a 47 00 09 00 04 00 00 
00 01 00 00 00 7b 00 00 00 02 00 00 00 03 00 00 00 7c 03 04 00 02 64 2f 0f 00 
00 00 17 80 00 00 02 80 00 00 03 80 00 00 0b 7f ff ff fd 0f 00 00 00 19 80 00 
00 01 80 00 00 00 80 00 00 07 7f ff ff fd 0f 00 00 00 1b 80 00 00 03 7f ff ff 
fc 7f ff ff fb 7f ff ff fd 0f 00 00 00 1e 7f ff ff ff 80 00 00 04 80 00 00 05 
0f 00 00 00 1f 7f ff ff ff 00 00 00 04 80 00 00 05 0f 00 00 00 20 80 00 00 00 
80 00 00 06 80 00 00 07 0f 00 00 00 21 80 00 00 01 80 00 00 00 0f 00 00 00 23 
7f ff ff ff 0c 00 00 00 a8 01 00 00 00 7d 3d 00 00 00 01 35 10 00 00 00 24 7f 
ff ff f4 80 00 00 09 80 00 00 08 10 00 00 00 25 80 00 00 0a 1c 00 00 00 00 01 
00 00 00 ae 0f 00 00 00 26 01 00 00 00 b1 0f 00 00 00 27 01 

root@luemmel-og:~# 
```

This rules out ucode and, worryingly, hints at the possibility that arbitrary executables might be affected by random memory alternations on this platform

derjohn, 2024-08-14:
@jow- Thanks for making the issue fully reproducible und bisecting it. I still wonder why I don't have more issue than that what I see in luci. The routing and wifi stuff seems not to be affected, so it only hits the userspace, i.e. something in the kernel compile process "knows" about the issue ? How can I lookup the exact platform / cpu id to compare on which ones it isn't working?

jow-, 2024-08-14:
I added the latest test case code here:
https://gist.github.com/jow-/a73da31cce8ef3a31584401e50cfe618

I'll also attach the test executable here. Would appreciate if other affected users here try this on their own devices and provide the resulting output here (if any). Sometimes it takes a while for the bug to manifest, so let the binary run for a while.

Executable:
[bug.mips.gz](https://github.com/user-attachments/files/16609494/bug.mips.gz)

brada4, 2024-08-18:
@ViperRu any particular router or really yourself a human person at fault?

brada4, 2024-08-18:
@ViperRu thanks for report, more interesying if other similar devices are affected.

derjohn, 2024-08-19:
> @brada4 I have the same problem on my WDR3600 with same simptoms (crash luci only).

@brada4 Which Version of the WDR3600 ? (printed on the bottom sticker)

brada4, 2024-08-19:
I dont own the device, only tried to explain most popular one bit flip cause.

jow-, 2024-08-21:
Some folks on IRC helped testing on further devices. Seems that at least the board revisions v1.4 and v1.5 of the WDR3600 are not affected by this issue.

To those who can reproduce the bug; what's the hardware revision and - in case you're able to open the device case - can you determine the RAM vendor used?

derjohn, 2024-08-21:
@jow- Isn't it vice versa? V1.4 and V1.5 _are_ affected in my case. 
I can open one and lookup RAM vendor.

sur5r, 2024-08-29:
I have one of these but inaccessible ATM. It's still on 19.x and I might be able to test stuff in a couple of weeks. Will watch this issue and report back.

es86de, 2024-09-01:
I own two WDR3600. One V1.1 without problems, and one V1.5 with ucode crashes.

V1.1:
[image]
[image]
[image]

V1.5:
[image]
[image]
[image]

brada4, 2024-09-02:
Please macro zoom on hynix memory chips in 2nd device?

es86de, 2024-09-02:
> Please macro zoom on hynix memory chips in 2nd device?

[image]

jow-, 2024-09-05:
I guess this rules out the ram vendor / kind theory.

pepe2k, 2024-09-05:
@es86de is your device running factory/vendor's U-Boot bootloader? Are you able to hookup serial console and provide logs from bootup?

derjohn, 2024-10-15:
Upgraded to OpenWrt 23.05.5
(  r24106-10cc5fcd00 )

The behavior of luci UI is still the same (I was hoping for some random kernel fix )

gbrd, 2024-10-19:
same problem openwrt 23.05.5 / TP-Link TL-WDR3600 v1

es86de, 2024-12-22:
I just tried OpenWrt 24.10.0-rc3 r28202-8667ca841b and it runs much more stable now on the V1.5 device. I haven't had any browser errors so far.

Although the behaviour of luci seems to be as intended, there is still a memory failure:
```
root@OpenWrt:~# date; /tmp/bug.mips; date
Sun Dec 22 11:15:20 UTC 2024
MEMORY CORRUPTION???

OLD MEMORY:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 7f ff 00 00 00 00 00

NEW MEMORY (different bytes suffixed with !):
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 7f ff 00 00 80!00 00

FINAL MEMORY:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 7f ff 00 00 80 00 00 00

Failure after 1985968 runs

Sun Dec 22 11:19:15 UTC 2024
```

es86de, 2024-12-29:
With OpenWrt 24.10.0-rc4 r28211-d55754ce0d / LuCI openwrt-24.10 branch 24.357.59006~41cbd31 ucode still crashes but less often:
```
root@OpenWrt:~# while true ; do ucode /www/cgi-bin/luci | wc ; done
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       74       155      1545
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       74       155      1545
       78       217      2932
       78       217      2932
       74       155      1545
       78       217      2932
       74       155      1545
       78       217      2932
       78       217      2932
       74       155      1545
       78       217      2932
Segmentation fault
        0         0         0
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
        0         0         0
Segmentation fault
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
       78       217      2932
```

es86de, 2025-02-02:
Summary of @jow-'s [bug.mips](https://github.com/user-attachments/files/16609494/bug.mips.gz) test on my V1.5 device:
```
22.03   no failure in 6.5 h
23.05   failure after 12 min (6772164 runs)
24.10   failure after 4 min (1985968 runs)
```

derjohn, 2025-03-18:
I switched to OpenWrt 24.10.0 r28427-6df0e3d02a, and at least I wasn't able too see segfault with this test
while true ; do ucode /www/cgi-bin/luci | wc ; done

Hardware:
Model	TP-Link TL-WDR3600 v1
Architecture	Atheros AR9344 rev 2